Louisvdw/dbus-serialbattery #852: Rename of the jk-bms only one times possible

gustogyver, 2023-11-05:
### Describe the problem

Hello
I‘ve got a problem by renaming my jk-bms. Only at the very first time when I plug in the bms to my pi with a new flashed sd card, I can rename my bms! After a reboot or unplug/replug, the name is gone and the opportunity to rename it! 
I mean, the line to rename the bms is totally gone.
What I’m doing wrong?
THX

### Driver version

The last one

### Venus OS device type

Raspberry Pi

### Venus OS version

3.10

### BMS type

JKBMS / Heltec BMS

### Cell count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
…
```

### Relevant log output

```shell
…
```

mr-manuel, 2023-11-05:
Since you did not compile the form like you should we cannot help you. Please add the needed data.

nik-bok, 2023-11-09:
Same here, two JKBMSes, one of them got permanent name and I can see custom name line in config file. That BMS name persist after restart but the other one is locked and no custom name in config file. After that there is no way to change name, change name row in GUI is gone.
If I change/add custom name tag with correct USB port of second BMS in config file it is ignored. Device is presented with default JKbms name and name can not be changed.

I hope you will sort this out soon because driver is not really useful as it is...

VenusOS 3.11, serial V1.0.20230728? (last dev)

mr-manuel, 2023-11-09:
Same to you. Without debugging data I cannot do anything.

nik-bok, 2023-11-09:
> Same to you. Without debugging data I cannot do anything.

Could you be more specific, what debugging data would you like to have

mr-manuel, 2023-11-09:
https://github.com/Louisvdw/dbus-serialbattery/issues/new?assignees=&labels=bug&projects=&template=bug_report.yml

gustogyver, 2023-11-10:
Sorry, but I don´t understand how I get the data you need!

mr-manuel, 2023-11-10:
And I don't understand what you don't understand :-)

probably you did not read carefully **ALL** the text when opening an issue. There is all described in every detail on how you find and get the data we need. There are also links you can click on.

If you have problems to understand english try with a translator or a friend that speaks english well.

I think the easiest would be, if you create a [new issue](https://github.com/Louisvdw/dbus-serialbattery/issues/new?assignees=&labels=bug&projects=&template=bug_report.yml) and read carefully the complete form and paste the needed data.

gustogyver, 2023-11-16:
GNU nano 4.9.3                  /u-boot/config.txt
kernel=u-boot.bin

[pi2]
device_tree=bcm2709-rpi-2-b.dtb

[pi3]
device_tree=bcm2710-rpi-3-b.dtb
dtoverlay=miniuart-bt
core_freq=400
core_freq_min=400

[pi3+]
device_tree=bcm2710-rpi-3-b-plus.dtb

[pi02]
device_tree=bcm2710-rpi-zero-2-w.dtb
dtoverlay=miniuart-bt,krnbt=on

[all]
dtparam=spi=on

closed 2023-11-25 by mr-manuel