Louisvdw/dbus-serialbattery #859: Battery gets charged above 55.2V

codeworkx, 2023-11-10:
### Describe the bug

We use a dynamic electricity tariff and control the charging and discharging of our battery via a node red flow. If the battery is to be charged to 100%, we set the minimum SoC to 100%. The charging process would now stop at 100%, even if the battery voltage has not reached 55.2V. To counter this problem, we set the mains setpoint to 32767W, which results in the battery becoming full.

Now it has already happened twice that the battery voltage of 55.2V has been exceeded. It looks like the regulation is slowly building up. The first time, the BMS disconnected due to overvoltage. The second time we were able to intervene manually at just under 56V and stopped the process. Important note: As the battery had not been full for some time, there was a large imbalance of 0.09V between the lowest and highest cell. We are not sure whether this could have had an influence. Maybe it's a VenusOS bug.

We are currently using the ESS mode "Keep batteries charged" to achieve our goal. So far, we have not encountered any problems with this.
[image]

### How to reproduce

Set min SoC to 100% and grid setpoint to 32767W.

### Expected behavior

The charging process should not exceed a battery voltage of 55.2V.

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.11

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 125.0

; -- CVL reset based on cell voltage diff (linear mode)
; Specify cell voltage diff where CVL limit is kept until diff is equal or lower
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.005

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.47, 3.45, 3.40, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.02, 0.05,  0.3,  0.5,    1

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  12, 14, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 12, 40,  45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3, 0.4,  1,  1, 0.5,  0

; Charge current limits
CC_CURRENT_LIMIT1_FRACTION = 0.2
CC_CURRENT_LIMIT2_FRACTION = 0.4
CC_CURRENT_LIMIT3_FRACTION = 0.6

; Discharge current limits
DC_CURRENT_LIMIT1_FRACTION = 0.2
DC_CURRENT_LIMIT2_FRACTION = 0.3
DC_CURRENT_LIMIT3_FRACTION = 0.5

; --------- Additional settings ---------
; Specify only one BMS type to load else leave empty to try to load all available
; -- Available BMS:
; Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; -- Available BMS, but disabled by default:
; https://louisvdw.github.io/dbus-serialbattery/general/install#how-to-enable-a-disabled-bms
; Ant, MNB, Sinowealth
BMS_TYPE = Jkbms

; Battery capacity (amps), if the BMS does not support reading it
BATTERY_CAPACITY = 125
```

### Relevant log output

```shell
See Picture
```

mr-manuel, 2023-11-10:
>  The charging process would now stop at 100%, even if the battery voltage has not reached 55.2V.

Why should the charging stop, if the battery has not reached CVL? What is your CVL in this case and what is the voltage drop between charger and battery?

> Now it has already happened twice that the battery voltage of 55.2V has been exceeded. It looks like the regulation is slowly building up.

Probably because of linear charging mode and the cell imbalance. You set `CELL_VOLTAGES_WHILE_CHARGING` to `3.55` V. Try to change it to `3.46` V. A cell difference of 5 % of the nominal cell voltage is not a big problem.

Please post also your CVL, CCL and DCL graph from the same moment.

Are you aware of [Dynamic ESS](https://www.victronenergy.com/live/drafts:dynamic_ess)?

codeworkx, 2023-11-10:
If you set the minimum soc in the ess menu to 100 %, it will be charged exactly to 100%, even if CVL has not been reached. This happens when the battery has not been fully charged for a long time and the SOC display is therefore inaccurate.

CVL is 55.2V.

MAX_CELL_VOLTAGE  = 3.450
16 Cells.

Does this mean that CVL will exceed the desired charging voltage of 55.2 V due to the linear limitation? I assumed this was just limiting the charging current?
That would be quite dangerous if the cells are heavily unbalanced for some reason.

mr-manuel, 2023-11-11:
> Does this mean that CVL will exceed the desired charging voltage of 55.2 V due to the linear limitation? I assumed this was just limiting the charging current?

It should not. Please post the requested screenshots to verify that.

> Dynamic ESS is known. But we're not happy with it in its current state. Guess there are some bugs left.

They just released some bug fixes yesterday.

codeworkx, 2023-11-19:
[image]
[image]
[image]

Seems to always happen in "Absorption dynamic".

mr-manuel, 2023-11-19:
> Please post also your CVL, CCL and DCL graph from the same moment.

This was requested in https://github.com/Louisvdw/dbus-serialbattery/issues/859#issuecomment-1806452721. You can enable it in the VRM Portal under Advanced -> Settings (top right) -> Battery Monitor -> BMS Charge and Discharge Limits.

It looks like this:

[image]

codeworkx, 2023-11-20:
[image]
[image]

Would it hurt to set CCL to 0 if battery pack voltage goes above CVL ?

codeworkx, 2023-11-21:
[image]
[image]
[image]
[image]

Happened again today.

mr-manuel, 2023-11-23:
Are you off or ongrid? I think the driver cannot do mucn in this case, since the charger is not doing what the driver needs.

codeworkx, 2023-11-26:
Ongrid.

mr-manuel, 2023-11-29:
Please open a question in the Victron Energy Community forum. The driver sets the voltage correctly.

mr-manuel, 2023-12-01:
You could test https://github.com/Louisvdw/dbus-serialbattery/pull/882, maybe it‘s changing something for you.

pos-ei-don, 2023-12-01:
Isnt it this Feature? 
Search for 0.4V in this link. 
Somwwhere with mqtt, you can change this value to a more narrow value. 

https://www.victronenergy.com/media/pg/Venus_GX/en/dvcc---distributed-voltage-and-current-control.html

mr-manuel, 2023-12-04:
@pos-ei-don thank you, for sharing this link. Indeed it has to be this. Therefore it's not a driver bug.

[11.6. DVCC for systems with the ESS Assistant](https://www.victronenergy.com/media/pg/Venus_GX/en/dvcc---distributed-voltage-and-current-control.html#UUID-0cda63b2-c80b-e81b-e174-f6a91ca5f848)

> A fixed solar offset of 0.4V (value for 48V system, divide by 4 for 12V) is applied when ESS-mode is set to Optimised in combination with the Feed-in excess solar charger power-setting enabled, or when ESS-mode is set to Keep batteries charged.