Louisvdw/dbus-serialbattery #861: Many attempts to have battery be recognized.

rldoose, 2023-11-11:
### Describe the problem

I have tried to get my battery to be recognized with no success.  Have tried on RPi 3 and RPi 4 with same results.  On each I have tried 2 different 485 converters which I have confirmed are on USB0.  Running svstat /service/dbus-serialbattery.tty* shows a new PID after running about 35 sec.

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.12

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.5
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.650
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
;BLUETOOTH_BMS = LltJbd_Ble C3:27:20:37:C1:A4
```

### Relevant log output

```shell
root@raspberrypi2:~# tail -n 120 /data/log/serial-starter/current | tai64nlocal
2023-11-11 02:09:31.763214500 *** starting serial-starter ***
2023-11-11 02:09:32.451225500 serstart starting
2023-11-11 02:09:32.464111500 INFO: loading config file /etc/venus/serial-starter.conf
2023-11-11 02:09:33.701595500 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
2023-11-11 02:09:36.563239500 ERROR: no service template for dbus-serialbattery
2023-11-11 02:09:40.424846500 INFO: Create daemontools service gps-dbus.ttyUSB0
2023-11-11 02:09:46.709863500 INFO: Start service gps-dbus.ttyUSB0 once
2023-11-11 02:09:57.577881500 INFO: Create daemontools service vedirect-interface.ttyUSB0
[... 116 lines cut ...]
sbattery

root@raspberrypi2:~# 
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 27 seconds, normally up
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 33 seconds, normally up
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 8149) 2 seconds
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 8149) 19 seconds
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 6 seconds, normally up
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 21 seconds, normally up
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 8716) 2 seconds
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 8716) 31 seconds
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: up (pid 8716) 35 seconds
root@raspberrypi2:~# svstat /service/dbus-serialbattery.tty*
/service/dbus-serialbattery.ttyUSB0: down 1 seconds, normally up
root@raspberrypi2:~#
```

mr-manuel, 2023-11-11:
1. Is the BMS working, if you connect it to the PC?
2. Did the BMS work in a previous version of this driver like `v0.14.3` or is this a new installation?
3. Did you try the nightly of the dev branch?

beliffm, 2023-11-11:
4. is the interface *really* RS485? The connector marked RS485 is not a sufficient indication for the presence of the interface. Maybe a picture showing the label might shed some light on the mattee

rldoose, 2023-11-11:
The BMS works via BLE and the Xiaoxiang app.  This is a new installation so I will try the builds you suggest below.  ReaganSent from iPhone

rldoose, 2023-11-13:
OK, I’m now questioning if my BMS is actually 485 (as you mentioned earlier).  I was thrown by  "Other BMS that use the Xiaoxiang phone app”, but I now realize that this probably only applies to Bluetooth as that is how the app connects to the BMS? 

The BMS is branded RadioB, and all that it says about the comm is “UART”.  The XiaoxiangBMS app shows it as JBD-SP04S034, so I’m sure it is a rebranded Jiabaida.  I now see that there are options from Jiabaida in both UART and UART RS485 so I can’t be sure of RS485, and don’t know how to tell.

So, I’m going to go back to trying Bluetooth with the nightly build and hope the Bluetooth install does not hang.  Would you agree with this approach?

￼

mr-manuel, 2023-11-13:
Yes, you could maybe try a USB to UART adapter instead of RS485.

beliffm, 2023-11-13:
Normally, those BMSs come in different flavours like UART, Heat, RS485 etc. They are listed on a label on front. If RS485 is not mentioned there..... The bluetooth interface is plugged into the UART connector.

mr-manuel, 2023-11-25:
@rldoose any updates on this?

rldoose, 2023-11-25:
So appreciate you checking.  My Starlink failed, and they have shipped me a new unit.  But I won’t have it until the 29th.  Then we will be taking a trip for ten days.  So, I won’t be able to resume this project until after Dec 10fh.  But, we will be sitting still after that, so I will hit it hard then.   ReaganSent from iPhone

rldoose, 2023-12-17:
I just got my Starlink back, so I’m ready to resume.  Will probably be able
to work on it tomorrow.  So appreciate your help and communication!   Above
and beyond.

Sent from iPhone

rldoose, 2023-12-18:
I just sat down to resume this project.  Since I mistakenly started out attempting to use a 485 converter on a UART BMS, I was planning on trying BT.  But I see the note that says “The Bluetooth and CAN connections are still not stable on some systems. If you want to have a stable connection use the serial connection.”  I also no longer see the instructions for BT in the documentation.  

I would rather be hard wired anyway, but I will have to order in the converter so I will be a few more days waiting on that.   But I’m a little confused.  I thought a UART was a piece of hardware that could be used for many different things/protocols, but I may be mistaken.  Is a UART to USB a one size fits all proposition, or do I need a specific "flavor”?  Any recommendations on what device to buy?

Reagan

mr-manuel, 2023-12-19:
I like the one from duppa.net, but currently it's sold out: https://www.duppa.net/shop/isolated-debug-board/

Maybe the one with four ports could also be interesting for you: https://www.duppa.net/shop/isolated-4-port-usb-to-ttl-uart-ftdi/

Albarge, 2023-12-22:
> Is a UART to USB a one size fits all proposition, or do I need a specific "flavor”? Any recommendations on what device to buy? Reagan
>

Maybe, but be careful on cheap non isolated adaptors picking up too much interference.

I can recommend the "Waveshare USB to RS232/485/TTL Converter Industrial Optically Isolated Converter with FT232RL Inside" https://amzn.eu/d/epY9Mym

My batteries are 10m from the Cerbo GX using a shielded USB cable, with powered USB HUB for the converters.

I have 3 of the Waveshare converters on 3 x JK BMS's with their RS485 converter (TTL to RS485, before I realised I didn't need it), but the Waveshare also accepts TTL directly or RS485, so for me, if the JK converter stops working I could just take the TTL directly from the BMS GPS port, but it will work for other BMS's with either, just make sure to use a twisted pair cable to it for RS485, It was absolutely necessary for me.

I originally bought a SDS Tech USB to console cable with CH340C chip, that claimed to be RS485 too- That didn't work though for RS485.

I then bought the cheap tiny USB to RS485 adaptors with A and B screws with the CH340E chip and the PL2303HX chip, found all over amazon and eBay - They _did_ work, but I had high dbus round trip times, and they'd drop out, so too unreliable, and not isolated.

So I then bit the bullet and got the Waveshare, and now they're solid with the dbus-serialbattery driver. Also they have their own isolated 3.3v or 5v power supply for the TTL, so no faffing around with a dc-dc converter from the BMS to power each converter like the duppa.net board needs.

rldoose, 2024-01-12:
Success!  What a trial this has been.  Finally received the proper converter only to have my RPi fail….  But, finally, with all the correct pieces and parts, it worked first crack out of the box!  Can’t thank you enough for sticking with me, and continuing to check in on me.  Now it is onto battery aggregation!

Reagan

closed 2024-01-13 by mr-manuel