Louisvdw/dbus-serialbattery #862: No powered Bluetooth adapters found

sarah1331sarah, 2023-11-16:
### Describe the problem

after an unspecified ammount of working time max 24 hr i get dissconnects and it refuses to reconnect unless i reboot raspberry pi 

also tried dev build 

### Driver version

v1.0.20230724beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.12

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4s

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 230.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 3.000
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;                - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;              As you have to adopt this value to your system, I reccomend to start with
;              OVP voltage - 0.030 (see Example).
;                - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;                  the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
;                - Try to decrease (lower) by 0.005 in steps, if the system hits the OVP too fast, before all
;                  cells could be balanced and the system goes into protection mode multiple times.
; Example: If OVP is 3.650, then start with 3.620 and increase/decrease by 0.005
; Note: The value has to be higher as the MAX_CELL_VOLTAGE
;SOC_RESET_VOLTAGE  = 3.650
; Specify after how many days the soc reset voltage should be reached again
; The timer is reset when the soc reset voltage is reached
[... 293 lines cut ...]
@4000000065567f9a27be4b24 INFO:SerialBattery:Starting dbus-serialbattery
@4000000065567f9a27cdaca4 INFO:SerialBattery:dbus-serialbattery v1.0.20231103dev
@4000000065567f9b0346c49c INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@4000000065567f9b0350a00c INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@4000000065567f9b1945cb94 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@4000000065567fa0040d6c14 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@4000000065567fa0042159a4 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
@4000000065567fa0042a23a4 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@4000000065567fa10c77b42c 
@4000000065567fa10c7bea4c INFO:Bluetooth details
@4000000065567fa10e600fb4 Attempting to disconnect from A4:C1:37:40:4E:F4
@4000000065567fa10e602724 Successful disconnected
@4000000065567fa6113065f4 Device A4:C1:37:40:4E:F4 (public)
@4000000065567fa61130814c 	Alias: 400AH_LiFePO4
@4000000065567fa61130891c 	Paired: no
@4000000065567fa6113090ec 	Trusted: no
@4000000065567fa6113098bc 	Blocked: no
@4000000065567fa611309ca4 	Connected: no
@4000000065567fa61130a474 	LegacyPairing: no
@4000000065567fa61130ac44 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@4000000065567fa6114ab06c 
@4000000065567fa71925a134 INFO:SerialBattery:
@4000000065567fa7192f0774 INFO:SerialBattery:Starting dbus-serialbattery
@4000000065567fa7193ee20c INFO:SerialBattery:dbus-serialbattery v1.0.20231103dev
```

sarah1331sarah, 2023-11-16:
`Last login: Thu Nov 16 21:15:31 2023 from 192.168.1.176
root@raspberrypi2:~# 
Display all 732 possibilities? (y or n)
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-blebattery.0/current | tai64nlocal
2023-11-16 21:13:43.490588500 
2023-11-16 21:13:43.490589500 During handling of the above exception, another exception occurred:
2023-11-16 21:13:43.490654500 
2023-11-16 21:13:43.490656500 Traceback (most recent call last):
2023-11-16 21:13:43.490658500   File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
2023-11-16 21:13:43.490662500     msg = self.format(record)
2023-11-16 21:13:43.490665500   File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
2023-11-16 21:13:43.490669500     return fmt.format(record)
2023-11-16 21:13:43.490671500   File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
2023-11-16 21:13:43.490698500     record.message = record.getMessage()
2023-11-16 21:13:43.490701500   File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
2023-11-16 21:13:43.490705500     msg = msg % self.args
2023-11-16 21:13:43.490708500 TypeError: not all arguments converted during string formatting
2023-11-16 21:13:43.490711500 Call stack:
2023-11-16 21:13:43.490713500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 163, in <module>
2023-11-16 21:13:43.490739500     main()
2023-11-16 21:13:43.490741500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 150, in main
2023-11-16 21:13:43.490746500     if not helper.setup_vedbus():
2023-11-16 21:13:43.490749500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 108, in setup_vedbus
2023-11-16 21:13:43.490753500     if not self.battery.get_settings():
2023-11-16 21:13:43.490756500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 261, in get_settings
2023-11-16 21:13:43.490782500     discharge_over_current = self.read_serial_data_llt(readCmd(REG_DSGOC))
2023-11-16 21:13:43.490833500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 171, in read_serial_data_llt
2023-11-16 21:13:43.490839500     logger.error(">>> ERROR: No reply - returning", e)
2023-11-16 21:13:43.490842500 Message: '>>> ERROR: No reply - returning'
2023-11-16 21:13:43.490845500 Arguments: (TypeError('can only concatenate str (not "int") to str'),)
2023-11-16 21:13:43.490850500 INFO:SerialBattery:publish config values = 1
2023-11-16 21:14:39.767620500 INFO:SerialBattery:BLE client disconnected
2023-11-16 21:15:42.763828500 --- Logging error ---
2023-11-16 21:15:42.763834500 Traceback (most recent call last):
2023-11-16 21:15:42.763837500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 152, in async_read_serial_data_llt
2023-11-16 21:15:42.763843500     result = await asyncio.wait_for(asyncio.wrap_future(bt_task), 20)
2023-11-16 21:15:42.763847500   File "/usr/lib/python3.8/asyncio/tasks.py", line 494, in wait_for
2023-11-16 21:15:42.763851500     return fut.result()
2023-11-16 21:15:42.763853500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 138, in send_command
2023-11-16 21:15:42.763980500     await self.bt_client.start_notify(BLE_CHARACTERISTICS_RX_UUID, rx_collector)
2023-11-16 21:15:42.763985500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 834, in start_notify
2023-11-16 21:15:42.763990500     await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
2023-11-16 21:15:42.763995500   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 950, in start_notify
2023-11-16 21:15:42.764025500     reply = await self._bus.call(
2023-11-16 21:15:42.764028500   File "/usr/lib/python3.8/site-packages/dbus_fast/aio/message_bus.py", line 399, in call
2023-11-16 21:15:42.764033500     await future
2023-11-16 21:15:42.764035500   File "/usr/lib/python3.8/site-packages/dbus_fast/aio/message_reader.py", line 22, in _message_reader
2023-11-16 21:15:42.764041500     message = unmarshaller._unmarshall()
2023-11-16 21:15:42.764154500   File "/usr/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 777, in _unmarshall
2023-11-16 21:15:42.764160500     self._read_header()
2023-11-16 21:15:42.764162500   File "/usr/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 638, in _read_header
2023-11-16 21:15:42.764168500     self._read_to_pos(HEADER_SIGNATURE_SIZE)
2023-11-16 21:15:42.764171500   File "/usr/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 374, in _read_to_pos
2023-11-16 21:15:42.764200500     self._read_sock_with_fds(pos, missing_bytes)
2023-11-16 21:15:42.764204500   File "/usr/lib/python3.8/site-packages/dbus_fast/_private/unmarshaller.py", line 317, in _read_sock_with_fds
2023-11-16 21:15:42.764210500     raise EOFError()
2023-11-16 21:15:42.764212500 EOFError
2023-11-16 21:15:42.764213500 
2023-11-16 21:15:42.764214500 During handling of the above exception, another exception occurred:
2023-11-16 21:15:42.764240500 
2023-11-16 21:15:42.764241500 Traceback (most recent call last):
2023-11-16 21:15:42.764244500   File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
2023-11-16 21:15:42.764248500     msg = self.format(record)
2023-11-16 21:15:42.764250500   File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
2023-11-16 21:15:42.764255500     return fmt.format(record)
[... cut ...]
2023-11-16 21:16:46.758910500     gen_data = self.read_serial_data_llt(self.command_general)
2023-11-16 21:16:46.758981500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 168, in read_serial_data_llt
2023-11-16 21:16:46.758989500     logger.error(">>> ERROR: No reply - canceled - returning", e)
2023-11-16 21:16:46.758994500 Message: '>>> ERROR: No reply - canceled - returning'
2023-11-16 21:16:46.758997500 Arguments: (CancelledError(),)
2023-11-16 21:16:46.759000500 INFO:SerialBattery:BLE client disconnected
        
`
Just happened again

sarah1331sarah, 2023-11-17:
crashed again last night
any help apreaciated 
`@400000006556a3032aab6c7c     data = asyncio.run(self.async_read_serial_data_llt(command))
@400000006556a3032c699b24   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
@400000006556a3032c69b294     return loop.run_until_complete(main)
@400000006556a3032c69be4c   File "/usr/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
@400000006556a3032c69d1d4     self.run_forever()
@400000006556a3032c69dd8c   File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
@400000006556a3032c69ed2c     self._run_once()
@400000006556a3032c6b7f84   File "/usr/lib/python3.8/asyncio/base_events.py", line 1859, in _run_once
@400000006556a3032c6b930c     handle._run()
@400000006556a3032c6b9adc   File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
@400000006556a3032c6baa7c     self._context.run(self._callback, *self._args)
@400000006556a3032c6bba1c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 158, in async_read_serial_data_llt
@400000006556a3032c6bd18c     logger.error(">>> ERROR: No reply - returning", e)
@400000006556a3032c6ce6e4 Message: '>>> ERROR: No reply - returning'
@400000006556a3032c6cf684 Arguments: (BleakDBusError('org.bluez.Error.InProgress', 'In Progress'),)
@400000006556a3032c6d0624 
@400000006556a3032c6d0a0c #0
@400000006556a3032c6d0df4 #1
@400000006556a3032c6d15c4 #2
@400000006556a3032c6d19ac #3
@400000006556ae4134696584 Exception in thread LltJbd_Ble_Loop:
@400000006556ae41346988ac Traceback (most recent call last):
@400000006556ae4134699464   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
@400000006556ae413469a404     self.run()
@400000006556ae413469abd4   File "/usr/lib/python3.8/threading.py", line 870, in run
@400000006556ae413469b78c     self._target(*self._args, **self._kwargs)
@400000006556ae413469c72c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 80, in background_loop
@400000006556ae41346bf9ac     asyncio.run(self.bt_main_loop())
@400000006556ae41346c0564   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
@400000006556ae41346c1504     return loop.run_until_complete(main)
@400000006556ae41346c20bc   File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
@400000006556ae41346c3444     return future.result()
@400000006556ae41346c3c14   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 75, in bt_main_loop
@400000006556ae41346cb144     await asyncio.sleep(0.1)
@400000006556ae41346cb914   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 569, in __aexit__
@400000006556ae41346ccc9c     await self.disconnect()
@400000006556ae41346cd46c   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 614, in disconnect
@400000006556ae41346ce7f4     return await self._backend.disconnect()
@400000006556ae41346d499c   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 412, in disconnect
@400000006556ae41346e245c     assert_reply(reply)
@400000006556ae41346e2c2c   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
@400000006556ae41346e439c     raise BleakDBusError(reply.error_name, reply.body)
@400000006556ae41346e4f54 bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.NoReply] Message recipient disconnected from message bus without replying`

mr-manuel, 2023-11-17:
Did you try the `nightly` of the `dev` branch? If not please do so.

sarah1331sarah, 2023-11-17:
Yes I did it lasts a lot longer than all the other installs but refuses to last a full day 

I was using bradcagle Bluetooth bms perfect fine but wanted the “extras” yours included 

I also started a brand new pi image and started from scratch to exclude “conflicts”

mr-manuel, 2023-11-17:
Maybe this solves the issue also for you?

https://github.com/Louisvdw/dbus-serialbattery/issues/864#issuecomment-1816973598

sarah1331sarah, 2023-11-19:
testing will report thank you

sarah1331sarah, 2023-11-19:
`@400000006559d49b35e31cd4 
@400000006559d49b35e33444 INFO:Bluetooth details
@400000006559d49b380ad31c Device a4:c1:37:40:4e:f4 not available
@400000006559d4a03b231f3c Device a4:c1:37:40:4e:f4 not available
@400000006559d4a03b45e97c 
@400000006559d4a22bf9aae4 INFO:SerialBattery:
@400000006559d4a22c180854 INFO:SerialBattery:Starting dbus-serialbattery
@400000006559d4a22c28f45c INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006559d4a3085faf34 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4a30866a08c INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4a32f8830a4 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006559d4a809296004 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006559d4a8093ee7bc ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
@400000006559d4a8094a2e74 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006559d4a923422174 
@400000006559d4a92344195c INFO:Bluetooth details
@400000006559d4a924ed50d4 Device a4:c1:37:40:4e:f4 not available
@400000006559d4ae27ecd91c Device a4:c1:37:40:4e:f4 not available
@400000006559d4ae27ed002c 
@400000006559d4af311e3ddc INFO:SerialBattery:
@400000006559d4af3124c1d4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006559d4af3132fe5c INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006559d4b00c56320c INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4b00c5d13c4 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4b022858c1c ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006559d4b50d0a0534 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006559d4b50d385c7c ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
@400000006559d4b50d40a1ac ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006559d4b616a222d4 
@400000006559d4b616a48c04 INFO:Bluetooth details
@400000006559d4b61873aa9c Device a4:c1:37:40:4e:f4 not available
@400000006559d4bb1ad8d35c Device a4:c1:37:40:4e:f4 not available
@400000006559d4bb1af878ec 
@400000006559d4bc22e02f3c INFO:SerialBattery:
@400000006559d4bc22e046ac INFO:SerialBattery:Starting dbus-serialbattery
@400000006559d4bc22f0928c INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006559d4bc39837ce4 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4bc398a3f5c INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4bd1432670c ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006559d4c13a32a4bc ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006559d4c13a55e044 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
@400000006559d4c13a5c8f34 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006559d4c30802413c 
@400000006559d4c30803eb04 INFO:Bluetooth details
@400000006559d4c3099a36bc Device a4:c1:37:40:4e:f4 not available
@400000006559d4c80c357b0c Device a4:c1:37:40:4e:f4 not available
@400000006559d4c80c4676b4 
@400000006559d4c9159eb0dc INFO:SerialBattery:
@400000006559d4c915aa499c INFO:SerialBattery:Starting dbus-serialbattery
@400000006559d4c915be71c4 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006559d4c92d0830cc INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4c92d0f3d7c INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4ca07ad4b3c ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006559d4ce2db919dc ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006559d4ce2ddc70bc ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
@400000006559d4ce2de5afec ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006559d4cf35be8d74 
@400000006559d4cf35c03354 INFO:Bluetooth details
@400000006559d4cf374f2824 Device a4:c1:37:40:4e:f4 not available
@400000006559d4d439bd2a84 Device a4:c1:37:40:4e:f4 not available
@400000006559d4d439cd6e94 
@400000006559d4d6054357ec INFO:SerialBattery:
@400000006559d4d60549c474 INFO:SerialBattery:Starting dbus-serialbattery
@400000006559d4d605587244 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006559d4d61b9c8cd4 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4d61ba3571c INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
@400000006559d4d6311e4994 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006559d4db1c4b3f7c ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006559d4db1c6edcac ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
[... cut ...] 7
@400000006559d7210e4509bc INFO:SerialBattery:publish config values = 1
@400000006559df7430903294 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006559df8032880324 INFO:SerialBattery:BLE client disconnected
@400000006559df8037635c04 ERROR:SerialBattery:>>> ERROR: No reply - canceled - returning
@400000006559df80376b5ecc ERROR:SerialBattery:
@400000006559dfb0052f146c #0
@400000006559dfb0052f27f4 #1
@400000006559dfb0052f2fc4 #2
@400000006559dfb0052f33ac #3`

loss connection after 1 hr

mr-manuel, 2023-11-19:
https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#driver-log-files

[image]

Please use trippe backticks ``` in the line before and after your code/logs.

mr-manuel, 2023-11-19:
Which Raspberry Pi do you have? Maybe version 2? Try to turn off Bluetooth and use an externa USB to Bluetooth adapter.

sarah1331sarah, 2023-11-19:
thanks again i have 3 b+ can you identify the error from the logs ? ill try tal64nlocal

mr-manuel, 2023-11-19:
Not without timestamps. It could also be a hardware issue.

sarah1331sarah, 2023-11-23:
started again this is errors now on dissconnect ```login as: root
root@192.168.1.200's password:
Last login: Thu Nov 23 12:02:57 2023 from 192.168.1.148
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal
2023-11-23 12:20:01.620128500
2023-11-23 12:20:02.729579500 ERROR:SerialBattery:>>> ERROR: Maximum value of CELL_VOLTAGES_WHILE_CHARGING is set to a value lower than MAX_CELL_VOLTAGE. Please check the configuration.
2023-11-23 12:20:02.840355500 INFO:SerialBattery:
2023-11-23 12:20:02.840761500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:20:02.841743500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:20:03.249688500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:03.250170500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:03.609367500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 12:20:08.261616500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 12:20:08.264242500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 12:20:08.264700500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 12:20:09.397421500
2023-11-23 12:20:09.397528500 INFO:Bluetooth details
2023-11-23 12:20:09.432298500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 12:20:09.432306500 Successful disconnected
2023-11-23 12:20:14.477169500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 12:20:14.477175500   Alias: 400AH_LiFePO4
2023-11-23 12:20:14.477177500   Paired: no
2023-11-23 12:20:14.477181500   Trusted: no
2023-11-23 12:20:14.477183500   Blocked: no
2023-11-23 12:20:14.477185500   Connected: no
2023-11-23 12:20:14.477186500   LegacyPairing: no
2023-11-23 12:20:14.477189500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 12:20:14.478076500
2023-11-23 12:20:15.477343500 ERROR:SerialBattery:>>> ERROR: Maximum value of CELL_VOLTAGES_WHILE_CHARGING is set to a value lower than MAX_CELL_VOLTAGE. Please check the configuration.
2023-11-23 12:20:15.582616500 INFO:SerialBattery:
2023-11-23 12:20:15.583022500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:20:15.583995500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:20:15.956822500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:15.957261500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:16.332534500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 12:20:20.969669500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 12:20:20.972062500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 12:20:20.972494500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 12:20:22.100204500
2023-11-23 12:20:22.100312500 INFO:Bluetooth details
2023-11-23 12:20:22.129702500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 12:20:22.129709500 Successful disconnected
2023-11-23 12:20:27.175084500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 12:20:27.175091500   Alias: 400AH_LiFePO4
2023-11-23 12:20:27.175093500   Paired: no
2023-11-23 12:20:27.175095500   Trusted: no
2023-11-23 12:20:27.175097500   Blocked: no
2023-11-23 12:20:27.175098500   Connected: no
2023-11-23 12:20:27.175100500   LegacyPairing: no
2023-11-23 12:20:27.175102500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 12:20:27.175980500
2023-11-23 12:20:28.244159500 ERROR:SerialBattery:>>> ERROR: Maximum value of CELL_VOLTAGES_WHILE_CHARGING is set to a value lower than MAX_CELL_VOLTAGE. Please check the configuration.
2023-11-23 12:20:28.355766500 INFO:SerialBattery:
2023-11-23 12:20:28.356481500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:20:28.358058500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:20:28.821957500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:28.822576500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:29.183698500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 12:20:33.835012500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 12:20:33.838339500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 12:20:33.849393500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 12:20:35.001389500
2023-11-23 12:20:35.001498500 INFO:Bluetooth details
2023-11-23 12:20:35.034209500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 12:20:35.034216500 Successful disconnected
2023-11-23 12:20:40.078059500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 12:20:40.078065500   Alias: 400AH_LiFePO4
2023-11-23 12:20:40.078067500   Paired: no
2023-11-23 12:20:40.078069500   Trusted: no
2023-11-23 12:20:40.078071500   Blocked: no
[... 10 lines cut ...]
[... cut ...]
2023-11-23 12:20:53.926680500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:20:54.331692500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:54.332159500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:20:54.687481500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
```

sarah1331sarah, 2023-11-23:
i get this error before it disconnects 
```
login as: root
root@192.168.1.116's password:
Last login: Thu Nov 23 12:51:32 2023
root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal
2023-11-23 12:48:21.916246500
2023-11-23 12:48:21.916250500 INFO:Bluetooth details
2023-11-23 12:48:21.946840500 Device 70:3e:97:d2:e6:10 not available
2023-11-23 12:48:27.003527500 Device 70:3E:97:D2:E6:10 (public)
2023-11-23 12:48:27.003531500   Alias: 400AH_LiFePO4
2023-11-23 12:48:27.003534500   Paired: no
2023-11-23 12:48:27.003536500   Trusted: no
2023-11-23 12:48:27.003539500   Blocked: no
2023-11-23 12:48:27.003541500   Connected: no
2023-11-23 12:48:27.003543500   LegacyPairing: no
2023-11-23 12:48:27.003545500   RSSI: -57
2023-11-23 12:48:27.004530500
2023-11-23 12:48:28.438230500 ERROR:SerialBattery:>>> ERROR: SOC_RESET_VOLTAGE is set to a value less than MAX_CELL_VOLTAGE. Please check the configuration.
2023-11-23 12:48:28.441646500 ERROR:SerialBattery:>>> ERROR: Maximum value of CELL_VOLTAGES_WHILE_CHARGING is set to a value lower than MAX_CELL_VOLTAGE. Please check the configuration.
2023-11-23 12:48:28.785483500 INFO:SerialBattery:
2023-11-23 12:48:28.785755500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:48:28.786345500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:48:29.050415500 INFO:SerialBattery:Init of LltJbd_Ble at 70:3e:97:d2:e6:10
2023-11-23 12:48:29.050859500 INFO:SerialBattery:Test of LltJbd_Ble at 70:3e:97:d2:e6:10
2023-11-23 12:48:35.534464500 INFO:SerialBattery:Connection established to LltJbd_Ble
2023-11-23 12:48:35.534962500 INFO:SerialBattery:Battery LltJbd_Ble connected to dbus from ble703e97d2e610
2023-11-23 12:48:35.535340500 INFO:SerialBattery:========== Settings ==========
2023-11-23 12:48:35.535746500 INFO:SerialBattery:> Connection voltage: 16.47V | Current: 0.0A | SoC: 98.73%
2023-11-23 12:48:35.536116500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2023-11-23 12:48:35.536491500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-11-23 12:48:35.536872500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 230.0A
2023-11-23 12:48:35.537257500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 130.0A | MAX BATTERY DISCHARGE CURRENT: 130.0A (read from BMS)
2023-11-23 12:48:35.537621500 INFO:SerialBattery:> CVCM:     True
2023-11-23 12:48:35.538011500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.2V | MAX CELL VOLTAGE: 4.25V
2023-11-23 12:48:35.538388500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2023-11-23 12:48:35.538763500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-11-23 12:48:35.539136500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-11-23 12:48:35.539643500 INFO:SerialBattery:Serial Number/Unique Identifier: 42_5.5Ah
2023-11-23 12:48:35.563826500 INFO:SerialBattery:DeviceInstance = 1
2023-11-23 12:48:35.564284500 INFO:SerialBattery:com.victronenergy.battery.ble703e97d2e610
2023-11-23 12:48:36.371639500 INFO:SerialBattery:publish config values = 1
2023-11-23 12:50:19.304379500 *** CCGX booted (0) ***
2023-11-23 12:51:36.232305500
2023-11-23 12:51:36.232309500 INFO:Bluetooth details
2023-11-23 12:51:36.268012500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 12:51:36.268017500 Successful disconnected
2023-11-23 12:51:41.323843500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 12:51:41.323848500   Alias: 400AH_LiFePO4
2023-11-23 12:51:41.323851500   Paired: no
2023-11-23 12:51:41.323853500   Trusted: no
2023-11-23 12:51:41.323855500   Blocked: no
2023-11-23 12:51:41.323857500   Connected: no
2023-11-23 12:51:41.323860500   LegacyPairing: no
2023-11-23 12:51:41.323862500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 12:51:41.323865500   RSSI: -38
2023-11-23 12:51:41.324694500
2023-11-23 12:51:42.348910500 INFO:SerialBattery:
2023-11-23 12:51:42.349182500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:51:42.349772500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:51:42.607572500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:51:42.607858500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 12:51:48.147233500 ERROR:SerialBattery:Exception occurred: TypeError('can only concatenate str (not "int") to str') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #237
2023-11-23 12:51:48.327145500 ERROR:SerialBattery:Exception occurred: TypeError('can only concatenate str (not "int") to str') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #237
2023-11-23 12:51:48.536979500 ERROR:SerialBattery:Exception occurred: TypeError('can only concatenate str (not "int") to str') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #237
2023-11-23 12:51:48.747125500 ERROR:SerialBattery:Exception occurred: TypeError('can only concatenate str (not "int") to str') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #237
2023-11-23 12:51:48.927195500 ERROR:SerialBattery:Exception occurred: TypeError('can only concatenate str (not "int") to str') of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #237
2023-11-23 12:51:49.347787500 INFO:SerialBattery:Connection established to LltJbd_Ble
[... cut ...]
2023-11-23 12:51:41.323860500   LegacyPairing: no
2023-11-23 12:51:41.323862500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 12:51:41.323865500   RSSI: -38
2023-11-23 12:51:41.324694500
2023-11-23 12:51:42.348910500 INFO:SerialBattery:
2023-11-23 12:51:42.349182500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 12:51:42.349772500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 12:51:42.607572500 INFO:SerialBattery:Init of LltJbd_
```

sarah1331sarah, 2023-11-23:
sorry for tripple post i get the errors on both rpi 3 b + and rpi 4 b 1.5 
i have also tested the bluetooth on the BMS same errors before (line 237) and after dissconnect (line 59)

i can confirm this doesnt happen on another JBD Bms i have but its only 50a and need the 150 it currently errors with

sarah1331sarah, 2023-11-23:
so it failed again and after an hour i had chance to run the tal64 once i ran the command it started scanning again 
 ```login as: root
root@192.168.1.200's password:
Last login: Thu Nov 23 14:53:15 2023
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal
2023-11-23 17:47:08.729064500   Trusted: no
2023-11-23 17:47:08.729066500   Blocked: no
2023-11-23 17:47:08.729068500   Connected: no
2023-11-23 17:47:08.729069500   LegacyPairing: no
2023-11-23 17:47:08.729071500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 17:47:08.730864500
2023-11-23 17:47:09.873756500 INFO:SerialBattery:
2023-11-23 17:47:09.874420500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 17:47:09.875514500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 17:47:10.264878500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:10.265536500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:10.641097500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 17:47:15.279965500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 17:47:15.279973500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 17:47:15.279977500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 17:47:16.432530500
2023-11-23 17:47:16.432817500 INFO:Bluetooth details
2023-11-23 17:47:16.468682500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 17:47:16.468689500 Successful disconnected
2023-11-23 17:47:21.517936500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 17:47:21.517944500   Alias: 400AH_LiFePO4
2023-11-23 17:47:21.517947500   Paired: no
2023-11-23 17:47:21.517948500   Trusted: no
2023-11-23 17:47:21.517950500   Blocked: no
2023-11-23 17:47:21.517952500   Connected: no
2023-11-23 17:47:21.517953500   LegacyPairing: no
2023-11-23 17:47:21.517955500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 17:47:21.520335500
2023-11-23 17:47:22.627079500 INFO:SerialBattery:
2023-11-23 17:47:22.627793500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 17:47:22.628867500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 17:47:23.001319500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:23.001958500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:23.376217500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 17:47:28.016423500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 17:47:28.016430500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 17:47:28.016434500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 17:47:29.144264500
2023-11-23 17:47:29.144526500 INFO:Bluetooth details
2023-11-23 17:47:29.178155500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 17:47:29.178162500 Successful disconnected
2023-11-23 17:47:34.230343500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 17:47:34.230350500   Alias: 400AH_LiFePO4
2023-11-23 17:47:34.230352500   Paired: no
2023-11-23 17:47:34.230354500   Trusted: no
2023-11-23 17:47:34.230355500   Blocked: no
2023-11-23 17:47:34.230357500   Connected: no
2023-11-23 17:47:34.230359500   LegacyPairing: no
2023-11-23 17:47:34.230361500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-11-23 17:47:34.231737500
2023-11-23 17:47:35.375211500 INFO:SerialBattery:
2023-11-23 17:47:35.375849500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-23 17:47:35.376869500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-23 17:47:35.752013500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:35.752650500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:40:4e:f4
2023-11-23 17:47:36.113503500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 17:47:40.764722500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 17:47:40.767306500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 17:47:40.768075500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 17:47:41.883424500
2023-11-23 17:47:41.883690500 INFO:Bluetooth details
2023-11-23 17:47:41.914113500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 17:47:41.914120500 Successful disconnected
2023-11-23 17:47:46.961536500 Device A4:C1:37:40:4E:F4 (public)
2023-11-23 17:47:46.961542500   Alias: 400AH_LiFePO4
[... 14 lines cut ...]
2023-11-23 17:47:53.499253500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 17:47:53.499257500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-11-23 17:47:54.665902500
2023-11-23 17:47:54.666166500 INFO:Bluetooth details
2023-11-23 17:47:54.699161500 Attempting to disconnect from A4:C1:37:40:4E:F4
2023-11-23 17:47:54.699169500 Successful disconnected
[... cut ...]
2023-11-23 17:48:01.793735500 ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-11-23 17:48:06.439032500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-11-23 17:48:06.439039500 ERROR:SerialBattery:No BMS found at a4:c1:37:40:4e:f4
2023-11-23 17:48:06.439042500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
```

baranator, 2023-11-23:
I had similar problems while 

> Which Raspberry Pi do you have? Maybe version 2? Try to turn off Bluetooth and use an externa USB to Bluetooth adapter.

Also [see here](https://github.com/Louisvdw/dbus-serialbattery/pull/372#issuecomment-1368039669), we had problems with the internal BT crashing every 15 hours.

sarah1331sarah, 2023-11-23:
Think I have sorted it I had re-named the device to 400ah lifepo4 reinstating the original name I am not getting errors will test for a couple days and update

mr-manuel, 2023-11-25:
You mean you re-named something on the BMS itself or in Venus OS under Device -> Name?

sarah1331sarah, 2023-11-26:
The bms itself but its failed again I just reset will post another log 

I tried 
pip3 install bleak==0.20.2
And again with  pip3 install dbus-fast==1.87.0
As I was getting class 'bleak.exc.BleakDBusError' will post log on next error again

sarah1331sarah, 2023-12-08:
just installed a new BMS can you please shed light on this with the dev branch 

for referance dev was the only one working before (with dissconnects)

```
2023-12-08 15:43:47.261987500 [CHG] Controller D8:3A:DD:6A:4E:52 Discovering: no
2023-12-08 15:43:47.261991500 Successful disconnected
2023-12-08 15:43:52.351933500 Device A4:C1:37:40:4E:F4 (public)
2023-12-08 15:43:52.351938500   Alias: 400AH_LiFePO4
2023-12-08 15:43:52.351941500   Paired: no
2023-12-08 15:43:52.351943500   Trusted: no
2023-12-08 15:43:52.351946500   Blocked: no
2023-12-08 15:43:52.351948500   Connected: no
2023-12-08 15:43:52.351950500   LegacyPairing: no
2023-12-08 15:43:52.351952500   UUID: Device Information        (0000180a-0000-1                                                                                                                                                             000-8000-00805f9b34fb)
2023-12-08 15:43:52.351956500   RSSI: -40
2023-12-08 15:43:52.351958500 [CHG] Device 5E:90:A7:75:16:5E RSSI: -93
2023-12-08 15:43:52.352048500 [CHG] Device 5E:90:A7:75:16:5E ManufacturerData Ke                                                                                                                                                             y: 0x004c
2023-12-08 15:43:52.352052500 [CHG] Device 5E:90:A7:75:16:5E ManufacturerData Va                                                                                                                                                             lue:
2023-12-08 15:43:52.353095500
2023-12-08 15:43:53.026732500 INFO:SerialBattery:
2023-12-08 15:43:53.026996500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-08 15:43:53.027575500 INFO:SerialBattery:dbus-serialbattery v1.0.2023112                                                                                                                                                             8dev
2023-12-08 15:43:53.255112500 INFO:SerialBattery:Init of LltJbd_Ble at a4:c1:37:                                                                                                                                                             40:4e:f4
2023-12-08 15:43:53.255391500 INFO:SerialBattery:Test of LltJbd_Ble at a4:c1:37:                                                                                                                                                             40:4e:f4
2023-12-08 15:43:59.965972500 INFO:SerialBattery:Connection established to LltJb                                                                                                                                                             d_Ble
2023-12-08 15:43:59.966296500 INFO:SerialBattery:Battery LltJbd_Ble connected to                                                                                                                                                              dbus from blea4c137404ef4
2023-12-08 15:43:59.966562500 INFO:SerialBattery:========== Settings ==========
2023-12-08 15:43:59.966877500 INFO:SerialBattery:> Connection voltage: 13.32V |                                                                                                                                                              Current: -0.64A | SoC: 99.47%
2023-12-08 15:43:59.967148500 INFO:SerialBattery:> Cell count: 4 | Cells populat                                                                                                                                                             ed: 4
2023-12-08 15:43:59.967422500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: Tru                                                                                                                                                             e
2023-12-08 15:43:59.967702500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 1                                                                                                                                                             00.0A | MAX BATTERY DISCHARGE CURRENT: 230.0A
2023-12-08 15:43:59.967977500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 2                                                                                                                                                             00.0A | MAX BATTERY DISCHARGE CURRENT: 240.0A (read from BMS)
2023-12-08 15:43:59.968238500 INFO:SerialBattery:> CVCM:     True
2023-12-08 15:43:59.968508500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX                                                                                                                                                              CELL VOLTAGE: 3.6V
2023-12-08 15:43:59.968778500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  T                                                                                                                                                             rue
2023-12-08 15:43:59.969046500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   T                                                                                                                                                             rue
2023-12-08 15:43:59.969315500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: T                                                                                                                                                             rue
[... cut ...]
2023-12-08 15:44:37.314059500   Paired: no
2023-12-08 15:44:37.314061500   Trusted: no
2023-12-08 15:44:37.314063500   Blocked: no
2023-12-08 15:44:37.314065500   Connected: no
2023-12-08 15:44:37.314068500   LegacyPairing: no
2023-12-08 15:44:37.314070500   UUID: Device Information        (0000180a-0000-1                                                                                                                                                             000-8000-00805f9b34fb)
2023-12-08 15:44:37.314073500   RSSI: -38
```

closed 2023-12-09 by sarah1331sarah