Louisvdw/dbus-serialbattery #864: Driver not work

DaweeCZ, 2023-11-17:
### Describe the problem

During the initial installation on the beta version of Venus OS 3.20 ~ 16, everything went fine, except that the BMSs kept connecting and disconnecting via Bluetooth, so I uninstalled the installation according to the instructions. Now, I wanted to reinstall the driver again, and since then, it hasn't been working. During the installation, it reports an error with a non-existent folder /service/gui/, and in the BT log, it throws 'ERROR:SerialBattery:Unexpected err=ModuleNotFoundError("No module named 'dbus_fast._private.constants'"), type(err)=<class 'ModuleNotFoundError'>.

### Driver version

latest

### Venus OS device type

Cerbo GX

### Venus OS version

3.20 ~ 26

### BMS type

JKBMS / Heltec BMS

### Cell count

15

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.def>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BLUETOOTH_BMS = Jkbms_Ble C8:47:80:01:92:CE
```

### Relevant log output

```shell
@400000006557892532a723e4 Missing device address argument
@400000006557892719630bdc INFO:SerialBattery:Starting dbus-serialbattery
@40000000655789271a74c22c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@400000006557892804c68e9c INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892805746e54 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892814b070ac ERROR:SerialBattery:Unexpected err=ModuleNotFoundError("No module named 'dbus_fast.>
@400000006557892814bdc2d4 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@400000006557892814c2aca4
@40000000655789282c8d72ec Missing device address argument
@400000006557892a1276f39c INFO:SerialBattery:Starting dbus-serialbattery
@400000006557892a129d9e0c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@400000006557892a358b736c INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892a359677bc INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892b02b8a5cc ERROR:SerialBattery:Unexpected err=ModuleNotFoundError("No module named 'dbus_fast.>
@400000006557892b02c3d12c ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@400000006557892b02c8268c
@400000006557892b15004a14 Missing device address argument
@400000006557892d04e65f24 INFO:SerialBattery:Starting dbus-serialbattery
@400000006557892d04fb5654 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@400000006557892e0771f7bc INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892e077ec8fc INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:01:92:CE
@400000006557892e107e8e9c ERROR:SerialBattery:Unexpected err=ModuleNotFoundError("No module named 'dbus_fast.>
```

mr-manuel, 2023-11-17:
[image]

But at least we can see it in the logs... Please install the latest `nightly` from the `dev` branch. There the error is fixed. Else see https://github.com/Louisvdw/dbus-serialbattery/issues/785

DaweeCZ, 2023-11-17:
Oh, I apologize, I overlooked this.. I installed the latest nightly dev, and now the log looks like this:

==> /data/log/dbus-blebattery.0/current <==
2023-11-17 17:35:53.043652500 ERROR:SerialBattery:No BMS found at C8:47:80:01:92:CE
2023-11-17 17:35:53.043662500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-11-17 17:35:53.431146500
2023-11-17 17:35:53.431569500 INFO:Bluetooth details
2023-11-17 17:35:53.552588500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552598500 [CHG] Device C8:47:80:01:92:CE RSSI: -62
2023-11-17 17:35:53.552601500 [CHG] Device C8:47:80:01:92:CE RSSI: -70
2023-11-17 17:35:53.552604500 [CHG] Device C8:47:80:01:92:CE RSSI: -61
2023-11-17 17:35:53.552606500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552609500 [CHG] Device C8:47:80:01:92:CE RSSI: -63
2023-11-17 17:35:53.552612500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552776500 [CHG] Device C8:47:80:01:92:CE RSSI: -63
2023-11-17 17:35:53.552779500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552782500 [CHG] Device C8:47:80:01:92:CE RSSI: -63
2023-11-17 17:35:53.552785500 [CHG] Device C8:47:80:01:92:CE RSSI: -72
2023-11-17 17:35:53.552787500 [CHG] Device C8:47:80:01:92:CE RSSI: -63
2023-11-17 17:35:53.552790500 [CHG] Device C8:47:80:01:92:CE RSSI: -62
2023-11-17 17:35:53.552819500 [CHG] Device C8:47:80:01:92:CE RSSI: -70
2023-11-17 17:35:53.552822500 [CHG] Device C8:47:80:01:92:CE RSSI: -62
2023-11-17 17:35:53.552825500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552828500 [CHG] Device C8:47:80:01:92:CE RSSI: -62
2023-11-17 17:35:53.552830500 [CHG] Device C8:47:80:01:92:CE RSSI: -73
2023-11-17 17:35:53.552833500 [CHG] Device C8:47:80:01:92:CE RSSI: -63
2023-11-17 17:35:53.552961500 [CHG] Device C8:47:80:01:92:CE RSSI: -71
2023-11-17 17:35:53.552964500 [CHG] Device C8:47:80:01:92:CE RSSI: -61
2023-11-17 17:35:53.706221500 Attempting to disconnect from C8:47:80:01:92:CE
2023-11-17 17:35:53.706230500 [CHG] Device B0:D2:78:4E:98:E7 RSSI is nil
2023-11-17 17:35:53.706233500 [CHG] Device D7:38:33:47:F9:6D RSSI is nil
2023-11-17 17:35:53.706236500 [CHG] Device D4:9D:C0:55:D7:F1 RSSI is nil
2023-11-17 17:35:53.706238500 [CHG] Device C8:47:80:01:92:6B RSSI is nil
2023-11-17 17:35:53.706241500 [CHG] Device C8:47:80:01:92:CE RSSI is nil
2023-11-17 17:35:53.706244500 [CHG] Controller 00:1A:7D:DA:71:15 Discovering: no
2023-11-17 17:35:53.706405500 Successful disconnected
2023-11-17 17:35:58.798148500 Device C8:47:80:01:92:CE (public)
2023-11-17 17:35:58.798157500   Alias: Battery 1
2023-11-17 17:35:58.798159500   Paired: no
2023-11-17 17:35:58.798160500   Trusted: no
2023-11-17 17:35:58.798161500   Blocked: no
2023-11-17 17:35:58.798163500   Connected: no
2023-11-17 17:35:58.798164500   LegacyPairing: no
2023-11-17 17:35:58.798166500   RSSI: -62
2023-11-17 17:35:58.799077500
2023-11-17 17:36:01.050505500 INFO:SerialBattery:
2023-11-17 17:36:01.051716500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-17 17:36:01.053745500 INFO:SerialBattery:dbus-serialbattery v1.0.20231103dev
2023-11-17 17:36:02.237230500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:01:92:CE
2023-11-17 17:36:02.238195500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:01:92:CE
2023-11-17 17:36:02.459013500 Exception in thread Thread-1:
2023-11-17 17:36:02.459022500 Traceback (most recent call last):
2023-11-17 17:36:02.459024500   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2023-11-17 17:36:02.459028500     self.run()
2023-11-17 17:36:02.459029500   File "/usr/lib/python3.8/threading.py", line 870, in run
2023-11-17 17:36:02.459032500     self._target(*self._args, **self._kwargs)
2023-11-17 17:36:02.459034500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 433, in connect_and_scrape
2023-11-17 17:36:02.459209500     asyncio.run(self.asy_connect_and_scrape())
2023-11-17 17:36:02.459214500   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
2023-11-17 17:36:02.459217500     return loop.run_until_complete(main)
2023-11-17 17:36:02.459219500   File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
2023-11-17 17:36:02.459223500     return future.result()
2023-11-17 17:36:02.459255500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 443, in asy_connect_and_scrape
2023-11-17 17:36:02.459260500     client = BleakClient(self.address)
2023-11-17 17:36:02.459262500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 518, in __init__
2023-11-17 17:36:02.459265500     get_platform_client_backend_type() if backend is None else backend
2023-11-17 17:36:02.459269500   File "/usr/lib/python3.8/site-packages/bleak/backends/client.py", line 260, in get_platform_client_backend_type
2023-11-17 17:36:02.459327500     from bleak.backends.bluezdbus.client import BleakClientBlueZDBus
2023-11-17 17:36:02.459331500   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 23, in <module>
2023-11-17 17:36:02.459335500     from dbus_fast.aio import MessageBus
2023-11-17 17:36:02.459337500   File "/usr/lib/python3.8/site-packages/dbus_fast/__init__.py", line 1, in <module>
2023-11-17 17:36:02.459380500     from . import introspection, message_bus, proxy_object, service
2023-11-17 17:36:02.459384500   File "/usr/lib/python3.8/site-packages/dbus_fast/introspection.py", line 5, in <module>
[... cut ...]
2023-11-17 17:36:02.459391500   File "/usr/lib/python3.8/site-packages/dbus_fast/errors.py", line 57, in <module>
2023-11-17 17:36:02.459394500     from .message import Message
2023-11-17 17:36:02.459508500   File "src/dbus_fast/message.py", line 3, in init dbus_fast.message
2023-11-17 17:36:02.459511500 ModuleNotFoundError: No module named 'dbus_fast._private.constants'
2023-11-17 17:36:11.759862500 ERROR:SerialBattery:No BMS found at C8:47:80:01:92:CE
2023-11-17 17:36:11.760646500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble

mr-manuel, 2023-11-17:
Did you reboot after install, right?

Seems that dbus-fast is still missing. Try to install it manually and see if it throws an error.

```bash
export SKIP_CYTHON=false
pip3 install dbus-fast
```

DaweeCZ, 2023-11-17:
Yeah, rebooted after every install and uninstall previously. 

Last login: Fri Nov 17 17:46:24 2023 from 192.168.1.22
root@einstein:~# export SKIP_CYTHON=false
root@einstein:~# pip3 install dbus-fast
Requirement already satisfied: dbus-fast in /usr/lib/python3.8/site-packages (2.14.0)

And still same errors on log.

mr-manuel, 2023-11-17:
```bash
export SKIP_CYTHON=false
pip3 install dbus-fast==1.91.3
```

DaweeCZ, 2023-11-17:
Perfect! Now it works! Thank you very much!

mr-manuel, 2023-11-17:
This is very strange. Sometimes it works, sometimes it does not work with the current dbus-fast version...

DaweeCZ, 2023-11-19:
Let me get back to the topic; I had to reinstall the new DBUS 2.14.0 because I found that ESS stopped functioning, not responding to commands like 'keep the battery charged' Hopefully, the RS485 - USB converters will arrive tomorrow, so I'll try connecting via cable to the new DBUS.

There was also an issue where one of the two BMS froze and stopped updating; Cerbo restarted twice overnight, I'm not sure why. Since I paused the driver, everything has been fine.

mr-manuel, 2023-11-19:
The cerbo most likely restarts because of overload, then the watchdog triggers a reboot.

DaweeCZ, 2023-11-19:
Yes, I think so too because during Bluetooth communication between two BMSs, even the remote console was quite sluggish. Once the converters arrive, I'll try to set it up again, hoping it'll work better than the Bluetooth connection. I'll be connecting a total of 4 BMS.

DaweeCZ, 2023-11-20:
So the converters arrived, I have it set up now in the classic way: BMS -> JK 485 adapter -> 485 to USB converter into the USB hub. After connecting and reinstalling the drivers on dbus-fast 2.14.0, nothing happened. I installed dbus-fast 1.91.3, reinstalled the driver again, and the BMS systems logged in. Then I installed dbus-fast 2.14.0, and the BMS systems are still connected. It looks like it's working. Rebooted after every install.

mr-manuel, 2023-11-20:
dbus-fast is not needed for the serial connection.

DaweeCZ, 2023-11-20:
So I'm not sure why the BMS systems didn't log in right after installation and restart, but only after reinstalling to dbus-fast 1.91.3, but I don't have to worry about that now. The main thing is that it's working now. The rest of the cells should arrive within a month, so then I'll set up the driver to aggregate the batteries into one. Thanks for the support!

mr-manuel, 2023-11-20:
You're welcome.

closed 2023-11-20 by mr-manuel