Louisvdw/dbus-serialbattery #866: Don't show SOC on Pages

Lacuranas, 2023-11-19:
### Describe the bug

After an update to Venus os 3.0 the SOC wasn't shown on Pages. It was shown in the Menu under BMS, and the BMS was correcly connected too. Also the SOC weren't in the VRM Portal

### How to reproduce

1.  Run clearly in Venus os 2.9.3
2. Update to Venus os 3.12

### Expected behavior

[image]
[image]

### Driver version

1.0.20230631

### Venus OS device type

Raspberry Pi

### Venus OS version

3.12

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

CVCM_ENABLE = False
CCCM_SOC_ENABLE = False
```

### Relevant log output

```shell
@400000006559da2d2a436f64 *** CCGX booted (0) ***
@400000006559da2e18ce45ec *** starting serial-starter ***
@400000006559da2e27db839c serstart starting
@400000006559da2e285fd2b4 INFO: loading config file /etc/venus/serial-starter.conf
@400000006559da2f159afba4 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@400000006559da3035d57c8c INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@400000006559da46356d5c4c INFO: Create daemontools service dbus-modbus-client.serial.ttyUSB1
@400000006559da482124b7e4 INFO: Create daemontools service dbus-cgwacs.ttyUSB2
@400000006559da4a01afc534 INFO: Create daemontools service mk2-dbus.ttyUSB3
@400000006559da4b051a09b4 INFO: Start service dbus-cgwacs.ttyUSB0 once
@400000006559da4d1286a33c INFO: Start service dbus-modbus-client.serial.ttyUSB1 once
@400000006559da4e24785f54 INFO: Start service dbus-cgwacs.ttyUSB2 once
@400000006559da500a320604 INFO: Start service mk2-dbus.ttyUSB3
@400000006559da531ef1c2b4 INFO: Create daemontools service dbus-serialbattery.ttyUSB1
@400000006559da5921d8aa4c INFO: Start service dbus-serialbattery.ttyUSB1 once
@400000006559db79038afd74 INFO: Create daemontools service gps-dbus.ttyUSB1
@400000006559db7f067e9af4 INFO: Start service gps-dbus.ttyUSB1 once
@400000006559db8825660b14 INFO: Create daemontools service vedirect-interface.ttyUSB1
@400000006559db8e286c84b4 INFO: Start service vedirect-interface.ttyUSB1 once
@400000006559db92047fb1fc INFO: Create daemontools service dbus-cgwacs.ttyUSB1
@400000006559db98078a715c INFO: Start service dbus-cgwacs.ttyUSB1 once
@400000006559db9d24fb2f9c INFO: Create daemontools service dbus-fzsonick-48tl.ttyUSB1
@400000006559dba3281fa11c INFO: Start service dbus-fzsonick-48tl.ttyUSB1 once
@400000006559dba704ffabb4 INFO: Create daemontools service dbus-imt-si-rs485tc.ttyUSB1
@400000006559dbad07ee6b1c INFO: Start service dbus-imt-si-rs485tc.ttyUSB1 once
@400000006559dbb01f8c7ffc INFO: Start service dbus-modbus-client.serial.ttyUSB1 once
@400000006559dbb739e15c24 INFO: Start service dbus-serialbattery.ttyUSB1 once
```

mr-manuel, 2023-11-19:
This is probably because under Settings -> DVCC -> Controlling BMS it did not select the SerialBattery as controlling BMS.

Lacuranas, 2023-11-20:
Thanks for the fast answer. I tried this and restart the driver/ reboot the pi, but unfortunately this didn't work

mr-manuel, 2023-11-25:
Check also what you selected under Settings -> System Setup -> Battery monitor.

Lacuranas, 2023-11-25:
Thak you very much. That fixed the problem

closed 2023-11-25 by mr-manuel