Louisvdw/dbus-serialbattery #872: Cell Voltages dissapear

DaweeCZ, 2023-11-25:
### Describe the problem

I noticed that with one of the two connected BMS units, the 'cell voltages' disappear, thus halting the graph recording. After a restart, it works for a few hours again, then disappears once more. The second battery seems to consistently display the data correctly.

### Driver version

1.0.20231117dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.20 ~ 30

### BMS type

JKBMS / Heltec BMS

### Cell count

15

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.500

LINEAR_LIMITATION_ENABLE = False
CVCM_ENABLE = False
CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False
BMS_TYPE = Jkbms

CUSTOM_BATTERY_NAMES = /dev/ttyUSB0:Battery 1,/dev/ttyUSB1:Battery 2,/dev/ttyUSB2:Battery 3,/dev/ttyUSB3:Battery 4
```

### Relevant log output

```shell
@40000000655dea1534e3b9ec ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea1634f66344 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea17350f7d0c ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea18351a1bcc ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea193531ce34 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea1a3566ff64 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
@40000000655dea1b3599ac0c ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
[... 156 lines cut ...]
@400000006561bdab15ff1c74 INFO:SerialBattery:
@400000006561bdab16de39a4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006561bdab17e31eb4 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006561bdbb18d3eb3c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006561bdbb18d4124c INFO:SerialBattery:Testing Jkbms
@400000006561bdbb2182451c INFO:SerialBattery:Connection established to Jkbms
@400000006561bdbb2274d2dc INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
@400000006561bdbb22c81744 INFO:SerialBattery:========== Settings ==========
@400000006561bdbb22d90b1c INFO:SerialBattery:> Connection voltage: 49.89V | Current: -0.0A | SoC: 93%
@400000006561bdbb22ea87ac INFO:SerialBattery:> Cell count: 15 | Cells populated: 0
@400000006561bdbb22fa2b94 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
@400000006561bdbb23c1195c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006561bdbb241ba8f4 INFO:SerialBattery:> CVCM:     False
@400000006561bdbb25edc914 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
@400000006561bdbb26023f5c INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
@400000006561bdbb26123934 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
@400000006561bdbb266f35e4 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@400000006561bdbb269b4b0c INFO:SerialBattery:Serial Number/Unique Identifier: Battery_1
@400000006561bdbb3319b404 INFO:SerialBattery:DeviceInstance = 4
@400000006561bdbb333cef8c INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
@400000006561bdbc01ac17cc INFO:SerialBattery:publish config values = 1
@400000006561bdc51bdd627c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006561bdc61c1e9db4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006561bdc71c373a7c ERROR:SerialBattery:>>> ERROR: No reply - returning
```

mr-manuel, 2023-11-25:
[image]

mr-manuel, 2023-11-25:
https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#driver-log-files

[image]

DaweeCZ, 2023-11-25:
I apologize, here are the logs for USB0 (Battery 1) and USB1 (Battery 2).

root@einstein:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2023-11-24 18:23:43.001347500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-11-24 18:23:43.013075500 INFO:SerialBattery:Serial Number/Unique Identifier: Battery_1
2023-11-24 18:23:43.191489500 INFO:SerialBattery:DeviceInstance = 4
2023-11-24 18:23:43.192447500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2023-11-24 18:23:43.402658500 INFO:SerialBattery:publish config values = 1
2023-11-24 18:23:51.805172500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:23:52.816621500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:24:09.607182500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:224/5252]
2023-11-24 18:36:14.253692500 *** CCGX booted (0) ***
2023-11-24 18:37:16.806685500 INFO:SerialBattery:
2023-11-24 18:37:16.813019500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-24 18:37:16.820805500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-24 18:37:32.835182500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-11-24 18:37:32.835188500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:37:32.977975500 INFO:SerialBattery:Connection established to Jkbms
2023-11-24 18:37:32.980038500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2023-11-24 18:37:32.998029500 INFO:SerialBattery:========== Settings ==========
2023-11-24 18:37:33.004197500 INFO:SerialBattery:> Connection voltage: 48.22V | Current: -29.63A | SoC: 25%
2023-11-24 18:37:33.006949500 INFO:SerialBattery:> Cell count: 15 | Cells populated: 0
2023-11-24 18:37:33.009695500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2023-11-24 18:37:33.015248500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2023-11-24 18:37:33.017933500 INFO:SerialBattery:> CVCM:     False
2023-11-24 18:37:33.021984500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2023-11-24 18:37:33.034266500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2023-11-24 18:37:33.035604500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2023-11-24 18:37:33.038112500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-11-24 18:37:33.066177500 INFO:SerialBattery:Serial Number/Unique Identifier: Battery_1
2023-11-24 18:37:33.246654500 INFO:SerialBattery:DeviceInstance = 4
2023-11-24 18:37:33.248121500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2023-11-24 18:37:33.363775500 INFO:SerialBattery:publish config values = 1
2023-11-24 18:45:59.239037500 *** CCGX booted (0) ***
2023-11-24 18:46:17.243621500 Traceback (most recent call last):
2023-11-24 18:46:17.243632500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 25, in <module>
2023-11-24 18:46:17.243636500     from bms.ecs import Ecs
2023-11-24 18:46:17.243638500 ModuleNotFoundError: No module named 'bms.ecs'
2023-11-24 18:46:34.264760500 INFO:SerialBattery:
2023-11-24 18:46:34.266046500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-24 18:46:34.267526500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-24 18:46:50.275276500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-11-24 18:46:50.276171500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:46:50.577265500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:46:51.080511500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-11-24 18:46:51.080520500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:46:51.366892500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:46:51.868653500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-11-24 18:46:51.870702500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:47:07.959570500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:73/14392]
2023-11-24 18:47:08.480092500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2023-11-24 18:47:18.893241500 INFO:SerialBattery:
2023-11-24 18:47:18.894162500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-24 18:47:18.895879500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-24 18:47:34.905470500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-11-24 18:47:34.906192500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:47:35.206218500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:47:35.709550500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-11-24 18:47:35.709559500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:47:36.018262500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-24 18:47:36.528014500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-11-24 18:47:36.528024500 INFO:SerialBattery:Testing Jkbms
2023-11-24 18:47:36.668802500 INFO:SerialBattery:Connection established to Jkbms
2023-11-24 18:47:36.672607500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2023-11-24 18:47:36.677969500 INFO:SerialBattery:========== Settings ==========
2023-11-24 18:47:36.679305500 INFO:SerialBattery:> Connection voltage: 48.79V | Current: -3.25A | SoC: 24%
2023-11-24 18:47:36.680523500 INFO:SerialBattery:> Cell count: 15 | Cells populated: 0
2023-11-24 18:47:36.684107500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2023-11-24 18:47:36.690528500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
[... cut ...]
2023-11-25 14:51:25.163302500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:287/287]
2023-11-25 14:56:00.783610500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x5895, Got 0x587d
2023-11-25 14:56:43.016403500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-11-25 14:58:19.012356500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x58a5, Got 0x58a4
2023-11-25 14:58:38.033812500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x58ac, Got 0x587c
2023-11-25 14:59:17.094666500 ERROR:SerialBattery:CRC checksum mismatch: Expected 0x58b0, Got 0x5880

DaweeCZ, 2023-11-25:
And this command:
tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal

Does nothing. SSH froze, and not respond.

mr-manuel, 2023-11-25:
[image]

If there are no logs, you will not see anything :-)

DaweeCZ, 2023-11-25:
Apologies, I'm not very familiar with SSH or Linux... I've had this command open for about 5 minutes, and no log has appeared, so is this normal behavior?

DaweeCZ, 2023-11-27:
Hello, any updates?

mr-manuel, 2023-11-27:
A log like this mostly comes from not shielded cables.

1. Are your connection cables shielded?
2. Is the data cable near the battery cable?
 
Please post some pictures.

DaweeCZ, 2023-11-27:
[image]
[image]

Both battery connected connected from JK BMS -> JK BMS GPS to RS485 -> RS485 to USB adapter -> USB hub -> UBS extension -> Cerbo GX. But only second battery stops sending cell voltages.

DaweeCZ, 2023-11-27:
But other information is being transmitted, like SOC, voltage, etc. Only the submenu for Cell Voltages disappears. Could this also be caused by interference?

mr-manuel, 2023-11-27:
It's also very often that the USB Hub causes this type of problems.

pos-ei-don, 2023-11-27:
> It's also very often that the USB Hub causes this type of problems.

Which I had!! especially som new Hbs hat this problem, if more than one battery was connected! 
An Old D-Link quality hub was my solution, this one works very stable!!

DaweeCZ, 2023-11-27:
Do you think trying a different USB hub might help, or is there currently no way to avoid this? I'm concerned because I'm using EM540, so I only have one free USB and I'd like to connect 4x JKBMS.

DaweeCZ, 2023-11-27:
> > It's also very often that the USB Hub causes this type of problems.
> 
> Which I had!! especially som new Hbs hat this problem, if more than one battery was connected! An Old D-Link quality hub was my solution, this one works very stable!!

Could you please specify which USB hub exactly you're using?

DaweeCZ, 2023-11-27:
> It's also very often that the USB Hub causes this type of problems.

When I finish setting up the rest of the batteries and do the cable management, do you recommend making the RS485 or USB cables longer? What is more affected by longer cable lengths?

pos-ei-don, 2023-11-27:
I have not tried long r485 cables, but my USB is 5m.

closed 2023-12-17 by mr-manuel