Louisvdw/dbus-serialbattery #877: Frequent driver restarts stop, when logging is reconfigured to DEBUG

beliffm, 2023-11-26:
### Describe the problem

I have been experiencing frequent driver restarts on both of my two BMSes. They all take place in a similar pattern which can be seen in the log below, In order to gain more information to tackle the problem i switched the log level to debug, only to find that the restarts stopped. I monitored the situation for more than two hours, not a single driver restart. What happened after i reconfigured to default logging can be seen below. Only after some minutes there was a restart.
My guess is, given that the serial connection is only 9600 baud, the polling is somewhat to "fast" and the serial buffer is not yet filled, which is then interpreted wrongly and triggers the restart. A higher log lever slows the polling loop down, thus avoiding the situation of an empty buffer. 

### Driver version

v1.0.20231117dev

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

V3.20~26

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0
BMS_TYPE = LltJbd
; LOGGING = DEBUG
```

### Relevant log output

```shell
2023-11-26 17:24:14.664824500 INFO:SerialBattery:
2023-11-26 17:24:14.665999500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-26 17:24:14.668060500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-26 17:24:30.673672500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-11-26 17:24:30.674295500 INFO:SerialBattery:Testing LltJbd
2023-11-26 17:24:31.281704500 INFO:SerialBattery:Connection established to LltJbd
2023-11-26 17:24:31.282234500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB1
2023-11-26 17:24:31.282742500 INFO:SerialBattery:========== Settings ==========
2023-11-26 17:24:31.283254500 INFO:SerialBattery:> Connection voltage: 52.49V | Current: 2.0A | SoC: 82.17%
2023-11-26 17:24:31.283669500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-11-26 17:24:31.284077500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-11-26 17:24:31.284480500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-11-26 17:24:31.284880500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2023-11-26 17:24:31.285277500 INFO:SerialBattery:> CVCM:     True
2023-11-26 17:24:31.285691500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-11-26 17:24:31.286085500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-11-26 17:24:31.286468500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-11-26 17:24:31.286844500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-11-26 17:24:31.287308500 INFO:SerialBattery:Serial Number/Unique Identifier: 21_280.0Ah
2023-11-26 17:24:31.345424500 INFO:SerialBattery:DeviceInstance = 1
2023-11-26 17:24:31.346240500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2023-11-26 17:24:31.680194500 INFO:SerialBattery:publish config values = 1
2023-11-26 17:32:07.435580500 ERROR:SerialBattery:>>> ERROR: Invalid response packet. Expected begin packet character 0xDD
2023-11-26 17:32:07.443495500 Traceback (most recent call last):
2023-11-26 17:32:07.443506500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 351, in publish_battery
2023-11-26 17:32:07.443511500     result = self.battery.refresh_data()
2023-11-26 17:32:07.443513500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 444, in refresh_data
2023-11-26 17:32:07.443516500     return self.read_gen_data() and self.read_cell_data()
2023-11-26 17:32:07.443519500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 532, in read_gen_data
2023-11-26 17:32:07.443635500     gen_data = self.read_serial_data_llt(self.command_general)
2023-11-26 17:32:07.443638500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 653, in read_serial_data_llt
2023-11-26 17:32:07.443642500     return self.validate_packet(data)
2023-11-26 17:32:07.443644500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 625, in validate_packet
2023-11-26 17:32:07.443648500     logger.warn(">>> WARN: BMS rejected request. Status " + status)
2023-11-26 17:32:07.443669500 TypeError: can only concatenate str (not "int") to str
2023-11-26 17:32:07.443882500 #0
2023-11-26 17:32:07.443885500 #1
2023-11-26 17:32:07.443886500 #2
2023-11-26 17:32:07.443887500 #3
2023-11-26 17:32:07.443888500 #4
2023-11-26 17:32:07.443889500 #5
2023-11-26 17:32:07.443889500 #6
2023-11-26 17:32:07.443890500 #7
2023-11-26 17:32:07.443891500 #8
2023-11-26 17:32:07.443892500 #9
2023-11-26 17:32:07.443893500 #10
2023-11-26 17:32:07.443894500 #11
2023-11-26 17:32:07.443895500 #12
2023-11-26 17:32:07.443896500 #13
2023-11-26 17:32:07.443897500 #14
2023-11-26 17:32:07.443898500 #15
2023-11-26 17:32:40.834222500 INFO:SerialBattery:
2023-11-26 17:32:40.835451500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-26 17:32:40.837590500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-26 17:32:56.853645500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-11-26 17:32:56.854219500 INFO:SerialBattery:Testing LltJbd
2023-11-26 17:32:57.479204500 INFO:SerialBattery:Connection established to LltJbd
2023-11-26 17:32:57.479746500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB1
2023-11-26 17:32:57.480152500 INFO:SerialBattery:========== Settings ==========
2023-11-26 17:32:57.480620500 INFO:SerialBattery:> Connection voltage: 52.49V | Current: 2.08A | SoC: 81.49%
2023-11-26 17:32:57.481006500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-11-26 17:32:57.481403500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-11-26 17:32:57.481809500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-11-26 17:32:57.482210500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2023-11-26 17:32:57.482578500 INFO:SerialBattery:> CVCM:     True
2023-11-26 17:32:57.483095500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-11-26 17:32:57.483503500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-11-26 17:32:57.483887500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-11-26 17:32:57.484268500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-11-26 17:32:57.484744500 INFO:SerialBattery:Serial Number/Unique Identifier: 21_280.0Ah
2023-11-26 17:32:57.541866500 INFO:SerialBattery:DeviceInstance = 1
2023-11-26 17:32:57.542778500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2023-11-26 17:32:57.887155500 INFO:SerialBattery:publish config values = 1
2023-11-26 17:37:29.240003500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/119]
```

stixif, 2023-11-27:
here the same with 2 Batterys

It looks like it only happens during discharge

Nr. 1
`@400000006564d9be339b0964 INFO:SerialBattery:
@400000006564d9be33ae4344 INFO:SerialBattery:Starting dbus-serialbattery
@400000006564d9be33da509c INFO:SerialBattery:dbus-serialbattery v1.0.20231126dev
@400000006564d9c918578b64 INFO:SerialBattery:
@400000006564d9c9186041dc INFO:SerialBattery:Starting dbus-serialbattery
@400000006564d9c91874279c INFO:SerialBattery:dbus-serialbattery v1.0.20231126dev
@400000006564d9d91931554c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006564d9d9193c422c INFO:SerialBattery:Testing LltJbd
@400000006564d9da116a9c4c INFO:SerialBattery:Connection established to LltJbd
@400000006564d9da11736a34 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@400000006564d9da117af3e4 INFO:SerialBattery:========== Settings ==========
@400000006564d9da11825684 INFO:SerialBattery:> Connection voltage: 56.88V | Current: 0.0A | SoC: 28.44%
@400000006564d9da118a45c4 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@400000006564d9da1191d35c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@400000006564d9da119a08ec INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006564d9da11a22edc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0A | MAX BATTERY DISCHARGE CURRENT: 190.0A (read from BMS)
@400000006564d9da11aa452c INFO:SerialBattery:> CVCM:     True
@400000006564d9da11b2058c INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006564d9da11b9be1c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006564d9da11c16edc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006564d9da11e9599c INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@400000006564d9da11ed0aec INFO:SerialBattery:Serial Number/Unique Identifier: 40_280.0Ah
@400000006564d9da2f0fe7ac INFO:SerialBattery:Found existing battery with DeviceInstance = 1
@400000006564d9da39b1284c INFO:SerialBattery:DeviceInstance = 1
@400000006564d9da39bd6904 INFO:SerialBattery:Used device instances: ['1']
@400000006564d9da39c78eac INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@400000006564d9eb01d064c4 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:31/36]
@400000006564d9eb020c298c ERROR:SerialBattery:ERROR >>> Problem with battery set up at /dev/ttyACM0
@400000006564d9eb022877a4 #0
@400000006564d9eb022892fc #1
@400000006564d9eb022896e4 #2
@400000006564d9eb02289acc #3
@400000006564d9eb0228a29c #4
@400000006564d9eb0228a684 #5
@400000006564d9eb0228aa6c #6
@400000006564d9eb0228b23c #7
@400000006564d9eb0228b624 #8
@400000006564d9eb0228ba0c #9
@400000006564d9eb0228bdf4 #10
@400000006564d9eb0228c5c4 #11
@400000006564d9eb0228c9ac #12
@400000006564d9eb0228cd94 #13
@400000006564d9eb0228d564 #14
@400000006564d9eb0228d94c #15
@400000006564d9eb0228dd34 #16
@400000006564d9eb0228e504 #17
@400000006564d9eb022969d4 Setted setting /Settings/Devices/serialbattery_40_280_0Ah/LastSeen to 1701108176
@400000006564d9fa00ffb2fc INFO:SerialBattery:
@400000006564d9fa010734dc INFO:SerialBattery:Starting dbus-serialbattery
@400000006564d9fa0116d0f4 INFO:SerialBattery:dbus-serialbattery v1.0.20231126dev
@400000006564da0a01ec9784 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006564da0a01f5750c INFO:SerialBattery:Testing LltJbd
@400000006564da0a36208f4c INFO:SerialBattery:Connection established to LltJbd
@400000006564da0a3630d744 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@400000006564da0a36430b6c INFO:SerialBattery:========== Settings ==========
@400000006564da0a364326c4 INFO:SerialBattery:> Connection voltage: 56.91V | Current: 1.29A | SoC: 28.44%
@400000006564da0a366e789c INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@400000006564da0a368081cc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@400000006564da0a368563cc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006564da0a36b9dd64 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0A | MAX BATTERY DISCHARGE CURRENT: 190.0A (read from BMS)
@400000006564da0a36bbd934 INFO:SerialBattery:> CVCM:     True
@400000006564da0a3715ebcc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006564da0a3720b584 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006564da0a3729642c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006564da0a3731e3f4 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
[... 53 lines cut ...]
@400000006564dd8d02ca5cf4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd8e11630acc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd8f033714fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd90122e982c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd91031eaafc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd920377a454 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd93036cde84 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd94044fa14c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd9503768efc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd9603db925c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd9703932f1c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd9803b2d0c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
[... cut ...]
@400000006564dd9e04e85edc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dd9f0513c054 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dda005402784 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dda105d0de64 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564dda205fba784 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564ddab12499a3c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564ddac0628b2c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
`

stixif, 2023-11-27:
i changed it...

but the failure is the same...

**Nr1.**
@400000006564f62d34b6fccc INFO:SerialBattery:
@400000006564f62d34bdafa4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006564f62d34f63c34 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006564f6380f07c934 INFO:SerialBattery:
@400000006564f6380f107fac INFO:SerialBattery:Starting dbus-serialbattery
@400000006564f6380f2a416c INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006564f6481010c524 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006564f6481017e55c INFO:SerialBattery:Testing LltJbd
@400000006564f6583b638f54 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:14/164]
@400000006564f6591dc7776c INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@400000006564f6591dcece54 INFO:SerialBattery:Testing LltJbd
@400000006564f65a15895034 INFO:SerialBattery:Connection established to LltJbd
@400000006564f65a1590d9e4 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@400000006564f65a15984c24 INFO:SerialBattery:========== Settings ==========
@400000006564f65a15a0106c INFO:SerialBattery:> Connection voltage: 58.45V | Current: 0.0A | SoC: 30.18%
@400000006564f65a15a730a4 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@400000006564f65a15ae1a2c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@400000006564f65a15b5fdb4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006564f65a15bd73dc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0A | MAX BATTERY DISCHARGE CURRENT: 190.0A (read from BMS)
@400000006564f65a15c4614c INFO:SerialBattery:> CVCM:     True
@400000006564f65a15cb98f4 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006564f65a15d316ec INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006564f65a15daa484 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006564f65a15e212dc INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@400000006564f65a15ea486c INFO:SerialBattery:Serial Number/Unique Identifier: 40_280.0Ah
@400000006564f65a1899ef7c INFO:SerialBattery:DeviceInstance = 2
@400000006564f65a18a4113c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@400000006564f65a3985846c INFO:SerialBattery:publish config values = 1

**Nr.2**
@400000006564f62e09c843cc INFO:SerialBattery:
@400000006564f62e09cf4c94 INFO:SerialBattery:Starting dbus-serialbattery
@400000006564f62e09e19ffc INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006564f6380e35deec INFO:SerialBattery:
@400000006564f6380e3e0cac INFO:SerialBattery:Starting dbus-serialbattery
@400000006564f6380e531b4c INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
@400000006564f6480f55ed1c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@400000006564f6480f5d7e9c INFO:SerialBattery:Testing LltJbd
@400000006564f64909058c24 INFO:SerialBattery:Connection established to LltJbd
@400000006564f649090fa9fc INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM1
@400000006564f649091658ec INFO:SerialBattery:========== Settings ==========
@400000006564f649091db7a4 INFO:SerialBattery:> Connection voltage: 59.65V | Current: 76.22A | SoC: 30.73%
@400000006564f649092462ac INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
@400000006564f649092bcd1c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@400000006564f6490935cbb4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@400000006564f649093e37f4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 190.0A | MAX BATTERY DISCHARGE CURRENT: 190.0A (read from BMS)
@400000006564f64909460024 INFO:SerialBattery:> CVCM:     True
@400000006564f649094db4cc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006564f649095561a4 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006564f649095da2ec INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006564f64909653854 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@400000006564f649096e7b6c INFO:SerialBattery:Serial Number/Unique Identifier: 40_279.0Ah
@400000006564f6490dfc8f0c INFO:SerialBattery:DeviceInstance = 3
@400000006564f6490e09aa84 INFO:SerialBattery:com.victronenergy.battery.ttyACM1
@400000006564f6492ef0f1e4 INFO:SerialBattery:publish config values = 1
@400000006564f6ca0ffdf8a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6cb1066e7ec ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6cc1e1485fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6cd0fece1a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6ce1ecd6234 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6cf0ff76124 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6d01de3c19c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6d11075208c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6d21e35258c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f6d31050a4b4 ERROR:SerialBattery:>>> ERROR: No reply - returning
[... 124 lines cut ...]
@400000006564f77a1be2f82c ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f77b1beb0a94 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f77c1c4f69e4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f77d1c38a9ac ERROR:SerialBattery:>>> ERROR: No reply - returning
[... cut ...]
@400000006564f78d1d9cc5bc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f78e1d313e64 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f78f2cbe9124 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f7901d9edce4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f7911df80134 ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f7921d8e3efc ERROR:SerialBattery:>>> ERROR: No reply - returning
@400000006564f7931da63b9c ERROR:SerialBattery:>>> ERROR: No reply - returning

beliffm, 2023-11-27:
@stixif Does your problem go away when you change the logging to DEBUG? To me it does not look like it. So it might be something else, given that it is only the one BMS that fails not both.

stixif, 2023-11-28:
@beliffm yes to both...

In the Last months it worket normaly

stixif, 2023-11-28:
with the newest dev from 28.11 it looks like good!

mr-manuel, 2023-11-29:
Are you using Ruuvi tags?

stixif, 2023-11-29:
@mr-manuel 
no not realy...

stixif, 2023-11-29:
@mr-manuel
now today by charging the battery all is OK...
but on discharging the errors come back...

@4000000065676b3611eb8c1c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000065676b3611f2fa74 INFO:SerialBattery:Testing LltJbd
@4000000065676b362222b754 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b37049b1d84 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@4000000065676b37049b34f4 INFO:SerialBattery:Testing LltJbd
@4000000065676b37145ea72c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b37327171cc INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@4000000065676b373285f3cc INFO:SerialBattery:Testing LltJbd
@4000000065676b3806b01304 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b3825173764 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM1
@4000000065676b481dfb607c INFO:SerialBattery:
@4000000065676b481e0883c4 INFO:SerialBattery:Starting dbus-serialbattery
@4000000065676b481e1ea3d4 INFO:SerialBattery:dbus-serialbattery v1.0.20231128dev
@4000000065676b581f33ffbc INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000065676b581f341b14 INFO:SerialBattery:Testing LltJbd
@4000000065676b582faed72c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b59120230fc INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@4000000065676b5912024c54 INFO:SerialBattery:Testing LltJbd
@4000000065676b5922cde3a4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b5a051b9ff4 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@4000000065676b5a05300e6c INFO:SerialBattery:Testing LltJbd
@4000000065676b5a15134d44 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065676b5a33648844 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM1
@4000000065676b6a2d69ec54 INFO:SerialBattery:

mr-manuel, 2023-11-29:
Probably one value is out of the bounds. I will add some debugging to check that.

stixif, 2023-11-29:
i checked my values in the BMS and found an  empty one...
"Discharge under Temperature release"
i set them now and will check again...
thanks for your help

beliffm, 2023-11-30:
> @beliffm could you also test the latest build?

Will do...

beliffm, 2023-11-30:
No change, restarts happening on both BMS every ten minutes or so.

mr-manuel, 2023-12-01:
Try to set this to 2000.

https://github.com/Louisvdw/dbus-serialbattery/blob/c6d5bb6fcdc0d08404a0849345d6c332536b710c/etc/dbus-serialbattery/battery.py#L63

beliffm, 2023-12-01:
> Try to set this to 2000.
> 
> https://github.com/Louisvdw/dbus-serialbattery/blob/c6d5bb6fcdc0d08404a0849345d6c332536b710c/etc/dbus-serialbattery/battery.py#L63

That helped. No more restarts. All that is left now since 8:07, when i made the change, is what can be seen in the log excerpt below. Those errors may result from a different issue though.

 ==> /data/log/dbus-serialbattery.ttyUSB1/current <==
2023-12-01 08:18:24.037539500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/104]
2023-12-01 08:41:13.552683500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/258]
2023-12-01 09:15:37.817786500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/98]

==> /data/log/dbus-serialbattery.ttyUSB2/current <==
2023-12-01 09:27:34.704079500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:33/259]

==> /data/log/dbus-serialbattery.ttyUSB1/current <==
2023-12-01 09:33:31.003588500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:34/94]

mr-manuel, 2023-12-01:
Maybe your BMS gets overloaded with polling every 2 seconds. What is your cpu usage of the gx card when you set it again to 1000?

beliffm, 2023-12-01:
> Maybe your BMS gets overloaded with polling every 2 seconds. What is your cpu usage of the gx card when you set it again to 1000?

CPU usage is about 10% in both cases... Maybe polling every second really is too fast for the BMSes. Since this morning there have been some occurances, but by far not as many as before.

mr-manuel, 2023-12-01:
If you did not already try other USB to serial converters it could also be that the converter is the problem.

closed 2023-12-03 by mr-manuel

beliffm, 2023-12-10:
Finally the quad-converter from duppa.net arrived yesterday. Since i changed my build-up to this converter the restarts have stopped completely even if i reset the polling rate back to 1000ms. Mind you, the LLTJBD-BMS has 12V supply on the UART connector which has to be converted to 5V to suit the Duppa.net converter. Formerly i used two converters which came with the BMSs together with an Auge USB hub. Maybe the GX card does not provide enough power for such a construction at high polling rates.
So thank you all for your patience and helping hands.