Louisvdw/dbus-serialbattery #887: Serval web links in the section how to install do not work , this makes it all very confusing how to install

merlep, 2023-12-11:
### Describe the problem

On webpage https://louisvdw.github.io/dbus-serialbattery/general/general/install#install-over-ssh  , under "Latest release " the link named "Install script " does not go to a valid location ( url https://louisvdw.github.io/dbus-serialbattery/general/general/install#install-over-ssh ) . Several other links on that page also do not work - makes it impossible to understand what the correct command is to install the latest release version

### Driver version

v1.00

### Venus OS device type

Cerbo GX

### Venus OS version

3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
asd
```

### Relevant log output

```shell
we
```

### Any other information that may be helpful

sad

mr-manuel, 2023-12-11:
Good question why. The links in the source code are correct. I have to take a deeper look.

mr-manuel, 2024-01-05:
Unfortunately the error is still present. When building locally there is no issue.

mr-manuel, 2024-01-05:
Found it. Problem fixed.

commit f16eba62 (2024-01-05): Fixes #887 (#905)

* Fixes #887
https://github.com/Louisvdw/dbus-serialbattery/issues/887

* Updated issue templates.
closed 2024-01-05 by mr-manuel