Louisvdw/dbus-serialbattery #891: DIY 18S Batteriepack+Raspberry+Multiplus 2 low Battery Alarm

1984chris, 2023-12-13:
### Describe the problem

Hello
I have a problem and can't get any further.
I have an 18S battery pack with a Jiabaida AP21S002 connected to a JBD-UART-TOOLS-RS485 smart bms rs485 to the USB port on the Raspberry, and a Multiplus 2 5000 with the MK3 adapter also on the Raspberry. Venus OS configured as ESS.
If I disconnect the battery pack from the Raspberry and switch on the Multiplus, the battery is discharged in the ESS menu Set Gridpointhave I set e.g. -50 watts that works.
But when I connect the battery pack to the Raspberry and switch on the Multiplus, the red low battery LED on the Multiplus lights up and switches off the inverter. Attached photos 

LG Christoph

Translated with DeepL.com (free version)

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.13

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

18

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MIN_CELL_VOLTAGE = 2.8
FLOAT_CELL_VOLTAGE = 3.4
BMS_TYPE = LltJbd
;SOC_LOW_WARNING = 0
;SOC_LOW_ALARM = 0

LINEAR_LIMITATION_ENABLE = True

CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False
```

### Relevant log output

```shell
2023-12-13 19:58:47.232241500 *** CCGX booted (0) ***
2023-12-13 19:59:12.898944500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-13 19:59:12.900202500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-12-13 19:59:12.900705500 INFO:SerialBattery:Testing LltJbd
2023-12-13 19:59:13.659855500 INFO:SerialBattery:Connection established to LltJbd
2023-12-13 19:59:13.659860500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
2023-12-13 19:59:13.659863500 INFO:SerialBattery:========== Settings ==========
2023-12-13 19:59:13.659866500 INFO:SerialBattery:> Connection voltage: 59.94V | Current: 0.0A | SoC: 69.97%
2023-12-13 19:59:13.659870500 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
2023-12-13 19:59:13.659873500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-12-13 19:59:13.660032500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 60.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2023-12-13 19:59:13.660037500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: -200.0A (read from BMS)
2023-12-13 19:59:13.660041500 INFO:SerialBattery:> CVCM:     True
2023-12-13 19:59:13.660044500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.8V | MAX CELL VOLTAGE: 3.45V
2023-12-13 19:59:13.660047500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-12-13 19:59:13.660076500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2023-12-13 19:59:13.660080500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-12-13 19:59:13.710060500 INFO:SerialBattery:DeviceInstance = 1
2023-12-13 19:59:13.711823500 INFO:SerialBattery:com.victronenergy.battery.ttyACM0
2023-12-13 19:59:14.123694500 INFO:SerialBattery:publish config values = 1
```

### Any other information that may be helpful

[image]

iIs it normal for the DCL discharge current limitation to show a minus value?

[image]

1984chris, 2023-12-16:
Hello

Maybe someone will have the same problem.
I did it by selecting DVCC - Controlling BMS - No BMS control under Settings.
that's how it works for me.

LG Christoph

[image]
[image]

1984chris, 2023-12-16:
Erledigt

closed 2023-12-16 by 1984chris

mr-manuel, 2023-12-16:
Thats very strange that it shows a negative value. I will fix that in the next release.

mr-manuel, 2023-12-17:
This is fixed with `v1.0.20231217`.

commit 8813763d (2023-12-17): Fixes #891
https://github.com/Louisvdw/dbus-serialbattery/issues/891