Louisvdw/dbus-serialbattery #892: Too Many Cell Imbalance warnings due to hard coded values. Request remove hard coded values.

CRaccani, 2023-12-17:
### Is your feature request related to a problem? Please describe.

Congrats on this project. It is fabulous.

I'm getting too many Cell imbalance warnings during discharge.
My imbalance is definitely grater than 0.2 and 0.12
     

/jkbms_ble.py

       # trigger cell imbalance warning when delta is to great
        if st["cell_info"]["delta_cell_voltage"] > min(
            st["settings"]["cell_ovp"] * 0.05, 0.200
        ):
            self.protection.cell_imbalance = 2
        elif st["cell_info"]["delta_cell_voltage"] > min(
            st["settings"]["cell_ovp"] * 0.03, 0.120
        ):
            self.protection.cell_imbalance = 1
        else:
            self.protection.cell_imbalance = 0

### Describe the solution you'd like

There are CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL and CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT in the config parameters. Are these values not used. Seems like they should be used to set the acceptable cell imbalance.

mr-manuel, 2023-12-17:
1. Did you do a top balancing?
2. What for cells do you have?
3. Have they different size/age?
4. In which voltage range do you have a big cell imbalance?

cliffraccani, 2023-12-17:
1. Yes, top balanced every morning to 0.005V delta. Battery charged at off peak grid power overnight.
2. The battery is a 14S80P lion 18650 Tesla DIY battery design. OVP 4.2V, UVP 2.65V, Absorption 4.15V, Float 4.09V
3. This has been running for the past 5-6 years no problems. The cells are all laptop used cells and each of the 14, 80P cells were well matched at the beginning but now tend to drift apart during heavy discharge but do come together when left to balance.
4. The cell imbalance warning starts at around 3.5V down to 2.65V. The imbalance can increase to about 0.45V.

Would it be possible to turn off the warning to the Venus OS or enable a wider range for cell imbalance.

Thank you for the quick reply.
Kind regards,
Cliff

mr-manuel, 2024-01-03:
If I change it to only percentage values instead of percentage and fixed value, would that suit for you?

closed 2024-01-03 by mr-manuel

Honusnap, 2024-01-08:
Even if it is a rare situation, having the max acceptable delta as a parameter seems a good idea.

I got 2x14kWh pack and, i never go over 20mV of voltage delta, it happend on high charge rates when pack is at high SOC.

I agree with Manuel, cell discrepancy is not a good situation, since it generally happend at high SOC, maybe you are charging at a too high voltage, too high for the matching that  have been done on those cells..

"Yes, top balanced every morning to 0.005V" : Maybe you misunderstood Manuel, a topbalancing is not done everyday.. week .. or even month, In general it's a one shot action.

My methode is this one : Charge all cells to the max voltage, 4.2V in your case. Then parallel them and wait, note the voltage each day and see whjere the graph is "stabilising" = start to be horizontal. 
This is the voltage per cell you want to aim at, globally it's your worst cell higher spot. Like in a row boat team .. you are trying to find the weakest rower then you give this rythm to everybody => nobody is out of it own capabilities....
This way you will reduce natural inbalance, it still can exist if those cells are really badly matched, for example .. capacity wise....