Louisvdw/dbus-serialbattery #894: ERROR Exception during bluetooth installation (jkbms)

MisterX1000, 2023-12-19:
### Describe the problem

Hi,

I try to connect my jkbms via bluetooth to my Multiplus 2-GX because the cable connection sometimes "lost" my jkbms.

The installation log showed me some errors. All log files are attached. 

I also tried to install "opkg update" "opkg install python3-misc" & "python3-pip" by hand - same error message.

Not sure if the beta firmware (v3.20~34) of my MP2-GX causes this problems. 

It would be very kind of you if you could have a look and help me here.

Thank you very much in advance
Alex

### Driver version

v1.0.20231128beta

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

v3.20~34

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Bluetooth BMS ---------
; Description: Specify the Bluetooth BMS and it's MAC address that you want to install. Leave emty to disable
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example for one BMS:
; BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
; Example for multiple BMS:
; BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E5:1C:EB

; --------- SOC limitation (affecting CCL/DCL) ---------
CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
[... 224 lines cut ...]
BLUETOOTH battery connection: There are a few more steps to complete installation.

    1. Add your Bluetooth BMS to the config file "/data/etc/dbus-serialbattery/config.ini".
       Check the default config file "/data/etc/dbus-serialbattery/config.default.ini" for more informations.
       If your Bluetooth BMS are nearby you can show the MAC address with "bluetoothctl devices".

    2. Make sure to disable Bluetooth in "Settings -> Bluetooth" in the remote console/GUI to prevent reconnects every minute.

    3. Re-run "/data/etc/dbus-serialbattery/reinstall-local.sh", if the Bluetooth BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more details.

CAN battery connection: There are a few more steps to complete installation.

    1. Add your CAN port to the config file "/data/etc/dbus-serialbattery/config.ini".
       Check the default config file "/data/etc/dbus-serialbattery/config.default.ini" for more informations.

    2. Make sure to select a profile with 250 kbit/s in "Settings -> Services -> VE.Can port -> CAN-bus profile" in the remote console/GUI.

    3. Re-run "/data/etc/dbus-serialbattery/reinstall-local.sh", if the CAN port was not added to the "config.ini" before.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you want to change in "/data/etc/dbus-serialbattery/config.default.ini"
                 and add them to "/data/etc/dbus-serialbattery/config.ini" to persist future driver updates.

mr-manuel, 2023-12-19:
See https://github.com/Louisvdw/dbus-serialbattery/issues?q=No+module+named+%27bleak%27

MisterX1000, 2023-12-20:
> See https://github.com/Louisvdw/dbus-serialbattery/issues?q=No+module+named+%27bleak%27

Than you very much for your help Manuel,

I tried to install bleak separately, with this commands:

**opkg update
opkg install python3-misc python3-pip
pip3 install bleak**

Or do you mean something else? Or did I miss something?

Thank you very much for your support!
Best regards

Alex

mr-manuel, 2023-12-20:
You could try to install an older version of bleak like this:

https://github.com/Louisvdw/dbus-serialbattery/issues/819#issuecomment-1791392653

MisterX1000, 2023-12-20:
Hey Manuel,

unfortunately I got the same error. Log is below. 

Best regards
Alex

root@nanopi:~# **pip3 install dbus-fast==v1.93.0**
Collecting dbus-fast==v1.93.0
  Downloading dbus_fast-1.93.0.tar.gz (67 kB)
     |████████████████████████████████| 67 kB 1.5 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requiremen                                                                                               t
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_meta                                                                                               data
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-build-env-zncmbttt/overlay/lib/python3.8/site-packages/poetry/core/masonry/api.py", lin                                                                                               e 12, in <module>
    from poetry.core.factory import Factory
  File "/var/volatile/tmp/pip-build-env-zncmbttt/overlay/lib/python3.8/site-packages/poetry/core/factory.py", line 15                                                                                               , in <module>
    from poetry.core.utils.helpers import combine_unicode
  File "/var/volatile/tmp/pip-build-env-zncmbttt/overlay/lib/python3.8/site-packages/poetry/core/utils/helpers.py", l                                                                                               ine 19, in <module>
    from poetry.core.version.pep440 import PEP440Version
  File "/var/volatile/tmp/pip-build-env-zncmbttt/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/__ini                                                                                               t__.py", line 3, in <module>
    from poetry.core.version.pep440.segments import LocalSegmentType
  File "/var/volatile/tmp/pip-build-env-zncmbttt/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/segme                                                                                               nts.py", line 3, in <module>
    import dataclasses
ModuleNotFoundError: No module named 'dataclasses'

root@nanopi:~#

MisterX1000, 2023-12-20:
Now somethig has changeg, it is not better but now I get the following response from my GX:

**root@nanopi:~# pip3 install bleak**
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
    from pip._internal.build_env import get_runnable_pip
  File "/usr/lib/python3.8/site-packages/pip/_internal/build_env.py", line 19, in <module>
    from pip._internal.cli.spinners import open_spinner
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/spinners.py", line 9, in <module>
    from pip._internal.utils.logging import get_indentation
  File "/usr/lib/python3.8/site-packages/pip/_internal/utils/logging.py", line 8, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
root@nanopi:~# ^C

**root@nanopi:~# pip3 install dbus-fast==v1.93.0**
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 10, in <module>
    from pip._internal.cli.autocompletion import autocomplete
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/autocompletion.py", line 10, in <module>
    from pip._internal.cli.main_parser import create_main_parser
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/main_parser.py", line 9, in <module>
    from pip._internal.build_env import get_runnable_pip
  File "/usr/lib/python3.8/site-packages/pip/_internal/build_env.py", line 19, in <module>
    from pip._internal.cli.spinners import open_spinner
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/spinners.py", line 9, in <module>
    from pip._internal.utils.logging import get_indentation
  File "/usr/lib/python3.8/site-packages/pip/_internal/utils/logging.py", line 8, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'

mr-manuel, 2023-12-21:
Seems that something is broken in your installation. Is this a fresh install?

This should fix this error, but probably a new one arises after.

`pip3 install dataclasses`

MisterX1000, 2023-12-21:
Hey Manuel,

you are right, I also got an error message when I try to install the "dataclasses"

I absolute agree with you, seems to be a problem with the installation. The system is a Victron MP2-GX.

If there is a way to do a complete new installation I don´t know it. I only now how to update the Venus OS and the VE.bus firmware.

[image]

mr-manuel, 2023-12-21:
Google is your friend: [18. Reset to factory defaults procedure](https://www.victronenergy.com/media/pg/Cerbo-S_GX/en/reset-to-factory-defaults-procedure.html)

MisterX1000, 2023-12-21:
> Google is your friend: [18. Reset to factory defaults procedure](https://www.victronenergy.com/media/pg/Cerbo-S_GX/en/reset-to-factory-defaults-procedure.html)

I agree ;-) but I don´t have a Cerbo.

But I will figure out how to bring my MP2-GX back to factory defaults - Then I try the serialbattery BT connection with the official Venus OS.

Thanks

mr-manuel, 2023-12-21:
This is the same for all GX devices.

closed 2024-01-22 by mr-manuel