Louisvdw/dbus-serialbattery #898: nightly from dev does not start up.

cflenker, 2023-12-24:
### Describe the problem

Driver does not start up.

### Driver version

v1.1.20231223dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.00

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
; CF: MAX_CELL_VOLTAGE auf 3.55 gesetzt, damit penalty nicht zu frueh greift
;     und die CVL runterholt was zu einem starken toggeln führt
;     Die wirkliche Absobtion-Spannung wird über DVCC auf 55.2V begrenzt.
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375
;
;
;
;
; Specify in seconds how often the linear values should be recalculated
LINEAR_RECALCULATION_EVERY = 60
; Specify in percent when the linear values should be recalculated immediately
; Example: 5 for a immediate change, when the value changes by more than 5%
LINEAR_RECALCULATION_ON_PERC_CHANGE = 5

; --------- Charge Voltage limitation (affecting CVL) ---------
; Description: Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;              voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
[... 103 lines cut ...]
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.05
```

### Relevant log output

```shell
@400000006587845c360fe994 KeyError: 'CUSTOM_BATTERY_NAMES'
@4000000065878481339a3e44 Traceback (most recent call last):
@4000000065878481339a616c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 18, in <module>
@4000000065878481339a710c     from dbushelper import DbusHelper
@4000000065878481339a78dc   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 8, in <module>
@4000000065878481339be424     from utils import logger, publish_config_variables
@4000000065878481339befdc   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 237, in <module>
@4000000065878481339e8fbc     CUSTOM_BATTERY_NAMES = _get_list_from_config(
@4000000065878481339e9f5c   File "/opt/victronenergy/dbus-serialbattery/utils.py", line 30, in _get_list_from_config
@4000000065878481339eaefc     rawList = config[group][option].split(",")
@4000000065878481339ebab4   File "/usr/lib/python3.8/configparser.py", line 1254, in __getitem__
@4000000065878481339ec66c     raise KeyError(key)
@4000000065878481339f3f84 KeyError: 'CUSTOM_BATTERY_NAMES'
```

cflenker, 2023-12-24:
It was fixed when I added the following line in my config:

`CUSTOM_BATTERY_NAMES = /dev/ttyUSB0:JKBMS1_Flenker`

mr-manuel, 2023-12-24:
Thanks for reporting. Theoretically the custom names are now stored in the dbus instead of the config file. I will check tet immediately. Merry Christmas also to you :-)

mr-manuel, 2023-12-24:
Fixed

commit 8f1f16d8 (2023-12-24): Fix #898
https://github.com/Louisvdw/dbus-serialbattery/issues/898
commit 9779b2e1 (2023-12-24): Merge pull request #899 from mr-manuel/dev

Fixes #898
closed 2023-12-24 by mr-manuel