Louisvdw/dbus-serialbattery #900: JBDBMS not connecting via UART or BT after VenusOS and SerialBattery driver upgrade

petertmac, 2023-12-28:
### Describe the problem

I recently updated my VenusOS after an SD card failure. I installed the latest VenusOS release and V1.0 of SerialBattery. I have two JBDBMS connected to my Pi via USB cables. Prior to SD failure both BMS's show up in VenusOS. After update can only see one BMS. I then decided to try the BT connection on one of the BMS's. Still can only see the BMS connected via USB cable. The other BMS has the BT dongle reinstalled and I can see it in the XiaoxiangBMS app. I just installed the nightly build and have the same issue.

### Driver version

1.0.20231117dev

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.13

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 120.0

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS = LltJbd_Ble 8E:C2:30:38:C1:A4

CUSTOM_BATTERY_NAMES = /dev/ttyUSB2:Mac SmartBattery 2
```

### Relevant log output

```shell
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal
2023-12-28 01:56:47.497372500 [CHG] Device 5F:FD:B2:29:6B:5C RSSI is nil
2023-12-28 01:56:47.509752500 [CHG] Device C0:85:16:F7:01:A3 RSSI is nil
2023-12-28 01:56:47.509758500 [CHG] Device 56:1B:56:8E:E3:3A TxPower is nil
2023-12-28 01:56:47.509762500 [CHG] Device 56:1B:56:8E:E3:3A RSSI is nil
2023-12-28 01:56:47.509766500 [CHG] Device 54:B7:E5:63:5B:E8 TxPower is nil
2023-12-28 01:56:47.509770500 [CHG] Device 54:B7:E5:63:5B:E8 RSSI is nil
2023-12-28 01:56:47.509774500 [CHG] Device E1:0B:C7:2C:D8:BC RSSI is nil
2023-12-28 01:56:47.509843500 [CHG] Device 2B:98:A4:1F:E3:6E RSSI is nil
2023-12-28 01:56:47.509846500 [CHG] Device E7:1D:E2:0A:17:F9 RSSI is nil
2023-12-28 01:56:47.509850500 [CHG] Device 40:1E:FC:0F:F6:FA TxPower is nil
2023-12-28 01:56:47.509854500 [CHG] Device 40:1E:FC:0F:F6:FA RSSI is nil
2023-12-28 01:56:47.509858500 [CHG] Device A8:51:AB:CF:A3:A8 TxPower is nil
2023-12-28 01:56:47.509861500 [CHG] Device A8:51:AB:CF:A3:A8 RSSI is nil
2023-12-28 01:56:47.509889500 [CHG] Device 3E:96:5B:7A:51:45 RSSI is nil
2023-12-28 01:56:47.509893500 [CHG] Device CE:C0:8B:12:47:23 RSSI is nil
2023-12-28 01:56:47.509896500 [CHG] Device E0:9F:2A:FD:1C:AB TxPower is nil
2023-12-28 01:56:47.509900500 [CHG] Device E0:9F:2A:FD:1C:AB RSSI is nil
2023-12-28 01:56:47.509904500 [CHG] Device D8:E0:E1:2A:D7:99 RSSI is nil
2023-12-28 01:56:47.509908500 [CHG] Device 7D:F1:60:65:2E:48 TxPower is nil
2023-12-28 01:56:47.509932500 [CHG] Device 7D:F1:60:65:2E:48 RSSI is nil
2023-12-28 01:56:47.509936500 [CHG] Device 08:65:F0:2B:7E:21 RSSI is nil
2023-12-28 01:56:47.509940500 [CHG] Controller 43:45:C0:00:1F:AC Discovering: no
2023-12-28 01:56:52.616900500 Device 8E:C2:30:38:C1:A4 not available
2023-12-28 01:56:52.617909500
2023-12-28 01:56:53.768609500 INFO:SerialBattery:
2023-12-28 01:56:53.769142500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-28 01:56:53.770097500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-12-28 01:56:54.194514500 INFO:SerialBattery:Init of LltJbd_Ble at 8E:C2:30:38:C1:A4
2023-12-28 01:56:54.194972500 INFO:SerialBattery:Test of LltJbd_Ble at 8E:C2:30:38:C1:A4
2023-12-28 01:56:59.206282500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-12-28 01:56:59.208750500 ERROR:SerialBattery:No BMS found at 8E:C2:30:38:C1:A4
2023-12-28 01:56:59.209359500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-12-28 01:57:05.250377500
2023-12-28 01:57:05.250505500 INFO:Bluetooth details
```

petertmac, 2023-12-28:
After installing Ver: 1.0.20231117dev, I decided to disconnect the bluetooth dongle and reconnect the Serial Adaptor and I can now see both batteries in VenusOS. However the Bluetooth issue remains unresolved.

Dajestia, 2024-01-03:
same problem, but with VenusOs v.3.20-35

Dajestia, 2024-01-08:
2024-01-05 16:51:56.178291500 INFO: SerialBattery:
2024-01-05 16:51:56.179034500 INFO: SerialBattery:Starting dbus-serialbattery
2024-01-05 16:51:56.180286500 INFO: SerialBattery:dbus-serialbattery v1. 1.20231224dev
2024-01-05 16:51:56.445174500 Traceback (most recent call last):
2024-01-05 16:51:56.445184500
File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.
py", line 238, in <module>
2024-01-05 16:51:56.445188500
main ()
2024-01-05 16:51:56.445190500
File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.
py", line 170, in main
2024-01-05 16:51:56.445193500
2024-01-05 16:51:56.445196500
from bms. lltjbd_ble import LitJbd_Ble
# noga: F401
File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py",
line 13, in <module>
2024-01-05 16:51:56.445332500
from bleak import BleakClient, BleakScanner, BLEDevice
2024-01-05 16:51:56.445336500 ModuleNotFoundError: No module named 'bleak'
2024-01-05 16:51:56.674500500
2024-01-05 16:51:56.674952500 INFO: Bluetooth details
2024-01-05 16:51:56.762498500 Attempting to disconnect from 70: 3E: 97:08:10: 9C
2024-01-05 16:51:56.762507500 Successful disconnected
2024-01-05 16:52:36.065139500 *** CCGX booted (0) ***

mr-manuel, 2024-01-08:
Try this:

```bash
opkg update

opkg install python3-misc python3-pip

pip3 install bleak==0.21.0

export SKIP_CYTHON=false
pip3 install dbus-fast==1.94.1
```

Dajestia, 2024-01-09:
root@einstein:~# opkg update
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/einstein/Packages.gz.
Updated source 'einstein'.
root@einstein:~# opkg install python3-misc python3-pip
No packages installed or removed.
root@einstein:~# pip3 install bleak==0.21.0
Collecting bleak==0.21.0
  Downloading bleak-0.21.0-py3-none-any.whl (137 kB)
     |████████████████████████████████| 137 kB 440 kB/s 
Collecting typing-extensions>=4.7.0; python_version < "3.12"
  Using cached typing_extensions-4.9.0-py3-none-any.whl (32 kB)
Collecting dbus-fast<2.0.0,>=1.83.0; platform_system == "Linux"
  Downloading dbus_fast-1.95.2.tar.gz (67 kB)
     |████████████████████████████████| 67 kB 918 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/usr/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/usr/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/var/volatile/tmp/pip-build-env-5s_frk85/overlay/lib/python3.8/site-packages/poetry/core/masonry/api.py", line 12, in <module>
    from poetry.core.factory import Factory
  File "/var/volatile/tmp/pip-build-env-5s_frk85/overlay/lib/python3.8/site-packages/poetry/core/factory.py", line 15, in <module>
    from poetry.core.utils.helpers import combine_unicode
  File "/var/volatile/tmp/pip-build-env-5s_frk85/overlay/lib/python3.8/site-packages/poetry/core/utils/helpers.py", line 19, in <module>
    from poetry.core.version.pep440 import PEP440Version
  File "/var/volatile/tmp/pip-build-env-5s_frk85/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/__init__.py", line 3, in <module>
    from poetry.core.version.pep440.segments import LocalSegmentType
  File "/var/volatile/tmp/pip-build-env-5s_frk85/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/segments.py", line 3, in <module>
    import dataclasses
ModuleNotFoundError: No module named 'dataclasses'

root@einstein:~# opkg install python3-misc python3-pip
No packages installed or removed.
root@einstein:~# export SKIP_CYTHON=false
[... 27 lines cut ...]
    return self._call_hook('get_requires_for_build_wheel', {
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
[... cut ...]
  File "/var/volatile/tmp/pip-build-env-y_c2dprh/overlay/lib/python3.8/site-packages/poetry/core/utils/helpers.py", line 19, in <module>
    from poetry.core.version.pep440 import PEP440Version
  File "/var/volatile/tmp/pip-build-env-y_c2dprh/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/__init__.py", line 3, in <module>
    from poetry.core.version.pep440.segments import LocalSegmentType
  File "/var/volatile/tmp/pip-build-env-y_c2dprh/overlay/lib/python3.8/site-packages/poetry/core/version/pep440/segments.py", line 3, in <module>
    import dataclasses
ModuleNotFoundError: No module named 'dataclasses'

mr-manuel, 2024-01-09:
You could try to install all modules which are not found with `pip install modulename`, but normally this looks like a broken system. At least it was in other issues.

Dajestia, 2024-01-11:
returning the fw 3.13 works all right

AlexdeKairos, 2024-01-24:
Same here. Any of the 3.20 betas work. Same installation errors on 3.20 but 3.12, 3.13 and the new 3.14 works perfect for me on 1.0.20231117dev. 

Something wrong with Bluetooth connections on the new 3.20 betas.

coca7, 2024-01-24:
I can't connect the JBDBMS by bluetooth either.  I tried last week, and had more success than in the past but still could not connect.  I also found it odd that I could see the bms bluetooth using bluetoothctl in Venus OS, but not on my Mac (bluetoothutil on Mac) or linux machine.  However, I can connect to the BMS on Mac through the Smart BMS app.  I reconnected all my BMS on USB, but I have attacked the log from one of them.

 GNU nano 4.9.3                       current                                  
@4000000065b15fe423c36b1c
@4000000065b15fe423c9e744 INFO:Bluetooth details
@4000000065b15fe4288bd454 Device E0:9F:2A:E4:B0:F1 not available
@4000000065b15fe92ff71744 Device E0:9F:2A:E4:B0:F1 not available
@4000000065b15fe930145f5c
@4000000065b15feb38a073cc INFO:SerialBattery:
@4000000065b15feb38ab02ec INFO:SerialBattery:Starting dbus-serialbattery
@4000000065b15feb38be0234 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
@4000000065b15fec11bb5844 Traceback (most recent call last):
@4000000065b15fec11bb7b6c   File "/opt/victronenergy/dbus-serialbattery/dbus-se>
@4000000065b15fec11bb8b0c     main()
@4000000065b15fec11bb8ef4   File "/opt/victronenergy/dbus-serialbattery/dbus-se>
@4000000065b15fec11bb9e94     from bms.lltjbd_ble import LltJbd_Ble  # noqa: F4>
@4000000065b15fec11bba664   File "/opt/victronenergy/dbus-serialbattery/bms/llt>
@4000000065b15fec11bdf43c     from bleak import BleakClient, BleakScanner, BLED>
@4000000065b15fec11be03dc ModuleNotFoundError: No module named 'bleak'
@4000000065b15fec22afb8fc
@4000000065b15fec22f5f18c INFO:Bluetooth details
@4000000065b15fec2f8788ac Device E0:9F:2A:E4:B0:F1 not available
@4000000065b15ff2023be44c Device E0:9F:2A:E4:B0:F1 not available

mr-manuel, 2024-01-25:
@idstein have you the same problem with Venus OS `3.20~*`?

mr-manuel, 2024-02-16:
Any update from you guys that have problems? @petertmac @Dajestia @AlexdeKairos @coca7

petertmac, 2024-02-16:
I stayed with the wired connections which work. It is difficult to swap the cables due to BMS locations, so have left it for the moment. One battery I can give a custom name. The other doesn’t give the option in the GUI.

mr-manuel, 2024-02-16:
> One battery I can give a custom name. The other doesn’t give the option in the GUI.

Without further informations is hard to help. Please open a new issue for that.

closed 2024-02-23 by mr-manuel