Louisvdw/dbus-serialbattery #906: Wrong SOC Charge Limitation

mdkeil, 2024-01-04:
### Describe the bug

Battery SOC > 98% but CCL @ 37A (Charge Limitation SoC); CCL of 15A is expected.  

### How to reproduce

nothing special

### Expected behavior

CCL of 15A

### Driver version

1.0.20240102beta

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.20~35

### BMS type

Seplos

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE  
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

CVL_ICONTROLLER_MODE = True
LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True                                                                                

BMS_TYPE = Seplos

EXCLUDED_DEVICES = /dev/ttyUSB0

CC_SOC_LIMIT1 = 98
CC_SOC_LIMIT2 = 95
CC_SOC_LIMIT3 = 91

CC_CURRENT_LIMIT1_FRACTION = 0.15
CC_CURRENT_LIMIT2_FRACTION = 0.35
CC_CURRENT_LIMIT3_FRACTION = 0.50
```

### Relevant log output

[... 95 lines cut ...]
2024-01-04 11:52:06.492455500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 11:52:06.493148500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 11:52:06.493853500 INFO:SerialBattery:========== Settings ==========
2024-01-04 11:52:06.494517500 INFO:SerialBattery:> Connection voltage: 54.89V | Current: 0.54A | SoC: None%
2024-01-04 11:52:06.495200500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 11:52:06.495794500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 11:52:06.496553500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 11:52:06.497154500 INFO:SerialBattery:> CVCM:     True
2024-01-04 11:52:06.497829500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 11:52:06.498439500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 11:52:06.499182500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 11:52:06.500033500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 11:52:06.500819500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 11:52:06.981223500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 11:52:07.238327500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 11:52:07.239097500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 11:52:07.240032500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 11:52:07.279239500 INFO:SerialBattery:publish config values = True
```

### Any other information that may be helpful

`2024-01-04 11:52:06.494517500 INFO:SerialBattery:> Connection voltage: 54.89V | Current: 0.54A | SoC: None%`

SoC: None% could be a hint, but in venusOS the correct SOC is displayed without problems.

mdkeil, 2024-01-04:
I think there is a shift error in the CC_SOC_LIMIT/CC_CURRENT_LIMIT Array, because with exactly 98.0% SOC I get a CCL of 35A.

mr-manuel, 2024-01-04:
How many batteries do you have?

mdkeil, 2024-01-04:
only one.

mr-manuel, 2024-01-04:
With an older version it worked?

mdkeil, 2024-01-04:
Yeah.  I think an older dev-version from november.. didn't know exactly. Next time the SOC is > 90 i will take a look.

mr-manuel, 2024-01-04:
You can try a release from here: https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases

mdkeil, 2024-01-04:
I think, this was the last working version for me:

`v1.0.20231128beta`

```
2024-01-04 18:02:37.868565500 INFO:SerialBattery:
2024-01-04 18:02:37.869444500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 18:02:37.871675500 INFO:SerialBattery:dbus-serialbattery v1.0.20231128beta
2024-01-04 18:02:53.880876500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 18:02:53.881646500 INFO:SerialBattery:Testing Seplos
2024-01-04 18:02:56.909665500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('ascii', b'\xe0\x00\x00\x00', 0, 1, 'ordinal not in range(128)'), type(err)=<class 'UnicodeDecodeError'>
2024-01-04 18:02:56.910609500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-04 18:02:57.412337500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-01-04 18:02:57.413091500 INFO:SerialBattery:Testing Seplos
2024-01-04 18:02:58.538618500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 18:02:58.539334500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 18:02:58.540112500 INFO:SerialBattery:========== Settings ==========
2024-01-04 18:02:58.540821500 INFO:SerialBattery:> Connection voltage: 52.62V | Current: 0.57A | SoC: 58.2%
2024-01-04 18:02:58.541421500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 18:02:58.542034500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 18:02:58.542690500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 18:02:58.543282500 INFO:SerialBattery:> CVCM:     True
2024-01-04 18:02:58.543924500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 18:02:58.544527500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 18:02:58.545143500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 18:02:58.545730500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 18:02:58.546482500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 18:02:59.045005500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 18:02:59.271011500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 18:02:59.271796500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 18:02:59.272486500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 18:02:59.310166500 INFO:SerialBattery:publish config values = 1
2024-01-04 18:03:05.463272500 WARNING:SerialBattery:checksum error
2024-01-04 18:03:06.586922500 Traceback (most recent call last):
2024-01-04 18:03:06.586934500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 601, in publish_battery
2024-01-04 18:03:06.586940500     result = self.battery.refresh_data()
2024-01-04 18:03:06.586943500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 118, in refresh_data
2024-01-04 18:03:06.586949500     result_alarm = self.read_alarm_data()
2024-01-04 18:03:06.586953500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 132, in read_alarm_data
2024-01-04 18:03:06.587119500     data = self.read_serial_data_seplos(
2024-01-04 18:03:06.587124500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 302, in read_serial_data_seplos
2024-01-04 18:03:06.587130500     if not Seplos.is_valid_frame(data):
2024-01-04 18:03:06.587134500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 276, in is_valid_frame
2024-01-04 18:03:06.587140500     if chksum != Seplos.int_from_2byte_hex_ascii(data, -5):
2024-01-04 18:03:06.587184500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 32, in int_from_2byte_hex_ascii
2024-01-04 18:03:06.587191500     bytes.fromhex(data[offset : offset + 4].decode("ascii")),
2024-01-04 18:03:06.587196500 UnicodeDecodeError: 'ascii' codec can't decode byte 0xcc in position 0: ordinal not in range(128)
2024-01-04 18:03:06.587202500 Setted setting /Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/LastSeen to 1704391379
2024-01-04 18:03:08.772023500 INFO:SerialBattery:
2024-01-04 18:03:08.777528500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 18:03:08.779276500 INFO:SerialBattery:dbus-serialbattery v1.0.20231128beta
2024-01-04 18:03:24.802866500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 18:03:24.802878500 INFO:SerialBattery:Testing Seplos
2024-01-04 18:03:25.938772500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 18:03:25.939479500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 18:03:25.940335500 INFO:SerialBattery:========== Settings ==========
2024-01-04 18:03:25.941026500 INFO:SerialBattery:> Connection voltage: 52.64V | Current: 0.0A | SoC: 58.2%
2024-01-04 18:03:25.941625500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 18:03:25.942228500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 18:03:25.942837500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 18:03:25.943423500 INFO:SerialBattery:> CVCM:     True
2024-01-04 18:03:25.944032500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 18:03:25.944626500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 18:03:25.945242500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 18:03:25.945839500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 18:03:25.946559500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 18:03:26.438649500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 18:03:26.677842500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 18:03:26.678920500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 18:03:26.680036500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 18:03:26.716264500 INFO:SerialBattery:publish config values = 1
```

mr-manuel, 2024-01-04:
With `v1.0.20231217beta` it did not work?

mdkeil, 2024-01-04:
I think it will not work. Same "SOC Error".. will see it next charge.

```
2024-01-04 21:05:33.234559500 INFO:SerialBattery:
2024-01-04 21:05:33.235212500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 21:05:33.236854500 INFO:SerialBattery:dbus-serialbattery v1.1.20231224beta
2024-01-04 21:05:49.250875500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 21:05:49.251672500 INFO:SerialBattery:Testing Seplos
2024-01-04 21:05:50.377396500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 21:05:50.378106500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 21:05:50.378723500 INFO:SerialBattery:========== Settings ==========
2024-01-04 21:05:50.379404500 INFO:SerialBattery:> Connection voltage: 52.45V | Current: 0.67A | SoC: None%
2024-01-04 21:05:50.380237500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 21:05:50.380872500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 21:05:50.381502500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 21:05:50.382105500 INFO:SerialBattery:> CVCM:     True
2024-01-04 21:05:50.382729500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 21:05:50.383342500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 21:05:50.383930500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 21:05:50.384516500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 21:05:50.385244500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 21:05:50.843461500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 21:05:51.176702500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 21:05:51.180517500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 21:05:51.180705500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 21:05:51.257537500 INFO:SerialBattery:publish config values = True
2024-01-04 21:05:56.885516500 Traceback (most recent call last):
2024-01-04 21:05:56.885527500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 500, in read
2024-01-04 21:05:56.885533500     raise SerialException(
2024-01-04 21:05:56.885536500 serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-01-04 21:05:56.885546500
2024-01-04 21:05:56.885548500 During handling of the above exception, another exception occurred:
2024-01-04 21:05:56.885553500
2024-01-04 21:05:56.885757500 Traceback (most recent call last):
2024-01-04 21:05:56.885761500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 626, in publish_battery
2024-01-04 21:05:56.885767500     result = self.battery.refresh_data()
2024-01-04 21:05:56.885771500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 117, in refresh_data
2024-01-04 21:05:56.885776500     result_status = self.read_status_data()
2024-01-04 21:05:56.885780500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 199, in read_status_data
2024-01-04 21:05:56.885832500     data = self.read_serial_data_seplos(
2024-01-04 21:05:56.885836500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 300, in read_serial_data_seplos
2024-01-04 21:05:56.885843500     data = ser.readline()
2024-01-04 21:05:56.885846500   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 509, in read
2024-01-04 21:05:56.885851500     raise SerialException('read failed: {}'.format(e))
2024-01-04 21:05:56.885937500 serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-01-04 21:05:56.885946500 Setted setting /Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/LastSeen to 1704402350
2024-01-04 21:05:59.021904500 INFO:SerialBattery:
2024-01-04 21:05:59.022656500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 21:05:59.031204500 INFO:SerialBattery:dbus-serialbattery v1.1.20231224beta
2024-01-04 21:06:15.041388500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 21:06:15.042061500 INFO:SerialBattery:Testing Seplos
2024-01-04 21:06:16.171153500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 21:06:16.171872500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 21:06:16.172459500 INFO:SerialBattery:========== Settings ==========
2024-01-04 21:06:16.173108500 INFO:SerialBattery:> Connection voltage: 52.49V | Current: 0.98A | SoC: None%
2024-01-04 21:06:16.173691500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 21:06:16.174276500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 21:06:16.174886500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 21:06:16.175463500 INFO:SerialBattery:> CVCM:     True
2024-01-04 21:06:16.176068500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 21:06:16.176670500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 21:06:16.177263500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 21:06:16.178457500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 21:06:16.179238500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 21:06:16.609199500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 21:06:16.899295500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 21:06:16.900265500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 21:06:16.900982500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 21:06:16.938303500 INFO:SerialBattery:publish config values = True
```

mr-manuel, 2024-01-04:
Just for reference:

https://github.com/mr-manuel/venus-os_dbus-serialbattery/compare/v1.0.20231128beta...v1.0.20231217beta

https://github.com/mr-manuel/venus-os_dbus-serialbattery/compare/v1.0.20231217beta...v1.1.20231224beta

mdkeil, 2024-01-04:
Log from `v1.0.20231217beta`

Should work..

```
2024-01-04 22:01:54.322299500 INFO:SerialBattery:
2024-01-04 22:01:54.323790500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 22:01:54.325979500 INFO:SerialBattery:dbus-serialbattery v1.0.20231217beta
2024-01-04 22:02:10.346013500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 22:02:10.346023500 INFO:SerialBattery:Testing Seplos
2024-01-04 22:02:12.693462500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('ascii', b'\x00\xe0\xe0\xe0', 1, 2, 'ordinal not in range(128)'), type(err)=<class 'UnicodeDecodeError'>
2024-01-04 22:02:12.694263500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-04 22:02:13.196085500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-01-04 22:02:13.196850500 INFO:SerialBattery:Testing Seplos
2024-01-04 22:02:14.321522500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 22:02:14.322276500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 22:02:14.322910500 INFO:SerialBattery:========== Settings ==========
2024-01-04 22:02:14.323628500 INFO:SerialBattery:> Connection voltage: 52.35V | Current: 0.0A | SoC: 37.4%
2024-01-04 22:02:14.324247500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 22:02:14.324877500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 22:02:14.325523500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 22:02:14.326138500 INFO:SerialBattery:> CVCM:     True
2024-01-04 22:02:14.326775500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 22:02:14.327408500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 22:02:14.328024500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 22:02:14.328634500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 22:02:14.329368500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 22:02:14.809855500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 22:02:15.083052500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 22:02:15.083831500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 22:02:15.084540500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 22:02:15.123049500 INFO:SerialBattery:publish config values = 1
2024-01-04 22:02:21.278410500 WARNING:SerialBattery:checksum error
2024-01-04 22:02:22.388202500 Traceback (most recent call last):
2024-01-04 22:02:22.388214500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 601, in publish_battery
2024-01-04 22:02:22.388220500     result = self.battery.refresh_data()
2024-01-04 22:02:22.388224500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 118, in refresh_data
2024-01-04 22:02:22.388230500     result_alarm = self.read_alarm_data()
2024-01-04 22:02:22.388233500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 132, in read_alarm_data
2024-01-04 22:02:22.388408500     data = self.read_serial_data_seplos(
2024-01-04 22:02:22.388415500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 302, in read_serial_data_seplos
2024-01-04 22:02:22.388421500     if not Seplos.is_valid_frame(data):
2024-01-04 22:02:22.388425500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 276, in is_valid_frame
2024-01-04 22:02:22.388431500     if chksum != Seplos.int_from_2byte_hex_ascii(data, -5):
2024-01-04 22:02:22.388482500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 32, in int_from_2byte_hex_ascii
2024-01-04 22:02:22.388490500     bytes.fromhex(data[offset : offset + 4].decode("ascii")),
2024-01-04 22:02:22.388494500 UnicodeDecodeError: 'ascii' codec can't decode byte 0xcc in position 0: ordinal not in range(128)
2024-01-04 22:02:22.388500500 Setted setting /Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/LastSeen to 1704405734
2024-01-04 22:02:24.585990500 INFO:SerialBattery:
2024-01-04 22:02:24.587017500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-04 22:02:24.588694500 INFO:SerialBattery:dbus-serialbattery v1.0.20231217beta
2024-01-04 22:02:40.608989500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-04 22:02:40.609000500 INFO:SerialBattery:Testing Seplos
2024-01-04 22:02:41.730710500 INFO:SerialBattery:Connection established to Seplos
2024-01-04 22:02:41.731430500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB1
2024-01-04 22:02:41.732031500 INFO:SerialBattery:========== Settings ==========
2024-01-04 22:02:41.732712500 INFO:SerialBattery:> Connection voltage: 52.38V | Current: 0.0A | SoC: 37.4%
2024-01-04 22:02:41.733306500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-04 22:02:41.733912500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-04 22:02:41.734536500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-04 22:02:41.735126500 INFO:SerialBattery:> CVCM:     True
2024-01-04 22:02:41.735751500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-04 22:02:41.736351500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-04 22:02:41.736944500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-04 22:02:41.737533500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-04 22:02:41.738259500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-04 22:02:42.187065500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-04 22:02:42.426706500 INFO:SerialBattery:DeviceInstance = 1
2024-01-04 22:02:42.427942500 INFO:SerialBattery:Used device instances: ['1']
2024-01-04 22:02:42.428880500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-04 22:02:42.465314500 INFO:SerialBattery:publish config values = 1

```

mr-manuel, 2024-01-05:
@cflenker could you check, if that has to do with #868?

mdkeil, 2024-01-09:
I'm very confused-- trying out all you beta-version, but nothing works as expected..?! always the same wrong behaviour.. where do you save persistent data?-- I will try to make a clean.

mr-manuel, 2024-01-09:
Persistent data is saved on dbus `com.victronenergy.settings/Settings/Devices`. You can browse it with the `dbus-spy` command.

mdkeil, 2024-01-09:
..ok, nothing special, seems to be ok. But I have no idea, what I could still check or what caused this behaviour ;(
I think my workaround will be to disable CCCM_SOC.

mr-manuel, 2024-01-09:
You only have to be patient that we can check this. It will take some time.

mdkeil, 2024-01-10:
for todays testing I have disabled linear limitation.

mdkeil, 2024-01-10:
SOC_DCL with `LINEAR_LIMITATION=False` looks ok.
>=20% SOC - 50A
<20% SOC - 30A

mdkeil, 2024-01-10:
Update: correct SOC CC-Limitation with `LINEAR_LIMITATION=False`
SOC <= 91 - MAX_Charge_Current
91 < SOC <=95  : 50 A
95 < SOC <=98  : 35 A
98 < SOC : 15 A

mr-manuel, 2024-01-11:
Could you please post the CVL, CCL and DCL graph from the VRM dashboard and one where we can see the SOC pf the same time period.

mdkeil, 2024-01-11:
from day before yesterday with `LINEAR_LIMITATION=true` // some data is missing because I tested all beta versions
[export_SOC_CCL_DCL_CVL_LINEAR_LIMITATION_True.xlsx](https://github.com/Louisvdw/dbus-serialbattery/files/13898422/export_SOC_CCL_DCL_CVL_LINEAR_LIMITATION_True.xlsx)

from yesterday with `LINEAR_LIMITATION=False`
[export_SOC_CCL_DCL_CVL_LINEAR_LIMITATION_False.xlsx](https://github.com/Louisvdw/dbus-serialbattery/files/13898421/export_SOC_CCL_DCL_CVL_LINEAR_LIMITATION_False.xlsx)

mr-manuel, 2024-01-11:
Please provide the graph from the VRM, it's easier to understand watching a graph then raw numbers.

mdkeil, 2024-01-11:
> Please provide the graph from the VRM, it's easier to understand watching a graph then raw numbers.

you can easily make graphs from that raw data ;)

10.01.2024 / yesterday / `LINEAR_LIMITATION=False`
[image]

04.01.2024 / `LINEAR_LIMITATION=True`
[image]

09.01.2024 / `LINEAR_LIMITATION=True` // Testing all beta versions
[image]

mdkeil, 2024-02-07:
> I think there is a shift error in the CC_SOC_LIMIT/CC_CURRENT_LIMIT Array, because with exactly 98.0% SOC I get a CCL of 35A.

 - with SOC 95% I get a CCL of 50A

mr-manuel, 2024-02-07:
> any news?

Unfortunately not. I have to wait for days with more sun and try to reproduce the issue.

mr-manuel, 2024-02-16:
I tried different settings but found no issue. Have you two batteries connected?

Please provide a screenshot of the device list and of the `Parameters` page of the dbus-serialbattery.

mdkeil, 2024-02-16:
[image]

[image]
[image]
[image]

[image]
[image]
[image]

mr-manuel, 2024-02-19:
I reworked this part. Please try my latest `beta` or `dev` build. Remember to check the [changelog](https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/v1.2.20240219beta/CHANGELOG.md) first, since there are breaking changes.

mdkeil, 2024-02-19:
thxs! I will check the CCL with your latest beta after we get some sunshine.. 

```
SOC_WHILE_CHARGING                 =  100,   98,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.00, 0.15, 0.30, 0.45, 1.00
```

mdkeil, 2024-02-24:
Problem is solved for me now! ;) Thxs!

closed 2024-02-24 by mr-manuel