Louisvdw/dbus-serialbattery #907: Make soc_reset_last_reached persistent

jdeus, 2024-01-05:
### Is your feature request related to a problem? Please describe.

It would be appreciated if soc_reset_last_reached could be saved to survive a restart of the GX device or a driver restart, as commented https://github.com/Louisvdw/dbus-serialbattery/blob/dev/etc/dbus-serialbattery/battery.py#L496

At the moment, this operation is performed way too many times.

mr-manuel, 2024-01-05:
What do you mean exactly with?

> At the moment, this operation is performed way too many times.

mr-manuel, 2024-01-05:
This is already implemented. I only forgot to remove the comment.

https://github.com/Louisvdw/dbus-serialbattery/blob/9779b2e1773aa668afca9d1f647636c773f11bcd/etc/dbus-serialbattery/dbushelper.py#L1147-L1166

closed 2024-01-05 by mr-manuel