Louisvdw/dbus-serialbattery #908: Coordinate SOC reset voltage in a multi-batteries system

jdeus, 2024-01-05:
### Is your feature request related to a problem? Please describe.

At the moment, if using the few proposed solutions to aggregate the batteries, the "SOC reset voltage" procedure will only be conducted when both batteries require it at the same time. But the main issue is the voltage won't be held long enough for both batteries to high balance properly, as it will stop when the first battery goes into float, which will leave the second battery stuck on "SOC reset voltage" with a high voltage, which is undesirable.

So until a solution is implemented, the SOC reset voltage feature is unsafe to use when a system has more than one battery bank connected.

### Describe the solution you'd like

One solution would be to implement a flag on the dbus for each dbus-serialbattery instance called "ready to SOC reset". The aggregator will wait until all batteries (each using dbus-serialbattery) request this flag, and only then it will allow both to enter the SOC reset state.
All dbus-serialbattery would then each enter the state, and flag "done" on the battery when the "SOC reset" is done. The coordinator will wait until all battery banks report the "done" flag to stop the SOC reset process (possibly by setting another flag on each dbus-serialbattery instance on the dbus)

### Additional context

I consider this as essential. Let me know if a bounty would prioritize this feature

mr-manuel, 2024-01-05:
This is not true. Only one BMS can change CVL, CCL and DCL.

Please see also https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/52. The aggregator has to di this.

closed 2024-01-05 by mr-manuel

jdeus, 2024-01-05:
Of course only one BMS can change these values. But I was talking in an aggregated system

mr-manuel, 2024-01-06:
In an aggregated system the aggregator has to be selected as the "Controlling BMS" and it has to hold the voltage until all batteries are balanced. 

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/52#issuecomment-1784154222

> the new feature can be activated by settin KEEP_MAX_CVL = True. Please review (search this parameter in aggregatebatteries.py) and test.

If you set this, then all batteries should be balanced well. If not then you have to open an issue in the aggregator repository ;-)

jdeus, 2024-01-06:
Looks like KEEP_MAX_CVL is only referenced as a comment, so it's not a GA feature, which is untested and thus not usable.

mr-manuel, 2024-01-06:
If nobody will test it, then it will never be available.