Louisvdw/dbus-serialbattery #910: Percentage in Dashboard not shown

maxpd1, 2024-01-08:
### Describe the bug

I see the percenage of the battery in the serialbattery menu of the devices but not in the standard overview on the battery icon. I did several restarts. With luck it appears but disappears again after next restart.

[image]

### How to reproduce

I can't tell how it happens. 
It doesn't seem to happen during a specific SOC.

### Expected behavior

Percentage always shown in Dashboard

### Driver version

1.0.20231125

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.13

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 65.0
MAX_BATTERY_DISCHARGE_CURRENT = 75.0

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = True
; Discharge current control management referring to cell-voltage enable (True/False).
DCCM_CV_ENABLE = True
; The current will be changed linear between those steps if 
LINEAR_LIMITATION_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.35, 3.15, 3.0, 2.7
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.5,    1, 1, 0.6, 0.1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1

TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  15, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.4, 1, 1,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20, -10,   0,   5,  10, 15, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.3, 0.5, 1, 1,  1,  1,  0
```

### Relevant log output

```shell
Sorry, not sure which one you need from the manual linked above.
```

mr-manuel, 2024-01-08:
Did you first search in the existing issues before you opened one?

https://github.com/Louisvdw/dbus-serialbattery/issues/866

maxpd1, 2024-01-08:
Indeed I did, but I seaches for percentage, not SOC :(

closed 2024-01-08 by maxpd1