Louisvdw/dbus-serialbattery #911: Wrong discharge limitation shown

maxpd1, 2024-01-08:
### Describe the bug

Discharge Limitation is set to 0.1 C, so 7.5A when under 10% in default Config.
This value is shown in parameters

[image]

But I had 7% left and it took out 65 A

[image]

Honestly my config isn't perfect yet, as I entered Discharge Limitation based on Voltage but did not deactivate Limitation based on SOC. Therefore the parameter still shows Discharge Limitation: SOC. But also the voltage limitations wouldn't accept 65A under 10%.

I used Venus 3.12
Raspberry connected via RS485 to a Seplos BMS with serialbattery as batterymonitor.
ESS mode: Optimized without battery life
DVCC activated

### How to reproduce

Not quite sure because normale the Discharge limitation is taken into account. 
This was the first time I saw this effect.

### Expected behavior

If discharge limitation is shown as SOC with 7.5A in parameters it should be a reliable value.

### Driver version

1.0.20231125

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.12

### BMS type

Seplos

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 65.0
MAX_BATTERY_DISCHARGE_CURRENT = 75.0

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = True
; Discharge current control management referring to cell-voltage enable (True/False).
DCCM_CV_ENABLE = True
; The current will be changed linear between those steps if 
LINEAR_LIMITATION_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.35, 3.15, 3.0, 2.7
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.5,    1, 1, 0.6, 0.1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1

TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  15, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.4, 1, 1,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20, -10,   0,   5,  10, 15, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.3, 0.5, 1, 1,  1,  1,  0
```

### Relevant log output

```shell
Please support me in this step.
```

maxpd1, 2024-01-08:
As we found out in https://github.com/Louisvdw/dbus-serialbattery/issues/910 
Settings -> DVCC -> Controlling BMS
Settings -> System Setup -> Battery monitor
were empty. May this cause this wrong behaviour?

mdkeil, 2024-01-09:
> Battery monitor were empty. May this cause this wrong behaviour?

thats your problem.

mr-manuel, 2024-01-09:
> Settings -> DVCC -> Controlling BMS

1. What are your settings here?
2. Did you enable DVCC and is the BMS selected as controlling BMS?

If not that's the problem.

maxpd1, 2024-01-09:
> > Settings -> DVCC -> Controlling BMS
> 
> 1. What are your settings here?

It was empty. Now its serial battery again.

> 2. Did you enable DVCC and is the BMS selected as controlling BMS?
> 
> If not that's the problem.

It wasn't now it is again.
I assume the problem is either if I turn of Victron/Seplos battery or if I plug out the USB cables that this is causing the removing of the Controlling BMS as well as the Battery Monitor during the next restart.

closed 2024-01-09 by maxpd1