Louisvdw/dbus-serialbattery #912: Jbd BMS not recognized after update

ToxSox, 2024-01-09:
### Describe the problem

My old sd card is broken. On that was venus os 2.9x and an older serialbattery version.
I got a new one, installed the newest venus os version and the current serialbattery version.
But it not regognized my Jbd BMS.

I get the following error in the logfile:
`2024-01-09 20:20:44.129263500 INFO:SerialBattery:Testing LltJbd
2024-01-09 20:20:44.322643500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>`

I have tried the latest, the master and the dev branch.

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.13

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
2024-01-09 20:24:57.543089500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-09 20:24:57.544049500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-09 20:24:57.544513500 INFO:SerialBattery:Testing Daly
2024-01-09 20:24:57.671728500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:24:57.672246500 INFO:SerialBattery:Testing Daly
2024-01-09 20:24:57.798413500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:24:57.798914500 INFO:SerialBattery:Testing Ecs
2024-01-09 20:24:57.858891500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:24:57.859382500 INFO:SerialBattery:Testing HeltecModbus
2024-01-09 20:25:01.547166500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:01.547659500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-09 20:25:06.058437500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:06.058900500 INFO:SerialBattery:Testing Jkbms
2024-01-09 20:25:06.326832500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:06.328176500 INFO:SerialBattery:Testing Lifepower
2024-01-09 20:25:06.607005500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:06.608417500 INFO:SerialBattery:Testing LltJbd
2024-01-09 20:25:06.788824500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-09 20:25:06.789428500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:07.063441500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:07.064958500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:07.336697500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:07.338198500 INFO:SerialBattery:Testing Seplos
2024-01-09 20:25:08.345604500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:08.847814500 INFO:SerialBattery:Testing Daly
2024-01-09 20:25:08.973532500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:08.974189500 INFO:SerialBattery:Testing Daly
2024-01-09 20:25:09.100268500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:09.100929500 INFO:SerialBattery:Testing Ecs
2024-01-09 20:25:09.512645500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:09.513433500 INFO:SerialBattery:Testing HeltecModbus
2024-01-09 20:25:13.201358500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:13.201365500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-09 20:25:17.713407500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:17.713415500 INFO:SerialBattery:Testing Jkbms
2024-01-09 20:25:17.986537500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:17.988067500 INFO:SerialBattery:Testing Lifepower
2024-01-09 20:25:18.257986500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:18.259509500 INFO:SerialBattery:Testing LltJbd
2024-01-09 20:25:18.546890500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-09 20:25:18.547553500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:18.815437500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:18.816989500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:19.089844500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:19.091340500 INFO:SerialBattery:Testing Seplos
2024-01-09 20:25:20.097993500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:20.599441500 INFO:SerialBattery:Testing Daly
2024-01-09 20:25:20.726526500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:20.727165500 INFO:SerialBattery:Testing Daly
2024-01-09 20:25:20.856373500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:20.856864500 INFO:SerialBattery:Testing Ecs
2024-01-09 20:25:21.266984500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:21.267437500 INFO:SerialBattery:Testing HeltecModbus
2024-01-09 20:25:24.945790500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:24.945798500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-09 20:25:29.467886500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:29.468401500 INFO:SerialBattery:Testing Jkbms
2024-01-09 20:25:29.737474500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:29.738837500 INFO:SerialBattery:Testing Lifepower
2024-01-09 20:25:30.013663500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:30.015103500 INFO:SerialBattery:Testing LltJbd
2024-01-09 20:25:30.205006500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-09 20:25:30.205485500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:30.482124500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:30.483440500 INFO:SerialBattery:Testing Renogy
2024-01-09 20:25:30.752988500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:30.754465500 INFO:SerialBattery:Testing Seplos
2024-01-09 20:25:31.761719500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-09 20:25:32.263026500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-01-09 20:25:32.263218500 
2024-01-09 20:25:32.263220500 #0
2024-01-09 20:25:32.263221500 #1
2024-01-09 20:25:32.263222500 #2
2024-01-09 20:25:32.263224500 #3
2024-01-09 20:25:32.263225500 #4
2024-01-09 20:25:32.263226500 #5
2024-01-09 20:25:32.263227500 #6
2024-01-09 20:25:32.263229500 #7
2024-01-09 20:25:32.263230500 #8
2024-01-09 20:25:32.263231500 #9
2024-01-09 20:25:32.263233500 #10
2024-01-09 20:25:32.263234500 #11
2024-01-09 20:25:32.263235500 #12
2024-01-09 20:25:32.263236500 #13
2024-01-09 20:25:32.263238500 #14
2024-01-09 20:25:32.263239500 #15
2024-01-09 20:25:32.263240500 #0
2024-01-09 20:25:32.263270500 #1
2024-01-09 20:25:32.263272500 #2
2024-01-09 20:25:32.263273500 #3
2024-01-09 20:25:32.263274500 #4
2024-01-09 20:25:32.263275500 #5
2024-01-09 20:25:32.263277500 #6
2024-01-09 20:25:32.263278500 #7
2024-01-09 20:25:32.263279500 #8
2024-01-09 20:25:32.263280500 #9
2024-01-09 20:25:32.263282500 #10
2024-01-09 20:25:32.263283500 #11
2024-01-09 20:25:32.263284500 #12
2024-01-09 20:25:32.263286500 #13
2024-01-09 20:25:32.263287500 #14
2024-01-09 20:25:32.263288500 #15
2024-01-09 20:25:32.263290500 #0
2024-01-09 20:25:32.263291500 #1
2024-01-09 20:25:32.263315500 #2
2024-01-09 20:25:32.263317500 #3
2024-01-09 20:25:32.263318500 #4
2024-01-09 20:25:32.263319500 #5
2024-01-09 20:25:32.263321500 #6
2024-01-09 20:25:32.263322500 #7
2024-01-09 20:25:32.263323500 #8
2024-01-09 20:25:32.263324500 #9
2024-01-09 20:25:32.263326500 #10
2024-01-09 20:25:32.263327500 #11
2024-01-09 20:25:32.263328500 #12
2024-01-09 20:25:32.263330500 #13
2024-01-09 20:25:32.263331500 #14
2024-01-09 20:25:32.263332500 #15
```

mr-manuel, 2024-01-09:
Please use the latest nightly of the dev branch. The logs should look different.

ToxSox, 2024-01-10:
Here the log from the dev branch:
`2024-01-10 20:20:50.976349500 INFO:SerialBattery:
2024-01-10 20:20:50.976691500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-10 20:20:50.981996500 INFO:SerialBattery:dbus-serialbattery v1.1.20231224dev
2024-01-10 20:21:06.990284500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-10 20:21:06.990844500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-01-10 20:21:07.117791500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:07.118418500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-01-10 20:21:07.245165500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:07.245627500 INFO:SerialBattery:Testing Ecs
2024-01-10 20:21:07.305506500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:07.305965500 INFO:SerialBattery:Testing HeltecModbus
2024-01-10 20:21:10.632821500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:10.633244500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-10 20:21:11.462205500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:11.462212500 INFO:SerialBattery:Testing Jkbms
2024-01-10 20:21:11.733776500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:11.735690500 INFO:SerialBattery:Testing Lifepower
2024-01-10 20:21:12.006117500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:12.007654500 INFO:SerialBattery:Testing LltJbd
2024-01-10 20:21:12.335942500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-10 20:21:12.336613500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-01-10 20:21:12.609394500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:12.610904500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-01-10 20:21:12.883385500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:12.884963500 INFO:SerialBattery:Testing Seplos
2024-01-10 20:21:13.892524500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:14.394735500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-01-10 20:21:14.394742500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-01-10 20:21:14.520624500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:14.521202500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-01-10 20:21:14.647496500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:14.647986500 INFO:SerialBattery:Testing Ecs
2024-01-10 20:21:15.075842500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:15.075849500 INFO:SerialBattery:Testing HeltecModbus
2024-01-10 20:21:18.770706500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:18.770713500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-10 20:21:19.580066500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:19.580073500 INFO:SerialBattery:Testing Jkbms
2024-01-10 20:21:19.850730500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:19.852493500 INFO:SerialBattery:Testing Lifepower
2024-01-10 20:21:20.123728500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:20.125086500 INFO:SerialBattery:Testing LltJbd
2024-01-10 20:21:20.557683500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-10 20:21:20.558193500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-01-10 20:21:20.838465500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:20.839978500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2024-01-10 20:21:21.108918500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:21.110300500 INFO:SerialBattery:Testing Seplos
2024-01-10 20:21:22.117811500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:22.619180500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-01-10 20:21:22.619864500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-01-10 20:21:22.746395500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:22.747122500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-01-10 20:21:22.873548500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:22.874044500 INFO:SerialBattery:Testing Ecs
2024-01-10 20:21:23.283960500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:23.284397500 INFO:SerialBattery:Testing HeltecModbus
2024-01-10 20:21:26.966073500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:26.966523500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-10 20:21:27.774594500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:27.775176500 INFO:SerialBattery:Testing Jkbms
2024-01-10 20:21:28.042801500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:28.044073500 INFO:SerialBattery:Testing Lifepower
2024-01-10 20:21:28.314600500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-10 20:21:28.316211500 INFO:SerialBattery:Testing LltJbd
2024-01-10 20:21:28.640205500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-10 20:21:28.641021500 INFO:SerialBattery:Testing Renogy at address "\x30"
2024-01-10 20:21:28.923946500 ERROR:SerialBattery:>>> ERROR: No reply - returning
[... cut ...]
2024-01-10 20:21:30.711337500 #1
2024-01-10 20:21:30.711339500 #2
2024-01-10 20:21:30.711362500 #3
2024-01-10 20:21:30.711363500 #4
2024-01-10 20:21:30.711365500 #5
2024-01-10 20:21:30.711366500 #6
2024-01-10 20:21:30.711367500 #7
2024-01-10 20:21:30.711368500 #8
2024-01-10 20:21:30.711370500 #9
2024-01-10 20:21:30.711371500 #10
2024-01-10 20:21:30.711372500 #11
2024-01-10 20:21:30.711373500 #12
2024-01-10 20:21:30.711375500 #13
2024-01-10 20:21:30.711376500 #14
2024-01-10 20:21:30.711377500 #15
`

mr-manuel, 2024-01-10:
Please use tripple backticks in the line before and after your code for proper formatting :-)

ToxSox, 2024-01-10:
okay, sorry

ToxSox, 2024-01-11:
I have tried v0.14.3 and that works fine.

mr-manuel, 2024-01-12:
Could you please install the `nightly` from the `dev` branch that I changed a few seconds ago? If it still doesn't work please enable debugging in the config.

ToxSox, 2024-01-13:
```2024-01-11 18:10:46.565932500 *** CCGX booted (0) ***
2024-01-11 18:11:27.784085500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-11 18:11:27.784778500 INFO:SerialBattery:dbus-serialbattery v0.14.3
2024-01-11 18:11:27.785395500 INFO:SerialBattery:Testing LltJbd
2024-01-11 18:11:27.924076500 INFO:SerialBattery:Connection established to LltJbd
2024-01-11 18:11:27.924717500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB0
2024-01-11 18:11:27.925226500 INFO:SerialBattery:=== Settings ===
2024-01-11 18:11:27.925798500 INFO:SerialBattery:> Connection voltage NoneV | current NoneA | SOC None%
2024-01-11 18:11:27.926319500 INFO:SerialBattery:> Cell count None | cells populated 0
2024-01-11 18:11:27.926840500 INFO:SerialBattery:> CCCM SOC True | DCCM SOC True
2024-01-11 18:11:27.927350500 INFO:SerialBattery:> CCCM CV True | DCCM CV True
2024-01-11 18:11:27.927852500 INFO:SerialBattery:> CCCM T True | DCCM T True
2024-01-11 18:11:27.928396500 INFO:SerialBattery:> MIN_CELL_VOLTAGE 2.9V | MAX_CELL_VOLTAGE 3.45V
2024-01-11 18:11:27.960725500 INFO:SerialBattery:DeviceInstance = 1
2024-01-11 18:11:27.961271500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-01-13 07:54:51.475825500 *** CCGX booted (0) ***
2024-01-13 08:00:51.615654500 *** CCGX booted (0) ***
2024-01-13 08:02:08.089887500 INFO:SerialBattery:
2024-01-13 08:02:08.090234500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-13 08:02:08.091137500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-13 08:02:24.107085500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-13 08:02:24.107648500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-01-13 08:02:24.234389500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:24.234953500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-01-13 08:02:24.362126500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:24.362595500 INFO:SerialBattery:Testing Ecs
2024-01-13 08:02:24.422332500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:24.422801500 INFO:SerialBattery:Testing HeltecModbus
2024-01-13 08:02:27.759385500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:27.759934500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-01-13 08:02:28.569419500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:28.570007500 INFO:SerialBattery:Testing Jkbms
2024-01-13 08:02:28.839405500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:28.841068500 INFO:SerialBattery:Testing Lifepower
2024-01-13 08:02:29.114600500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-13 08:02:29.115939500 INFO:SerialBattery:Testing LltJbd
2024-01-13 08:02:29.451798500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 130
2024-01-13 08:02:29.498123500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:29.545433500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:29.577462500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:29.609509500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:29.926303500 INFO:SerialBattery:Connection established to LltJbd
2024-01-13 08:02:29.926761500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB0
2024-01-13 08:02:29.927180500 INFO:SerialBattery:========== Settings ==========
2024-01-13 08:02:29.927651500 INFO:SerialBattery:> Connection voltage: 52.93V | Current: 0.0A | SoC: None%
2024-01-13 08:02:29.928073500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-01-13 08:02:29.928509500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-13 08:02:29.928963500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2024-01-13 08:02:29.929391500 INFO:SerialBattery:> CVCM:     True
2024-01-13 08:02:29.929838500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-13 08:02:29.930297500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-13 08:02:29.930737500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-13 08:02:29.931175500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-13 08:02:29.931721500 INFO:SerialBattery:Serial Number/Unique Identifier: 35_107.0Ah
2024-01-13 08:02:30.114852500 INFO:SerialBattery:Remove /Settings/Devices/serialbattery_ttyUSB0 from dbus. Old entry. Delete result: False
2024-01-13 08:02:30.736536500 INFO:SerialBattery:DeviceInstance = 1
2024-01-13 08:02:30.738183500 INFO:SerialBattery:Used device instances: ['1']
2024-01-13 08:02:30.740175500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-01-13 08:02:30.938673500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 130
2024-01-13 08:02:30.987149500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:31.127815500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:31.165134500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:31.212392500 WARNING:SerialBattery:>>> WARN: BMS rejected request. Status 129
2024-01-13 08:02:31.239492500 INFO:SerialBattery:publish config values = True
2024-01-13 08:02:31.309574500 INFO:SerialBattery:DeviceInstance = 1
2024-01-13 08:02:31.310132500 INFO:SerialBattery:Changed DeviceInstance = 1
2024-01-13 08:02:31.311791500 INFO:SerialBattery:Changed CustomName = SerialBattery(LLT/JBD)```

Thanks! It works now, with some warnings in the log.

closed 2024-01-13 by mr-manuel