Louisvdw/dbus-serialbattery #916: Battery Temperature shows 100 degrees

JurgenDaniel, 2024-01-11:
### Describe the problem

Hi, 
just installed a JK bms and connect it through bluetooth to my cerbo gx.
Unfortunately i have a problem with the battery temp, the cerbo shows a temp of 100 degrees, while the Jk bms shows the correct temp in the Jk app. (see screenshot below)
In the log i see that the temperature is 6552.1 & 6552.3 **(The screenshot and log is not from the same time and temperature.)**
Is this a bug that can be fixed by software? I have already tried somethings, like using different versions and also de nightly dev.
Could someone help me? 

Screenshot JK BMS **(Please note that the temps are below 0)**
[image]

### Driver version

1.1.20231224dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

15

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:10:10:20
TEMP_BATTERY = 0
LOGGING = DEBUG
```

### Relevant log output

```shell
2024-01-11 17:15:51.548359500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  0
2024-01-11 17:15:51.548369500 DEBUG:SerialBattery:[]
2024-01-11 17:15:51.548371500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  128
2024-01-11 17:15:51.548375500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 212, 197, 12, 196, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 127, 0, 0, 197, 12, 2, 0, 3, 5, 57, 0, 66, 0, 53, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 51, 0, 51, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-01-11 17:15:51.584210500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  22
2024-01-11 17:15:51.585217500 DEBUG:SerialBattery:ncallback(): \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0b\x00\x00\x00\x00\x00
2024-01-11 17:15:51.585843500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2024-01-11 17:15:51.586667500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 212, 197, 12, 196, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 127, 0, 0, 197, 12, 2, 0, 3, 5, 57, 0, 66, 0, 53, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 51, 0, 51, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-01-11 17:15:51.587264500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2024-01-11 17:15:51.588323500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 212, 197, 12, 196, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 127, 0, 0, 197, 12, 2, 0, 3, 5, 57, 0, 66, 0, 53, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 51, 0, 51, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 11, 0, 0, 0, 0, 0]
2024-01-11 17:15:51.601579500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  128
[... 59 lines cut ...]
2024-01-11 17:15:52.733668500 DEBUG:SerialBattery:fb[54]: 0.0.0.0.0
2024-01-11 17:15:52.734335500 DEBUG:SerialBattery:fb[70]: 0.0.255.127.0
2024-01-11 17:15:52.734917500 DEBUG:SerialBattery:fb[134]: 0.0.0.0.0
2024-01-11 17:15:52.735481500 DEBUG:SerialBattery:fb[144]: 0.0.11.0.0
2024-01-11 17:15:52.736034500 DEBUG:SerialBattery:fb[289]: 254.255.127.220.15
2024-01-11 17:15:52.736567500 DEBUG:SerialBattery:bms_max_cell_count recognized: 32
2024-01-11 17:15:52.737161500 DEBUG:SerialBattery:processing frame with battery cell info
2024-01-11 17:15:52.743887500 DEBUG:SerialBattery:decode_cellinfo_jk02(): self.frame_buffer
2024-01-11 17:15:52.745978500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 214, 197, 12, 196, 12, 197, 12, 197, 12, 197, 12, 196, 12, 197, 12, 197, 12, 196, 12, 198, 12, 198, 12, 197, 12, 197, 12, 197, 12, 197, 12, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 127, 0, 0, 197, 12, 2, 0, 3, 5, 57, 0, 66, 0, 53, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 52, 0, 51, 0, 51, 0, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 11, 0, 0, 0, 0, 0, 133, 191, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 241, 255, 243, 255, 0, 0, 0, 0, 0, 0, 0, 66, 139, 219, 2, 0, 192, 69, 4, 0, 0, 0, 0, 0, 45, 0, 0, 0, 100, 0, 0, 0, 41, 43, 3, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 1, 0, 0, 0, 60, 4, 0, 0, 0, 0, 116, 50, 61, 64, 0, 0, 0, 0, 38, 19, 41, 22, 0, 1, 0, 1, 0, 5, 0, 0, 148, 44, 16, 0, 0, 0, 0, 0, 11, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 254, 255, 127, 220, 15, 1, 0, 0, 0, 0, 0, 0, 254]
2024-01-11 17:15:52.749298500 DEBUG:SerialBattery:{'model_nbr': 'BK-BLE-1.0', 'device_info': {'hw_rev': '11.XW', 'sw_rev': '11.261', 'uptime': 201600, 'vendor_id': 'JK_B1A20S15P', 'manufacturing_date': '231219', 'serial_number': '3040142672', 'production': 'Input Us'}, 'last_update': 1704993352.2149296, 'settings': {'cell_uvp': 2.6, 'cell_uvpr': 2.65, 'cell_ovp': 3.6, 'cell_ovpr': 3.5500000000000003, 'balance_trigger_voltage': 0.01, 'power_off_voltage': 2.5, 'max_charge_current': 25.0, 'max_discharge_current': 10.0, 'max_balance_current': 25.0, 'cell_count': 15, 'charging_switch': True, 'discharging_switch': True, 'balancing_switch': True}, 'cell_info': {'voltages': [3.269, 3.2680000000000002, 3.269, 3.269, 3.269, 3.2680000000000002, 3.269, 3.269, 3.2680000000000002, 3.27, 3.27, 3.269, 3.269, 3.269, 3.269], 'average_cell_voltage': 3.269, 'delta_cell_voltage': 0.002, 'max_voltage_cell': 3, 'min_voltage_cell': 5, 'resistances': [0.057, 0.066, 0.053, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.052000000000000005, 0.051000000000000004, 0.051000000000000004, 0.056, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'total_voltage': 49.029, 'current': 0.0, 'temperature_sensor_1': 6552.1, 'temperature_sensor_2': 6552.3, 'temperature_mos': 1.1, 'balancing_current': 0.0, 'balancing_action': 0.0, 'battery_soc': 66, 'capacity_remain': 187.275, 'capacity_nominal': 280.0, 'cycle_count': 0, 'cycle_capacity': 0.045, 'charging_switch_enabled': True, 'discharging_switch_enabled': True, 'balancing_active': False, 'error_bitmask_16': '0x0', 'error_bitmask_2': '0000000000000000', 'power': 0.0}, 'warnings': {'resistance_too_high': False, 'cell_count_wrong': False, 'charge_overtemp': False, 'charge_undertemp': False, 'discharge_overtemp': False, 'cell_overvoltage': False, 'cell_undervoltage': False, 'charge_overcurrent': False, 'discharge_overcurrent': False}}
2024-01-11 17:15:52.781189500 DEBUG:SerialBattery:logged to dbus [66]
2024-01-11 17:15:52.783177500 DEBUG:SerialBattery:Cells:[1]3.269V [2]3.2680000000000002V [3]3.269V [4]3.269V [5]3.269V [6]3.2680000000000002V [7]3.269V [8]3.269V [9]3.2680000000000002V [10]3.27V [11]3.27V [12]3.269V [13]3.269V [14]3.269V [15]3.269V [16]NoneV [17]NoneV [18]NoneV [19]NoneV [20]NoneV [21]NoneV [22]NoneV [23]NoneV [24]NoneV [25]NoneV [26]NoneV [27]NoneV [28]NoneV [29]NoneV [30]NoneV
tail: /data/log/dbus-blebattery.0/current has been replaced; following end of new file
2024-01-11 17:15:53.294998500 DEBUG:SerialBattery:compair recvd. crc: 6 vs calc. crc: 6
2024-01-11 17:15:53.295006500 DEBUG:SerialBattery:great success! frame complete and sane, lets decode
2024-01-11 17:15:53.295010500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 215, 197, 12, 196, 12, 197, 12, 198, 12, 197, 12, 196, 12, 196, 12, 197, 12, 196, 12, 197, 12, 197, 12, 197, 12, 197, 12, 197, 12, 1 [... cut ...]
2024-01-11 17:15:53.295218500 DEBUG:SerialBattery:fb[54]: 0.0.0.0.0
2024-01-11 17:15:53.295220500 DEBUG:SerialBattery:fb[70]: 0.0.255.127.0
2024-01-11 17:15:53.295222500 DEBUG:SerialBattery:fb[134]: 0.0.0.0.0
2024-01-11 17:15:53.295224500 DEBUG:SerialBattery:fb[144]: 0.0.11.0.0
2024-01-11 17:15:53.295227500 DEBUG:SerialBattery:fb[289]: 254.255.127.220.15
2024-01-11 17:15:53.295229500 DEBUG:SerialBattery:bms_max_cell_count recognized: 32
2024-01-11 17:15:53.295250500 DEBUG:SerialBattery:processing frame with battery cell info
```

JurgenDaniel, 2024-01-11:
Here is a screenshot of the 100 degree battery. . . little to high.
[image]

kommando828, 2024-01-11:
This looks like a uint 16 register being seen as int16.

So temps of -1.5 and -1.3 are seen as 6552.1 and 6552.3 when all are divided by 10

This tool allows you to see conversions

https://cryptii.com/pipes/integer-converter as seen in the screenshot

[image]

JurgenDaniel, 2024-01-12:
Thanks for your quick response!
It helped me a lot, i have looked into the code a little bit but couldn't find the self.translate function.
Today its alittle above 0 degrees where i live and right now it works fine. so it has probably something to do with the freezing temperature. 
Does anyone else have 100 degree battery temps in their history when its freezing?
[image]

kommando828, 2024-01-12:
Nearest I get to programming is Node-red but within Node-red I have to do a lot of int32 to int16 plus int16 to uint16 conversions inside nodes feeding modbus servers so I recognised the numbers. There will be code that is misreading 16623 and 16621 as above 100 instead of them actually be negative numbers. As modbus registers cannot contain either negative numbers or decimal points you have to know what type of number they are and apply a signed to unsigned conversion and apply a multiplier ie 01 /10 becomes 0.1.

mr-manuel, 2024-01-14:
What model and hardware version is your BMS?

pau1phi11ips, 2024-01-18:
I have a JK BMS, JK-B2A8S20P, connected via Bluetooth and also getting 100°C for the temps when it's below zero.

Serial Battery version `1.0.20231126dev`

mr-manuel, 2024-01-21:
The issue should be fixed with the `nightly` of the `dev` branch from my repo. Could you please try?

```bash
wget -O /tmp/install.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh

bash /tmp/install.sh
```

Select `nightly build` and then `dev (mr-manuel's repo)`.

pau1phi11ips, 2024-01-21:
Thanks, I'll give it a try tonight. The air temps have gone up a lot since but I'll stick one of the sensors in some frozen peas 😁

pau1phi11ips, 2024-01-23:
> Select `nightly build` and then `dev (mr-manuel's repo)`.

Hiya, sorry for the delay.

The air temp is currently 8°C. I put one of the probes in the frozen peas and it slowly crept down... 6, 4, 3, 2, 1... then jumped to -20°C. It didn't budge on the -20. Average battery temp was showing -6.

Should the -20 have been -2?

mr-manuel, 2024-01-28:
No there is a the max is limited to 100 and the min to -20.

Can you change

https://github.com/Louisvdw/dbus-serialbattery/blob/3c2401a606817f2955c1c17bf0fa4f2977fc06aa/etc/dbus-serialbattery/battery.py#L212-L221

to

```python
        if sensor == 0:
            self.temp_mos = value
        if sensor == 1:
            self.temp1 = value
        if sensor == 2:
            self.temp2 = value
        if sensor == 3:
            self.temp3 = value
        if sensor == 4:
            self.temp4 = value
```

So we see what temperature is calculated and how to fix it properly.

commit 2da6bae2 (2024-01-28): Fixes #916
https://github.com/Louisvdw/dbus-serialbattery/issues/916
closed 2024-01-28 by mr-manuel