Louisvdw/dbus-serialbattery #919: Daly_Can - on v1.1.20240112dev starts working (nearly)

hwater, 2024-01-12:
### Describe the bug

At first see my logfile:

After updating this seems to give an endless loop, see the Errror messages in the first half.

### How to reproduce

If you want me to tryout more with the Daly on CAN-Bus feel free to contact me, the settings is now working in my Lab.

Keep on with your fantastic Work, best Whishes from Austria. Holger Watermann

### Expected behavior

.After commenting out the IF inside battery.py it starts to work without further complains.

`        # if BMS limit is lower then config limit and therefore the values are not the same,
        # then the limit was also read from the BMS
        # if utils.MAX_BATTERY_CHARGE_CURRENT > self.max_battery_charge_current:
        #    charge_limits.update({self.max_battery_charge_current: "BMS Settings"})`

### Driver version

v1.1.20240112dev

### Venus OS device type

Raspberry Pi 2

### Venus OS version

v3.20~37

### BMS type

Daly Smart BMS

### Cell count

4

### Battery count

1

### Connection type

Raspberry Pi RS485 HAT

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
MAX_BATTERY_CHARGE_CURRENT = 10.0
MAX_BATTERY_DISCHARGE_CURRENT = 40.0

CAN_PORT = can8
```

### Relevant log output

```shell
2024-01-12 11:18:49.862609500 
2024-01-12 11:18:51.143060500 INFO:SerialBattery:
2024-01-12 11:18:51.143475500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-12 11:18:51.144375500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-12 11:18:51.913939500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-12 11:18:51.914587500 INFO:SerialBattery:Testing Daly_Can
2024-01-12 11:18:52.002564500 INFO:SerialBattery:DalyBMS 4 cells
2024-01-12 11:18:52.003243500 INFO:SerialBattery:Connection established to Daly_Can
2024-01-12 11:18:52.003816500 INFO:SerialBattery:Battery Daly_Can connected to dbus from can8
2024-01-12 11:18:52.004337500 INFO:SerialBattery:========== Settings ==========
2024-01-12 11:18:52.004878500 INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
2024-01-12 11:18:52.005403500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 0
2024-01-12 11:18:52.005937500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-12 11:18:52.006511500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 10.0A | MAX BATTERY DISCHARGE CURRENT: 40.0A
2024-01-12 11:18:52.007037500 INFO:SerialBattery:> CVCM:     True
2024-01-12 11:18:52.007674500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-12 11:18:52.008237500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-12 11:18:52.008769500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-12 11:18:52.009300500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-12 11:18:52.009906500 INFO:SerialBattery:Serial Number/Unique Identifier: DalyBMS4cells_NoneAh
2024-01-12 11:18:53.785838500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-01-12 11:18:54.015251500 INFO:SerialBattery:DeviceInstance = 2
2024-01-12 11:18:54.015760500 INFO:SerialBattery:Used device instances: ['1', '3', '12', '4', '9', '5', '10', '6', '7', '2', '14', '13', '8', '11']
2024-01-12 11:18:54.016291500 INFO:SerialBattery:com.victronenergy.battery.can8
2024-01-12 11:18:54.035883500 INFO:SerialBattery:publish config values = True
2024-01-12 11:18:55.097989500 Traceback (most recent call last):
2024-01-12 11:18:55.097996500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 684, in publish_battery
2024-01-12 11:18:55.098002500     self.battery.manage_charge_current()
2024-01-12 11:18:55.098005500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 669, in manage_charge_current
2024-01-12 11:18:55.098010500     # if utils.MAX_BATTERY_CHARGE_CURRENT > self.max_battery_charge_current:
2024-01-12 11:18:55.098015500 TypeError: '>' not supported between instances of 'float' and 'NoneType'
2024-01-12 11:18:55.098179500 Setted setting /Settings/Devices/serialbattery_DalyBMS4cells_NoneAh/LastSeen to 1705058333
2024-01-12 11:18:55.377246500 
2024-01-12 11:18:56.750579500 INFO:SerialBattery:
2024-01-12 11:18:56.751242500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-12 11:18:56.752268500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-12 11:18:57.515659500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-12 11:18:57.516438500 INFO:SerialBattery:Testing Daly_Can
2024-01-12 11:18:57.600726500 INFO:SerialBattery:DalyBMS 4 cells
2024-01-12 11:18:57.601483500 INFO:SerialBattery:Connection established to Daly_Can
2024-01-12 11:18:57.602044500 INFO:SerialBattery:Battery Daly_Can connected to dbus from can8
2024-01-12 11:18:57.602557500 INFO:SerialBattery:========== Settings ==========
2024-01-12 11:18:57.603115500 INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
2024-01-12 11:18:57.603644500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 0
2024-01-12 11:18:57.604196500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-12 11:18:57.604777500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 10.0A | MAX BATTERY DISCHARGE CURRENT: 40.0A
2024-01-12 11:18:57.605326500 INFO:SerialBattery:> CVCM:     True
2024-01-12 11:18:57.605876500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-12 11:18:57.606426500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-12 11:18:57.606970500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-12 11:18:57.607601500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-12 11:18:57.608239500 INFO:SerialBattery:Serial Number/Unique Identifier: DalyBMS4cells_NoneAh
2024-01-12 11:18:59.374166500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-01-12 11:18:59.595778500 INFO:SerialBattery:DeviceInstance = 2
2024-01-12 11:18:59.596321500 INFO:SerialBattery:Used device instances: ['1', '3', '12', '4', '9', '5', '10', '6', '7', '2', '14', '13', '8', '11']
2024-01-12 11:18:59.596865500 INFO:SerialBattery:com.victronenergy.battery.can8
2024-01-12 11:18:59.616459500 INFO:SerialBattery:publish config values = True
```

### Any other information that may be helpful

There is no CAN Profile selector here to chooose from, so please add it. I took RS485 HAT.

I use an normal CAN HAT on the pi and that gives me can8 connected to the included CAN L/H Lines of the DALY bms. 

Addditional Findings:
As it seems, the DALY_CAN has an enabled 120 Ohms Terminator connectet, so connect only one more at he other end of your CAN Bus! This gives you the normal 60 Ohm on the Bus (measure without power attached)

hwater, 2024-01-12:
> Can you please install the nightly from my branch and test if it works?
>
> In your case this should work with:
>
> bash /data/etc/dbus-serialbattery/install.sh
> Then select nightly build (option 3) -> dev (mr-manuel's repo) (option 3).

Can you please enable it again. I startet the install from an wrong pi. 
Now on the right one it is gone ;-(

Greetings, Holger

hwater, 2024-01-12:
Hello Manuel
this patch works, i will go on with more checking....

greetings, Holger

`Last login: Fri Jan 12 17:30:27 2024
root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-canbattery.can8/current | tai64nlocal
2024-01-12 17:30:32.389509500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-12 17:30:32.389519500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 10.0A | MAX BATTERY DISCHARGE CURRENT: 40.0A
2024-01-12 17:30:32.389524500 INFO:SerialBattery:> CVCM:     True
2024-01-12 17:30:32.389527500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-12 17:30:32.389532500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-12 17:30:32.389537500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-12 17:30:32.389660500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-12 17:30:32.389664500 INFO:SerialBattery:Serial Number/Unique Identifier: DalyBMS4cells_NoneAh
2024-01-12 17:30:34.440181500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-01-12 17:30:34.684706500 INFO:SerialBattery:DeviceInstance = 2
2024-01-12 17:30:34.685286500 INFO:SerialBattery:Used device instances: ['1', '3', '12', '4', '9', '5', '10', '6', '7', '2', '14', '13', '8', '11']
2024-01-12 17:30:34.685822500 INFO:SerialBattery:com.victronenergy.battery.can8
2024-01-12 17:30:34.706321500 INFO:SerialBattery:publish config values = True
`

hwater, 2024-01-12:
Sorry, this is the logfile AFTER rebooting:

Also, do i need to populate all the parameters into my **config.ini** ?
Now, as seen above i have only the CAN port and the MAX_BATTERY in it.

`2024-01-12 18:30:04.549606500 *** CCGX booted (0) ***
2024-01-12 18:31:56.995809500 
2024-01-12 18:31:58.967776500 INFO:SerialBattery:
2024-01-12 18:31:58.968400500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-12 18:31:58.969630500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-12 18:32:00.056268500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-12 18:32:00.056845500 INFO:SerialBattery:Testing Daly_Can
2024-01-12 18:32:00.187786500 INFO:SerialBattery:DalyBMS 4 cells
2024-01-12 18:32:00.188749500 INFO:SerialBattery:Connection established to Daly_Can
2024-01-12 18:32:00.189269500 INFO:SerialBattery:Battery Daly_Can connected to dbus from can8
2024-01-12 18:32:00.189715500 INFO:SerialBattery:========== Settings ==========
2024-01-12 18:32:00.190172500 INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
2024-01-12 18:32:00.190615500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 0
2024-01-12 18:32:00.191085500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-12 18:32:00.191641500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 10.0A | MAX BATTERY DISCHARGE CURRENT: 40.0A
2024-01-12 18:32:00.192087500 INFO:SerialBattery:> CVCM:     True
2024-01-12 18:32:00.192559500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-12 18:32:00.193004500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-12 18:32:00.193457500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-12 18:32:00.193891500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-12 18:32:00.194419500 INFO:SerialBattery:Serial Number/Unique Identifier: DalyBMS4cells_NoneAh
2024-01-12 18:32:02.322136500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-01-12 18:32:02.558154500 INFO:SerialBattery:DeviceInstance = 2
2024-01-12 18:32:02.558759500 INFO:SerialBattery:Used device instances: ['1', '3', '12', '4', '9', '5', '10', '6', '7', '2', '14', '13', '8', '11']
2024-01-12 18:32:02.559327500 INFO:SerialBattery:com.victronenergy.battery.can8
2024-01-12 18:32:02.579348500 INFO:SerialBattery:publish config values = True
`

mr-manuel, 2024-01-12:
> Can you please enable it again. 

What should I enable again?

To have proper code formatting add trippe backticks or tildes `~~~` in the line before and after the code.

> Also, do i need to populate all the parameters into my config.ini ?

No, only relevant for your system. All other are fetched from the `default.config.ini`.

So all is working and I can close this issue?

closed 2024-01-12 by mr-manuel

hwater, 2024-01-12:
Hello Manuel
About the "enable" - I had the intension, the "dev (mr-manuel's repo) (option 3)." was gone automatically after i installed it to the wrong Computer. My failure, it was allready there.

I will try the ~~~ thx for this Tip.

The Readings from CAN are working now, but i an not shure if it is all correct, i don´t see the Cell Voltages in the Venus - only the min and max Voltage.

But that will be another Issue - this one is ready to be closed.

HoW

~~~2024-01-12 19:16:15.446169500 *** CCGX booted (0) ***
2024-01-12 19:18:06.373312500 
2024-01-12 19:18:08.501008500 INFO:SerialBattery:
2024-01-12 19:18:08.501498500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-12 19:18:08.502483500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-12 19:18:09.537288500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-12 19:18:09.537869500 INFO:SerialBattery:Testing Daly_Can
2024-01-12 19:18:09.687542500 INFO:SerialBattery:DalyBMS 4 cells
2024-01-12 19:18:09.688139500 INFO:SerialBattery:Connection established to Daly_Can
2024-01-12 19:18:09.688687500 INFO:SerialBattery:Battery Daly_Can connected to dbus from can8
2024-01-12 19:18:09.689175500 INFO:SerialBattery:========== Settings ==========
2024-01-12 19:18:09.689733500 INFO:SerialBattery:> Connection voltage: NoneV | Current: NoneA | SoC: None%
2024-01-12 19:18:09.690225500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 0
2024-01-12 19:18:09.690818500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-12 19:18:09.691434500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 10.0A | MAX BATTERY DISCHARGE CURRENT: 40.0A
2024-01-12 19:18:09.691974500 INFO:SerialBattery:> CVCM:     True
2024-01-12 19:18:09.692709500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-12 19:18:09.693298500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-12 19:18:09.694004500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-12 19:18:09.694581500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-12 19:18:09.695219500 INFO:SerialBattery:Serial Number/Unique Identifier: DalyBMS4cells_NoneAh
2024-01-12 19:18:11.847755500 INFO:SerialBattery:Found existing battery with DeviceInstance = 2
2024-01-12 19:18:12.151129500 INFO:SerialBattery:DeviceInstance = 2
2024-01-12 19:18:12.151839500 INFO:SerialBattery:Used device instances: ['1', '3', '12', '4', '9', '5', '10', '6', '7', '2', '14', '13', '8', '11']
2024-01-12 19:18:12.152449500 INFO:SerialBattery:com.victronenergy.battery.can8
2024-01-12 19:18:12.175610500 INFO:SerialBattery:publish config values = True
~~~

commit ea69264d (2024-01-28): Fixes #919
https://github.com/Louisvdw/dbus-serialbattery/issues/919