Louisvdw/dbus-serialbattery #923: VRM instance id not unique with multiple Seplos BMS on separate USB->485 adapters

johndparker, 2024-01-17:
### Describe the bug

I have installed and configured this driver together with dbus-aggregate-batteries and all is working very well so far except both individual batteries have the same VRM id of 1 instead of having separate ids.
(The virtual combined battery has an id of 0)
I see other discussion around this issue, but not for Seplos.

### How to reproduce

Connect 2 Seplos BMS on separate USB to 485 adaptors

### Expected behavior

Each serialbattery instance should have a separate VRM id.

### Driver version

v1.0.20240102beta

### Venus OS device type

Cerbo GX

### Venus OS version

v3.13

### BMS type

Seplos

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 140.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
```

### Relevant log output

```shell
from /data/log/dbus-serialbattery.ttyUSB1/current

2024-01-17 18:07:06.361820500 *** CCGX booted (0) ***
2024-01-17 18:07:54.414990500 INFO:SerialBattery:
2024-01-17 18:07:54.415829500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-17 18:07:54.479641500 INFO:SerialBattery:dbus-serialbattery v1.0.20240102beta
2024-01-17 18:08:10.490165500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-01-17 18:08:10.491349500 INFO:SerialBattery:Testing Daly at address "\x40"
2024-01-17 18:08:10.629890500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-17 18:08:10.629900500 INFO:SerialBattery:Testing Daly at address "\x80"
2024-01-17 18:08:10.777539500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-01-17 18:08:10.777549500 INFO:SerialBattery:Testing Ecs
[... 286 lines cut ...]
2024-01-17 18:09:13.873875500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-17 18:09:13.874916500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-17 18:09:13.875899500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 140.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2024-01-17 18:09:13.876846500 INFO:SerialBattery:> CVCM:     True
2024-01-17 18:09:13.905635500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-17 18:09:13.906650500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-17 18:09:13.919860500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-17 18:09:13.922812500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-17 18:09:13.926762500 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_280.0Ah
2024-01-17 18:09:26.418013500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-17 18:09:27.212817500 INFO:SerialBattery:DeviceInstance = 1
2024-01-17 18:09:27.214103500 INFO:SerialBattery:Used device instances: ['1']
2024-01-17 18:09:27.215541500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB3
2024-01-17 18:09:27.326819500 INFO:SerialBattery:publish config values = True
2024-01-17 18:29:01.984596500 INFO:SerialBattery:Changed CustomName = Battery 1
2024-01-17 18:29:36.103516500 INFO:SerialBattery:Changed CustomName = Battery 0
```

### Any other information that may be helpful

I did not have to configure anything for both serialbatteries to appear in the RemoteConsole, and the aggregatebattery virtual found the instances and combined them again without any specific configuration.

Note also that this driver is interfering with ttyUSB2 which is actually one of my ET112 power meters. The other ET112 is on ttyUSB0 and that does not seem to be affected.

Perhaps the Seplos driver needs to use the connection (USB1 vs USB3) to create a unique ID?

johndparker, 2024-01-17:
Oh - I missed this log file

/data/log/serial-starter/current

```
2024-01-17 19:45:11.029505500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 19:45:17.110954500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 19:45:22.657169500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 19:45:32.494473500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 19:46:18.096106500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 19:46:27.897256500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 19:46:31.782333500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2024-01-17 19:57:08.220142500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 19:57:14.082369500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 19:57:19.849634500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 19:57:30.004049500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 19:58:15.171582500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 19:58:24.982766500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 19:58:28.464374500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2024-01-17 20:04:18.807548500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 20:04:24.773427500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 20:04:30.548925500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 20:04:40.274294500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 20:05:25.892674500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 20:05:35.573451500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 20:05:39.089072500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2024-01-17 20:14:34.189821500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 20:14:39.954296500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 20:14:45.631621500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 20:14:55.629409500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 20:15:40.881468500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 20:15:50.615057500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 20:15:56.120442500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2024-01-17 20:18:15.788605500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 20:18:21.449173500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 20:18:25.272817500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 20:18:35.222458500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 20:19:22.903842500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 20:19:33.147009500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 20:19:38.617022500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2024-01-17 20:28:03.963116500 INFO: Start service dbus-fzsonick-48tl.ttyUSB2 once
2024-01-17 20:28:09.941473500 INFO: Start service dbus-imt-si-rs485tc.ttyUSB2 once
2024-01-17 20:28:15.900071500 INFO: Start service dbus-modbus-client.serial.ttyUSB2 once
2024-01-17 20:28:25.808802500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2024-01-17 20:29:12.951994500 INFO: Start service gps-dbus.ttyUSB2 once
2024-01-17 20:29:23.005303500 INFO: Start service vedirect-interface.ttyUSB2 once
2024-01-17 20:29:28.528030500 INFO: Start service dbus-cgwacs.ttyUSB2 once
```

Looks like something is struggling to do something with USB2 when it should be leaving it alone!

johndparker, 2024-01-17:
I added ttyUSB2 to the exclude list and now it is not being bothered by the driver.

That just leaves the problem that both serialbattery intances have the same VRM id of "1"

mr-manuel, 2024-01-17:
If you have two identical BMS you have to increase the capacity of one battery by one, else you don't have an unique id for the battery.
E.g. change the capacity for each battery to be different
280
281
279
282
278
...

The small difference makes no serial impact.

johndparker, 2024-01-17:
Hi Mr Manuel

I'll try that but I'll also be looking for a better way to generate a unique ID - the adaptors have unique serial numbers in /dev/serial/by-id so we should be able to do that.

The problem with USB2 looks like a separate problem so I'll open a new ticket for that.

Having disabled the driver  it looks like the grid meter is being left alone and the system is not going into passtrough.

Kind regards

John

mr-manuel, 2024-01-17:
> I'll try that but I'll also be looking for a better way to generate a unique ID - the adaptors have unique serial numbers in /dev/serial/by-id so we should be able to do that.

The unique ID has to be fetched from the BMS and not the serial adapter. What if you swap the adapters? If you build a solution for this it won't be implemented, since it's not applicable for the wide community.

mr-manuel, 2024-01-17:
For your other problem try to swap the `0` with `1` in https://github.com/Louisvdw/dbus-serialbattery/blob/976ef061eca12e3fa4c8f95c6119cd4f3f9302ed/etc/dbus-serialbattery/dbus-serialbattery.py#L149

johndparker, 2024-01-17:
I'll try the sys.exit change in the morning, and will consider if there is something else unique in the Seplos BMS - it just does not feel right to mess with the capacity whereas linking it to how it is connected makes some sense - at least to me. I'll sleep on it.

Thank you for all your assistance.

Kind regards
John

mr-manuel, 2024-01-17:
Since your batteries are for sure not exactly 280 Ah but have a deviation of +/- 2 % which are 5,6 Ah on a 280 Ah battery it's no big deal.

closed 2024-01-17 by mr-manuel