Louisvdw/dbus-serialbattery #927: JKBMS B2A24S15P & v1.0.20230531 Never switches to FLOAT

BipedalPrimate, 2024-01-20:
### Describe the problem

As I understand it, the following parameters in config.ini (full file attached as config.ini.txt below) determine when to switch to Float:

MAX_CELL_VOLTAGE = 3.450
FLOAT_CELL_VOLTAGE = 3.350
CVCM_ENABLE = False (Started with this set to True but never achieved 55.2v because of initially unbalanced pack - now balanced)
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.160
MAX_VOLTAGE_TIME_SEC = 600 (note that I have tried setting this to 1 sec)
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

It appears that all of these conditions have been met on a number of occasions yet Float is never achieved.

Am I missing something? If so, what?

[image]
[image]
[image]
[image]
[image]
[image]

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
; MAX_BATTERY_CHARGE_CURRENT    = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 125.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
; MIN_CELL_VOLTAGE   = 2.900
MIN_CELL_VOLTAGE   = 3.150

; Max voltage can seen as absorption voltage
[... 386 lines cut ...]
2024-01-18 05:40:30.160814500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-18 05:40:30.161161500 INFO:SerialBattery:Testing Jkbms
2024-01-18 05:40:30.218530500 INFO:SerialBattery:Connection established to Jkbms
2024-01-18 05:40:30.218885500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB2
2024-01-18 05:40:30.219175500 INFO:SerialBattery:========== Settings ==========
2024-01-18 05:40:30.219499500 INFO:SerialBattery:> Connection voltage: 55.2V | Current: -2.0A | SoC: 99%
2024-01-18 05:40:30.219876500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-01-18 05:40:30.220173500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-18 05:40:30.220480500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 125.0A
2024-01-18 05:40:30.220764500 INFO:SerialBattery:> CVCM:     False
2024-01-18 05:40:30.221058500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.15V | MAX CELL VOLTAGE: 3.45V
2024-01-18 05:40:30.221326500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-18 05:40:30.221595500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-18 05:40:30.221860500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
2024-01-18 05:40:30.222125500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A24S15P
2024-01-18 05:40:30.254207500 INFO:SerialBattery:DeviceInstance = 2
2024-01-18 05:40:30.254558500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB2
2024-01-18 05:40:30.277622500 INFO:SerialBattery:publish config values = 1
^C
root@raspberrypi4:~#
```

### Any other information that may be helpful

No output from: tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal

mr-manuel, 2024-01-20:
Did you read the `condif.default.ini`?

https://github.com/Louisvdw/dbus-serialbattery/blob/e3206f750e5c2c1651bae4c6073d0905f056bdd1/etc/dbus-serialbattery/config.default.ini#L90-L112

BipedalPrimate, 2024-01-20:
Thank you.  I get it now.

What you are saying is that FLOAT will never occur unless CVCM = TRUE regardless of the other conditions.

mr-manuel, 2024-01-20:
Thats correct.

closed 2024-01-20 by mr-manuel