Louisvdw/dbus-serialbattery #928: CPU Load reduction

Honusnap, 2024-01-20:
### Describe the problem

Hello All,

What are the options to reduce CPU Load of the serialbattery driver ?

7% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:8C:E9:0C:E6
3% {dbus_mqtt.py} /usr/bin/python3 -u /opt/victronenergy/dbus-mqtt/dbus_mqtt.py --init-broker
3% {dbus_systemcalc} /usr/bin/python3 -u /opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py

I've reduced the SOC calculus every 120s, is there any other options that can be changed, maybe the frequency at chich the serialbattery driver is calculating datas can be changed ?

Thanks a lot,

### Driver version

1.020230531 (Last stable)

### Venus OS device type

Cerbo GX

### Venus OS version

3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.41
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.40

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
[... 281 lines cut ...]
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; --------- Voltage drop ---------
; If you have a voltage drop between the BMS and the charger because of wire size or length
; then you can specify the voltage drop here. The driver will then add the voltage drop
; to the calculated CVL to compensate.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 V and the battery is now charged until the charger reaches 55.20 V.
;     The BMS now measures 55.05 V since there is a voltage drop of 0.15 V on the cable.
;     Since the dbus-serialbattery reads the voltage of 55.05 V from the BMS the max voltage
;     of 55.20 V is never reached and max voltage is kept forever.
;     By setting the VOLTAGE_DROP to 0.15 V the voltage on the charger is increased and the
;     target voltage on the BMS is reached.
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
...
```

mr-manuel, 2024-01-21:
First of all, did you try the nightly? We wrote a big note for this when you open an issue.

[image]

What is your system load?

Honusnap, 2024-01-21:
As stated, it's not a bug... no need to try nightly which i just get rid of for stable release.

I'm just asking what are the action that can be taken to reduce CPU load.

7% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:8C:E9:0C:E6
7% seems pretty high already for me.

mr-manuel, 2024-01-22:
When connected via Bluetooth the driver is using an active callback from the BMS, in your case the data is received probably every half a second.

You could change

https://github.com/Louisvdw/dbus-serialbattery/blob/976ef061eca12e3fa4c8f95c6119cd4f3f9302ed/etc/dbus-serialbattery/dbus-serialbattery.py#L225-L228

to (pay attention at the line indentation that changed also)

```python
    # try using active callback on this battery
    # if not battery.use_callback(lambda: poll_battery(mainloop)):
    # if not possible, poll the battery every poll_interval milliseconds
    gobject.timeout_add(battery.poll_interval, lambda: poll_battery(mainloop))
```

and then increase

https://github.com/Louisvdw/dbus-serialbattery/blob/976ef061eca12e3fa4c8f95c6119cd4f3f9302ed/etc/dbus-serialbattery/battery.py#L63

but this will not be supported and you will have problems at least once a day where the driver crashes.

Honusnap, 2024-01-23:
You mean that when active callback can't be initialised the driver will crash once a day ?

I'll check the code, maybe there is something to do in the poll_battery(mainloop) to drop the action in case the call is within the same Poll_Intervall.

closed 2024-01-28 by mr-manuel