Louisvdw/dbus-serialbattery #929: Only one of two BMSs showing up

rldoose, 2024-01-21:
### Describe the problem

Only one of two BMSs showing up.  I am using the nightly build from dev as this was necessary to resolve the last issue I submitted.  Either will work when used one at a time, but not together at the same time.

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.13

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]
; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 150
MAX_BATTERY_DISCHARGE_CURRENT = 200

; --------- Cell Voltages ---------
MIN_CELL_VOLTAGE   = 2.6
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.550
FLOAT_CELL_VOLTAGE = 3.4

; --------- BMS disconnect behaviour ---------
BLOCK_ON_DISCONNECT = False

; --------- Charge mode ---------
LINEAR_LIMITATION_ENABLE = True
LINEAR_RECALCULATION_EVERY = 60

; --------- Charge Voltage limitation (affecting CVL) ---------
; Charge voltage control management enable (True/False).
CVCM_ENABLE = True
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.080
MAX_VOLTAGE_TIME_SEC = 1800
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

; --------- Temperature limitation (affecting CCL/DCL) ---------
; Charge current control management (CCCM) & ; Discharge current control management (DCCM)
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

TEMPERATURE_LIMITS_WHILE_CHARGING    = 	0,   	2,	5,  	10,  	15, 	20, 	35,  	40, 	55
MAX_CHARGE_CURRENT_T_FRACTION        = 	0, 	0.1,	0.2, 	0.4, 	0.8,  	1,  	1, 	0.4,  	0
TEMPERATURE_LIMITS_WHILE_DISCHARGING = 	-20,   	0,   	5,  	10, 	15, 	45, 	55
MAX_DISCHARGE_CURRENT_T_FRACTION     =	0,	0, 	0.3, 	0.4,  	1,  	1, 	.5
	
; --------- Time-To-Go ---------
TIME_TO_GO_ENABLE = True
TIME_TO_SOC_RECALCULATE_EVERY = 60

BMS_TYPE = LltJbd

; Publish the config settings to the dbus path "/Info/Config/"
PUBLISH_CONFIG_VALUES = 1

; Select the format of cell data presented on dbus [Valid values 0,1,2,3]
BATTERY_CELL_DATA_FORMAT = 1

; Battery temperature
; Specify how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 to sensor 4
TEMP_BATTERY = 0
TEMP_1_NAME = Temp 1
TEMP_2_NAME = Temp 2

; --------- BMS specific settings ---------
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

	; --------- Battery monitor specific settings ---------
	; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
	; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
	; since max voltage is never reached.
	; Example:
	;     cell count: 16
	;     MAX_CELL_VOLTAGE = 3.45
	;     max voltage calculated = 16 * 3.45 = 55.20
	;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
	;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
	;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
	;     Set VOLTAGE_DROP to 0.15
;VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
root@raspberrypi4:/data/etc/dbus-serialbattery# head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyUSB0 <== gps
==> /data/var/lib/serial-starter/ttyUSB1 <== sbattery
==> /data/var/lib/serial-starter/ttyUSB2 <== vedirect
==> /data/var/lib/serial-starter/ttyUSB3 <== vedirect
==> /data/var/lib/serial-starter/ttyUSB4 <== mkx

This Looks correct except for the gps on USB0.  Battery 1 shows as USB1 in the Remote Console as indicated above.  

root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB1/current | tai64nlocal
2024-01-20 22:50:16.182336500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-20 22:50:16.183468500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-20 22:50:16.183759500 INFO:SerialBattery:Testing LltJbd
2024-01-20 22:50:16.695921500 INFO:SerialBattery:Connection established to LltJbd
2024-01-20 22:50:16.696208500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyUSB1
2024-01-20 22:50:16.696461500 INFO:SerialBattery:========== Settings ==========
2024-01-20 22:50:16.696768500 INFO:SerialBattery:> Connection voltage: 13.3V | Current: -6.33A | SoC: 77.82%
2024-01-20 22:50:16.697022500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2024-01-20 22:50:16.697288500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-01-20 22:50:16.697562500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A
2024-01-20 22:50:16.697838500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A (read from BMS)
2024-01-20 22:50:16.698086500 INFO:SerialBattery:> CVCM:     True
2024-01-20 22:50:16.698361500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.6V | MAX CELL VOLTAGE: 3.55V
2024-01-20 22:50:16.698623500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-20 22:50:16.698886500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-20 22:50:16.699145500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-20 22:50:16.727624500 INFO:SerialBattery:DeviceInstance = 3
2024-01-20 22:50:16.727925500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
2024-01-20 22:50:16.989119500 INFO:SerialBattery:publish config values = 1

root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-01-21 00:36:53.588490500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-21 00:36:53.589075500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-21 00:36:53.589361500 INFO:SerialBattery:Testing LltJbd
2024-01-21 00:36:53.776643500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 00:36:54.277672500 INFO:SerialBattery:Testing LltJbd
2024-01-21 00:36:54.464562500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 00:36:54.966101500 INFO:SerialBattery:Testing LltJbd
2024-01-21 00:36:55.153801500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 00:36:55.658846500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
```

### Any other information that may be helpful

Attempting to reassign USB0 using 'echo "sbattery" > /data/var/lib/serial-starter/ttyUSB0' responds correctly and reassignes the port, but this is not kept after a reboot.  I unplugged all the other USB devices and only left the two batteries connected.  After many attempts plugging the batteries in, I did get both batteries to show briefly in the Remote Console, but this was lost as soon as I plugged in another USB device.  The device shown under USB0 seems random as I have seen it listed as most all the different devices.

mr-manuel, 2024-01-21:
> I am using the nightly build from dev

Why you write than that you are using v1.0.20230531 and also the log tell so?

rldoose, 2024-01-21:
Which log is that in?

Sent from iPhone

rldoose, 2024-01-21:
Correct:

***@***.***:~# tail -F /data/log/dbus-serialbattery.*/current | tai64nlocal
s-serialbattery.ttyUSB0/current <==
2024-01-21 04:51:48.872253500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-21 04:51:48.872831500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-21 04:51:48.873123500 INFO:SerialBattery:Testing LltJbd
2024-01-21 04:51:49.103106500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 04:51:49.604112500 INFO:SerialBattery:Testing LltJbd
2024-01-21 04:51:49.803998500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 04:51:50.305035500 INFO:SerialBattery:Testing LltJbd
2024-01-21 04:51:50.490362500 ERROR:SerialBattery:Unexpected err=TypeError('can only concatenate str (not "int") to str'), type(err)=<class 'TypeError'>
2024-01-21 04:51:50.991410500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2024-01-21 0

mr-manuel, 2024-01-22:
[image]

Use the latest nightly.

> Either will work when used one at a time, but not together at the same time.

Check if in the logs all `Serial Number/Unique Identifier:` are the same. If yes, then you have to change the capacity of each battery to be individual. +/- 5 Ah does not change much, but help to identify every single battery. You could use 280 and 281 Ah for example.

rldoose, 2024-01-22:
Success!  Changing the battery capacities did the trick.  Thanks so much for the trick.  Would never have guessed that.

Different question if I may:  I have 3 identical batteries with JDB BMSs.  When I only had the one working, I was using that single battery as the controlling BMS  and my SmartShunt as the battery monitor, and that seems to work fine.  Do I really need to use a battery integrator?  What do I gain by using one that I don’t have now?  Would I loose any functionality, e.g. visibility to each batteries cell voltages if I used an integrator?

mr-manuel, 2024-01-22:
> Do I really need to use a battery integrator? What do I gain by using one that I don’t have now?

Yes, else all batteries switch to float, if the controlling BMS is charged. Or if another BMS has an alarm nothing happens.

> Would I loose any functionality, e.g. visibility to each batteries cell voltages if I used an integrator?

No, the aggregador only displays as an additional battery.

closed 2024-01-22 by mr-manuel