Louisvdw/dbus-serialbattery #930: Cerbox GX crashed by USB RS485 original Victron Adapter

mactoolz, 2024-01-21:
### Describe the problem

Hello, I currently have four USB RS485 adapters. They each connect a Seplos battery.
The problem is that the CERBO GX crashes and restarts after an indefinite period of time.

### Driver version

the latest Ninja Version, actual installied from 21.01.2024

### Venus OS device type

Cerbo GX

### Venus OS version

3.13

### BMS type

Seplos

### Cell count

16

### Battery count

4

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 90.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.50
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.40

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;                - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;              As you have to adopt this value to your system, I reccomend to start with
;              OVP voltage - 0.030 (see Example).
;                - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;                  the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
;                - Try to decrease (lower) by 0.005 in steps, if the system hits the OVP too fast, before all
;                  cells could be balanced and the system goes into protection mode multiple times.
; Example: If OVP is 3.650, then start with 3.620 and increase/decrease by 0.005
; Note: The value has to be higher as the MAX_CELL_VOLTAGE
[... 416 lines cut ...]
@4000000065ace83f21ac9524 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065ace83f22243a0c INFO:SerialBattery:Testing Renogy at address "\xf7"
@4000000065ace83f34e655e4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065ace83f354b0354 INFO:SerialBattery:Testing Seplos
@4000000065ace8410640e9fc INFO:SerialBattery:Connection established to Seplos
@4000000065ace841064ec0f4 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB3
@4000000065ace841065c613c INFO:SerialBattery:========== Settings ==========
@4000000065ace84106699bf4 INFO:SerialBattery:> Connection voltage: 53.6V | Current: 0.0A | SoC: 69.8%
@4000000065ace8410738ee2c INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
@4000000065ace8410739153c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000065ace841073920f4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 80.0A | MAX BATTERY DISCHARGE CURRENT: 90.0A
@4000000065ace84107393094 INFO:SerialBattery:> CVCM:     True
@4000000065ace841074e9524 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
@4000000065ace841074eac94 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000065ace8410757dc24 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000065ace84107649dc4 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000065ace84107720374 INFO:SerialBattery:Serial Number/Unique Identifier: SeplosBMS16cells_287.0Ah
@4000000065ace84119ac902c INFO:SerialBattery:DeviceInstance = 5
@4000000065ace84119c30dfc INFO:SerialBattery:com.victronenergy.battery.ttyUSB3
@4000000065ace8412956254c INFO:SerialBattery:publish config values = 1
```

### Any other information that may be helpful

My final result is that i use 8 USB RS485. Is that possible ?

mactoolz, 2024-01-21:
I have just noticed that my fourth adapter does not appear in the current files and no values are displayed

mr-manuel, 2024-01-21:
What is your system load? You can check with `uptime` or `top`.

mactoolz, 2024-01-21:
10:50:55 up 2 min,  load average: 5.95, 2.76, 1.04

mactoolz, 2024-01-21:
```
 1194  1162 root     S    39744   4%  11% {vrmlogger.py} /usr/bin/python3 -u /opt/victronenergy/vrmlogger/vrmlogger.py
 2307  1203 nodered  R     198m  20%  10% node-red
 2422  2398 root     R    39604   4%   8% {dbus_mqtt.py} /usr/bin/python3 -u /opt/victronenergy/dbus-mqtt/dbus_mqtt.py --init-broker
 1242  1221 root     R    22348   2%   5% {dbus_systemcalc} /usr/bin/python3 -u /opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py
  864   862 messageb R     4332   0%   5% dbus-daemon --system --nofork
 2208  1215 root     S<    143m  14%   4% /opt/victronenergy/gui/gui -nomouse -display Multi: LinuxFb: VNC:size=1024x600:depth=32:passwordFile=/data/conf/v
 1782  1265 root     S    23140   2%   4% python /data/BatteryAggregator/battery_service.py
 1273  1252 root     S    21388   2%   3% {dbus_generator.} /usr/bin/python3 -u /opt/victronenergy/dbus-generator-starter/dbus_generator.py
 2421  2404 mosquitt S    10572   1%   2% /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf
 1270  1250 root     S    45544   4%   2% {dbus-modbus-cli} /usr/bin/python3 -u /opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py
 1232  1213 root     S     8756   1%   1% /opt/victronenergy/hub4control/hub4control
 2451  2425 root     R    23500   2%   1% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB2
 1274  1254 root     S    11216   1%   1% /opt/victronenergy/dbus-fronius/dbus-fronius
 2430  2406 root     S     9772   1%   1% /opt/victronenergy/dbus-modbustcp/dbus-modbustcp
 2455  2427 root     R    22316   2%   1% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB1
 2447  2420 root     R    22828   2%   1% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB3
 1954  1845 root     S     3620   0%   1% /opt/victronenergy/mk2-dbus/mk2-dbus --log-before 25 --log-after 25 --banner -w -s /dev/ttyS4 -i -t mk3 --setting
 2448  2424 root     S    22344   2%   1% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB0
 2418  2400 root     S    21444   2%   1% {vesmart_server.} /usr/bin/python3 -u /opt/victronenergy/vesmart-server/vesmart_server.py -i hci1
 2131  2113 root     S     3900   0%   1% /opt/victronenergy/vecan-dbus/vecan-dbus -c socketcan:can0 --banner --log-before 25 --log-after 25 -vv
 1233  1209 root     S    26684   3%   1% {localsettings.p} /usr/bin/python3 -u /opt/victronenergy/localsettings/localsettings.py --path=/data/conf
 1277  1261 root     D     3616   0%   0% /opt/victronenergy/dbus-adc/dbus-adc --banner
 4514  3372 root     R     2780   0%   0% top
 1193  1173 simple-u S    12724   1%   0% /bin/simple-upnpd --xml /var/run/simple-upnpd.xml -d
 2289  1259 root     S     3596   0%   0% /opt/victronenergy/dbus-ble-sensors/dbus-ble-sensors --banner
 2121  2115 root     S     3172   0%   0% /opt/victronenergy/can-bus-bms/can-bus-bms --log-before 25 --log-after 25 -vv -c socketcan:can0 --banner
 1286  1263 root     S    64396   6%   0% python /data/SetupHelper/PackageManager.py
 1236  1217 root     S    19528   2%   0% {dbus_vebus_to_p} /usr/bin/python3 -u /opt/victronenergy/dbus-vebus-to-pvinverter/dbus_vebus_to_pvinverter.py
 1875  1870 root     S     3228   0%   0% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyS5
 1876  1872 root     S     3228   0%   0% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -s /dev/ttyS7
 2132  2127 root     S     3172   0%   0% /opt/victronenergy/can-bus-bms/can-bus-bms --log-before 25 --log-after 25 -vv -c socketcan:can1 --banner
 1268  1245 root     S    15548   2%   0% /opt/victronenergy/dbus-qwacs/dbus-qwacs
 1188  1165 root     S    13612   1%   0% /opt/victronenergy/venus-platform/venus-platform
  867     1 root     S     7592   1%   0% /usr/sbin/haveged -w 1024 -v 1
 3285  2171 root     S     5520   1%   0% sshd: root@pts/0

```

mactoolz, 2024-01-21:
```
10:56:27 up 7 min,  load average: 3.14, 3.23, 1.82

```

mactoolz, 2024-01-21:
It feels like the Cerbo GX crashes every 30 minutes.

The question is what causes the Cerbo to crash?

mr-manuel, 2024-01-21:
Please check this issue https://github.com/Louisvdw/dbus-serialbattery/issues/808

DirkReudenbach, 2024-01-21:
Ive had the same issue and i tried hard with mr-manuels support also #808 doesn't finally help. Nodered + serial driver on the Cerbo crashed on my installation again and again. The final solution to let both work is to run serial driver on the Cerbo and nodered on a dedicated Raspberry pi- but still the CPU load on the Cerbo only with the serial driver is at 50-70%. Think the high CPU load it's related true the 4 Batteries with 4 JKBMS im running. I tested with less than 4 Batteries and the load went down. Due to separate Nodered the system never rebooted anymore

mr-manuel, 2024-01-21:
I rather would recommend using the more powerful Raspberry Pi to connect the BMS, aggregate the data and send the aggregated data to the cerbo. On the cerbo use dbus-mqtt-battery to retrieve the data.

DirkReudenbach, 2024-01-21:
ok thanks for your recommendation - sounds a better solution due to more powerful raspi than mine.

 But how could the setup look like. Which Software should /must run on the raspi to connect the serial driver? Which connection could i use for this? RS485, bluetooth ?

mr-manuel, 2024-01-21:
Install Venus OS on the Raspberry Pi with the dbus-serialbattery driver and a battery aggregator. Install dbus-mqtt-battery on your Cerbo GX. Use Node-RED to fetch the aggregated battery data from the Raspberry Pi, combine the data to one topic and feed the dbus-mqtt-battery with this data.

mactoolz, 2024-01-21:
ok, that is of course very, very weak from the CERBO GX.
But what I see in the htop is that even the dbus mqtt is partially at 50%.

But I'm still not sure whether the performance of the Cerbo should really be questioned here.

I am very poorly positioned at Linux level. Is there any way to log the services that are listed in the "htop"?

mr-manuel, 2024-01-21:
> I am very poorly positioned at Linux level. Is there any way to log the services that are listed in the "htop"?

You could do it with Telegraf, an InfluxDB and visualize it with Grafana.

mr-manuel, 2024-01-22:
> I have just noticed that my fourth adapter does not appear in the current files and no values are displayed

Check if in the logs all `Serial Number/Unique Identifier:` are the same. If yes, then you have to change the capacity of each battery to be individual. +/- 5 Ah does not change much, but help to identify every single battery.

mactoolz, 2024-01-22:
What do you mean exactly ?

Then again the question about the crashes.
Until a few weeks ago, I only had two RS485 adapters in use.

Since I've been using the four adapters, the problems have increased dramatically.

I don't know exactly how to solve the problem?

My main aim is to use 7 of these adapters for the battery packs in the near future.

I currently see the problem that it might not work at all.

What is your opinion on this?

Translated with DeepL.com (free version)

mactoolz, 2024-01-22:
at the moment i see at "htop" that the service from the GUI have 40-55% !!!

mr-manuel, 2024-01-22:
Why the system reboots: https://github.com/Louisvdw/dbus-serialbattery/issues/808#issuecomment-1724240488

How to solve the problem: https://github.com/Louisvdw/dbus-serialbattery/issues/930#issuecomment-1902663820

> What do you mean exactly ?

Every battery has to have another unique identifier else it won't work with the latest driver version. The unique identifier in your case is created from the hardware version and the capacity, since there is no unique value that could be read from your BMS. To make every unique identifier unique the capacity of the batteries can't be the same.

If you have 11 batteries with 280 Ah you could do:
1. battery: 275 Ah
2. battery: 276 Ah
3. battery: 277 Ah
4. battery: 278 Ah
5. battery: 279 Ah
6. battery: 280 Ah
7. battery: 281 Ah
8. battery: 282 Ah
9. battery: 283 Ah
10. battery: 284 Ah
11. battery: 285 Ah

mactoolz, 2024-01-22:
ok, i do it ...

ok, but your suggestion is not the solution to make the system more stable, is it?

I'll take a look at the other topics too.
I have also seen that the GUI takes up a lot of time here.

mactoolz, 2024-01-22:
You say that there is no unique feature for integrating the adapters.

When I look at the attributes, each adapter has the serial number as an attribute.

ATTRS{serial} == "FT8DK8RE"

[image]

mr-manuel, 2024-01-22:
It depends what you exactly mean with "to make the system mor stable". If your system is too weak for the tasks you have to split the tasks to more powerful systems.

> You say that there is no unique feature for integrating the adapters.

I didn't said that. I said that there is no unique data we get from the Seplos BMS via serial connection. If you know how to get some, you are very welcome to improve the driver.

I know that some users have different serial adapters with different ID's, but that's not what we are looking for. The BMS has to have an unique ID, so that if you swap adapter or USB port the VRM ID stays the same.

mactoolz, 2024-01-22:
by more stable I meant whether your suggestion would lead to fewer crashes if you set the capacity of the battery packs to different power ratings.

Wouldn't it be enough to read the serial number from the RS485 to recognize the adapters?

mr-manuel, 2024-01-22:
> by more stable I meant whether your suggestion would lead to fewer crashes if you set the capacity of the battery packs to different power ratings.

No, if you don't do that the latest version of the driver will not load all BMS.

> Wouldn't it be enough to read the serial number from the RS485 to recognize the adapters?

Sure, but this is the theory. If you are able to do so in the practical way feel free to open a PR for that :-) I spend hours and was not able to. Also not all BMS are sending this data.

mactoolz, 2024-01-22:
Hi,

the data does not come from the BMS. The data can be read out via the PI with the command

`
dmesg | grep ttyUSB
`
all USB active adapters are displayed here

`
udevadm info --name=/dev/ttyUSB0 --attribute-walk
`
the command displays the USB attributes for each port
how the adapter is connected.

Maybe it would be a way to give the user this possibility, i.e. to use it on the PI and to be able to enter these attributes in the config file, e.g. the MAC address of the BLE. Since you can
then creates the assignment.

mr-manuel, 2024-01-22:
There is a long history behind that. We will not change that. You can search all issues for `VRM ID` and read the issues to understand it.

closed 2024-01-22 by mr-manuel

mactoolz, 2024-01-30:
?? I dont understand this.

The source code have to detect how many USB Driver are available and so on.

In the config file is it possible to define the name for the usb. 

Ok. Another question. 

Why need only one USB Driver e.g. 3-6% of one USB sriver binding into the VENUS ?