Louisvdw/dbus-serialbattery #936: SOC stuck at 100%

NorthTown2022, 2024-01-23:
### Describe the bug

For unknown reasons my SOC got stuck at 100% with SOC_CALCULATION = TRUE. 

I realized the SOC within BMS was reporting accurately and to test set SOC_CALCULATION = FALSE, the problem went away. I set SOC_CALCULATION = TRUE once again and problem has not yet returned. 

The problem persisted during the installation and uninstallation of various versions of the 1.1.202401** driver. 
When I installed the 1.0.20231117dev driver the problem also went away.

### How to reproduce

This is a screenshot from yesterday morning before I wiped the Pi clean and did a fresh install.

[image]

### Expected behavior

I have no idea but will report back if it happens again. I currently have Mr. Manual's 1.1.20240122dev driver installed but like I said the issue persisted through several variations of the 1.1.202401** driver.

### Driver version

1.1.20240122dev, 1.1.20240102beta 

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
BMS_TYPE = Jkbms
LOGGING = INFO

MAX_BATTERY_CHARGE_CURRENT = 140
MAX_BATTERY_DISCHARGE_CURRENT = 100

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max ba                                                                             >
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V =                                                                             >
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.350

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.620

; --------- SOC calculation ---------
SOC_CALCULATION = TRUE
SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 3000

; Temperature sensor 1 name
TEMP_1_NAME = Temp Cells 4|5

; Temperature sensor 2 name
TEMP_2_NAME = Temp Cells 12|13

LINEAR_LIMITATION_ENABLE = True

CVCM_ENABLE = True
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

; --------- Cell Voltage limitation (affecting CVL) ---------
CVL_ICONTROLLER_MODE = True
CVL_ICONTROLLER_FACTOR = 0.2
```

### Relevant log output

```shell
root@raspberrypi4:~# tail -F /data/log/dbus-serialbattery.ttyUSB2/current | tai64nlocal
2024-01-23 20:38:26.330713500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-01-23 20:38:26.331086500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-01-23 20:38:26.331457500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-01-23 20:38:26.331829500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-01-23 20:38:26.332319500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_B2A20S20P
2024-01-23 20:38:26.484270500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-01-23 20:38:26.589115500 INFO:SerialBattery:DeviceInstance = 1
2024-01-23 20:38:26.589584500 INFO:SerialBattery:Used device instances: ['1']
2024-01-23 20:38:26.589995500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB2
2024-01-23 20:38:26.606093500 INFO:SerialBattery:publish config values = True
```

### Any other information that may be helpful

My coding skills are weak but I'm happy to help. I will do more testing and report back if this happens again.

NorthTown2022, 2024-01-23:
Yikes! I'm not sure what happened there but it's been corrected. Thanks :)

mr-manuel, 2024-01-28:
@NorthTown2022 is the issue solved with the correction of the config typo?

NorthTown2022, 2024-01-28:
The problem has not come back.

closed 2024-01-28 by mr-manuel