Louisvdw/dbus-serialbattery #947: Not switching from absorption to float

NorthTown2022, 2024-02-03:
### Describe the problem

It appears I am falling short a few mV of max voltage (55.2V). I can't figure out why this is happening. I have tried adding (and reverted with a semi colin) a voltage drop,  additionally a wider range for imbalance to help start the timer even though cells seem to be withing spec:
[image]
[image]

### Driver version

1.1.2024.01.28dev (nightly from mr-manuals repository)

### Venus OS device type

Cerbo GX

### Venus OS version

3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
BMS_TYPE = Jkbms
LOGGING = INFO

MAX_BATTERY_CHARGE_CURRENT = 140
MAX_BATTERY_DISCHARGE_CURRENT = 100

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max ba                                                                             >
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V =                                                                             >
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.350

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.620

; --------- SOC calculation ---------
SOC_CALCULATION = TRUE
SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; Temperature sensor 1 name
TEMP_1_NAME = Temp Cells 4|5

; Temperature sensor 2 name
TEMP_2_NAME = Temp Cells 12|13

LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

; --------- Cell Voltage limitation (affecting CVL) ---------
CVL_ICONTROLLER_MODE = True
CVL_ICONTROLLER_FACTOR = 0.2

;CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.030
;VOLTAGE_DROP = 0.10
```

### Relevant log output

```shell
Not sure what logs would help here
```

mdkeil, 2024-02-07:
Do you limited max voltage in DVCC-settings? -- is serialbattery your controlling bms @DVCC-settings? CCL-voltage during charging?

mr-manuel, 2024-02-07:
Delete the `"""` in this part and restart the script. This way you see additional debugging data in the GUIv1 under the `Parameters` page.

https://github.com/Louisvdw/dbus-serialbattery/blob/5e025e66ff4887ee5ce3665789e5d2a14e63ec23/etc/dbus-serialbattery/battery.py#L534-L536

NorthTown2022, 2024-02-08:
> dbus-serialbattery/etc/dbus-serialbattery/battery.py

> Do you limited max voltage in DVCC-settings? -- is serialbattery your controlling bms @DVCC-settings? CCL-voltage during charging?

No to max voltage:
[image]

Yes to serialbattery as controlling bms:
[image]

NorthTown2022, 2024-02-08:
I put this line into putty and edited line 535:
[image]

Thank you. Debugging is now active:
[image]

Since I posted for help I have changed version and config in an effort to fix this problem. I have freshly reinstalled 1.1.2024.01.28dev using original config with the exception:
VOLTAGE_DROP = 0.11

mr-manuel, 2024-02-08:
If you enable `Limit max charge current` then the BMS is not able to dynamically change it. Can you post the debugging screen again at the moment the battery should switch to float?

NorthTown2022, 2024-02-09:
"Limit charge current" is now turned off in DVCC. 

Yes, It will take me a few days to capture the debugging screen at that moment.

NorthTown2022, 2024-02-11:
There is no sun today and had time so I charged battery full with an Eltek Flatpack rectifier. It's modified to output 55.2V. As soon as the cells became balanced to <0.010V the Absorption dynamic timer started its countdown. Here is the moment it switched to transition:
[image]
[image]
Charging with the rectifier, voltage was held in float transition at ~55.2V for quite a long time. I believe because the driver has no influence over the rectifier. After a while I felt it was time to turn the rectifier off, draw the battery down with a load and check for float @ 53.6V. After ~42 minutes it was drawn down to <53.6V and the driver went into float:
[image]
Last week when there was full sun the battery was charged only with PV, and got held at 55.2V for many hours. There is no sun to speak of in the forecast this coming week. Perhaps by keeping the battery nearly full there will be enough PV coming through the clouds to do this test again tomorrow.

NorthTown2022, 2024-02-12:
The sun was out briefly this morning but I lost todays screenshot of the moment "Float Transition" to "Float" occurred. It has definitely switched to "Float" though, so I am happy about this :) I'm puzzled as to why voltage was held at 55.2V on those sunny days last week..

I watched the "time_diff" counter start the moment the min/max cell voltages were within 0.010v of each other. The "time_diff" ran for exactly 900 seconds in "Absorption dynamic" before switching to "Float Transition".

An undetermined amount of time passed before switching into "Float" mode.

In the Example below does the red box refer to "step mode", and the green box "linear mode"?
[image]

If my cells are balanced within 0.010V when "time_diff" starts to count, should "time_diff" run for 300 seconds instead of the full 900? I think In part I am confused about this because "time_diff" ran for 900 seconds and have the following configs set:
LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True

What are the parameters that manage "Float Transition" to "Float"?

mr-manuel, 2024-02-12:
> An undetermined amount of time passed before switching into "Float" mode.

Float transition applies that there is no big voltage change and therefore hughe current spikes.

> If my cells are balanced within 0.010V when "time_diff" starts to count, should "time_diff" run for 300 seconds instead of the full 900? I think In part I am confused about this because "time_diff" ran for 900 seconds and have the following configs set:

https://github.com/Louisvdw/dbus-serialbattery/blob/ce2df1a3fed55ecabc8c5722d742f9c0f27aad6b/etc/dbus-serialbattery/config.default.ini#L145

https://github.com/Louisvdw/dbus-serialbattery/blob/ce2df1a3fed55ecabc8c5722d742f9c0f27aad6b/etc/dbus-serialbattery/battery.py#L406

> What are the parameters that manage "Float Transition" to "Float"?

This is fixed.

https://github.com/Louisvdw/dbus-serialbattery/blob/ce2df1a3fed55ecabc8c5722d742f9c0f27aad6b/etc/dbus-serialbattery/battery.py#L508

NorthTown2022, 2024-02-12:
Sorry I am still a little bit confused. Can you please clarify:

Is the red box "step mode" and the green box "linear mode"? 

Is 300 seconds in green box a typo that should read 900 seconds?

mr-manuel, 2024-02-15:
> Is the red box "step mode" and the green box "linear mode"?

I updated the description so it's more clear.

https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/4fe25b1732fab837a32d7dbd1c12df4f17bfbcb2/etc/dbus-serialbattery/config.default.ini#L107-L131

> Is the red box "step mode" and the green box "linear mode"?

It's was not a typo, but the survivor of an old version where the value was fixed (like the description some lines above say. Now that changed and I corrected it. Thanks for reporting that!

closed 2024-02-16 by mr-manuel