Louisvdw/dbus-serialbattery #953: SOC from Victron SmartShunt

Vitalic66, 2024-02-04:
### Is your feature request related to a problem? Please describe.

Hallo, is it possible to use the SOC reported from Victron SmartShunt instead the internal calculated SOC from JK?

### Describe the solution you'd like

With dbus-spy I can read the SOC from Victron SmartShunt. So where do I have to tell the driver to use and show this value to the Victron system instead of the one reported from JK?

dmsims, 2024-02-05:
On Venus console - Settings, system setup, Battery monitor

Vitalic66, 2024-02-05:
Sure but this doesn't fix the wrong SOC reported to the Victron system.

For example my system, Serial Bat says 75% SOC, as calculated by JKBMS. Victron SmartShunt says 50%. 

If I set a SOC discharge limit in Voctron system to e.g. 50% it would still discharge the battery, until JKBMS reports 50% which might end up in 20% seen by the Victron SmartShunt.

Vitalic66, 2024-02-05:
True but which charge and discharge algorithm is used than? If you chose serial battery as monitor, it uses the values in the driver for charge and discharge limits.

If you set the monitor to SmartShunt, I guess charge and discharge limits are set by the victron algorithm?

The Problem is not, whos shows what in VRM. The problem is, who controls the limits.

dmsims, 2024-02-05:
Think you need to read through the options in config.default.ini

DirkReudenbach, 2024-02-05:
@Vitalic66 if you want the Smart Shunt to control the battery you need to set up the smart shinzo as Battery Monitor (Venus console - Settings, system setup, Battery monitor). This setup controls SOC, current and Voltages of the system
Additionally setup DVCC and setup controlling BMS - this setup will be used the control CVL, CCL and DCL

Vitalic66, 2024-02-05:
@DirkReudenbach thanks for the advice. Always thought that the choosen battery monitor overwrites DVCC settings from serial battery.

mr-manuel, 2024-02-05:
We will not add different SOC as the SOC from the battery itself. With the latest beta/nightly or stable from my branch you can correct the reading from the BMS. Since you have a SmartShunt you can perfectly adjust the BMS readings and make it more accurate.

https://github.com/Louisvdw/dbus-serialbattery/blob/5e025e66ff4887ee5ce3665789e5d2a14e63ec23/etc/dbus-serialbattery/config.default.ini#L152-L172

closed 2024-02-05 by mr-manuel