Louisvdw/dbus-serialbattery #956: Understanding charge current curve

SergioRius, 2024-02-07:
### Describe the problem

I have two banks of 280Ah in 18s. And before I activate the driver as "BMS Controlling" I wanted to fine-tune the charge curve. After adjusting the MAX_BATTERY_CHARGE_CURRENT and MAX_BATTERY_DISCHARGE_CURRENT to less than the theoretical battery and system capabilities, I'm still getting a higher in excess curve in some areas.

[image]

The graph shows this past night grid charge and compared the Multiplus charge curve that was applied (yellow) vs the CCL the driver was reporting (in green).

I'm trying to get rid of the current rebound that occurs @4:40 that lets the CCL at 20A after a cliff of 40A. All of this occurs with the battery at max. voltage and I suppose that would produce a serious overcharge if it was applied.
Please, help me understand how to modify my configuration to get a better absorption and float.

### Driver version

v1.0.20240102beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.14

### BMS type

JKBMS / Heltec BMS

### Cell count

18

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

MAX_BATTERY_CHARGE_CURRENT    = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

MIN_CELL_VOLTAGE   = 2.930
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.400

SOC_RESET_VOLTAGE  = 3.570

SOC_CALCULATION = True
; SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 180

CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.43, 3.40
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.01,  0.2,  0.8,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.00, 3.02
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0, 0.05, 0.07,  0.1,    1

CVL_ICONTROLLER_MODE = True

TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2,   1,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3,   1,  1,  0

CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

BMS_TYPE = Jkbms
```

### Relevant log output

```shell
None
```

mr-manuel, 2024-02-07:
I assume that you are using LFP batteries. Keep the charging and discharging curve in mind. If you charge with 100 A (0.36 C in your case) until 3,40 A the curve is a bit steeper as in the diagram below. Therefore from 3.40 V to 3.45 Volts there is not much time and you have the current earlier, depending on what your goal is.

If you want a smoother curve try something like this:
```ini
CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.40, 3.35
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.01,  0.1,  0.2,    1
```

[image]

SergioRius, 2024-02-09:
Thanks @mr-manuel for your help.

Yes, I'm using LiFePO4 cells. I've tried your settings and those are the results for the past night:
[image]

I think that this curve does not fit, but I could be wrong. What I see is that at 43% I'm already getting almost a 50% of current reduction. At 61% the CCL is 23A.
[image]
I'm losing all that area in the charge curve. I could be wrong I think because if that CCL had been applied, the voltage shouldn't raise so much to start, and could be that it would set at a slightly lower voltage and go on. That would be only try and see.

Then the current lowers to 3A so the shunt would sync the 100% but that occurs at 96%. What still concerns me is that after that 3A valley the CCL starts climbing again, but sets on 11A, that may be Ok.

mr-manuel, 2024-02-10:
The parameters were only an example in order that you see how it behaves. You have to try and see a bit what works best for you.

That the CCL raises again a bit is normal, because you lower the charging amps and therefore the cell voltage drops a bit.

mr-manuel, 2024-02-12:
This screenshot does not look like how the driver normally looks. Are you sure you have installed it correctly?

SergioRius, 2024-02-12:
> This screenshot does not look like how the driver normally looks. Are you sure you have installed it correctly?

Sorry Manuel, this is a screenshot for the BatteryAggregator. Both the serialBattery JKBMS instances show the same data, but half the CCL and DCL.

mr-manuel, 2024-02-12:
No, the dbus-serialbattery shows much more informations. I can only help you with the dbus-serialbattery.

SergioRius, 2024-02-14:
> No, the dbus-serialbattery shows much more informations. I can only help you with the dbus-serialbattery.

[image]
[image]

It seems that if one bank reaches max voltage before the other, the BatteryAggregator sets the lower CVL of the two as the aggregated battery CVL and then one bank never goes to absorption and never balances.

[image]

Do you think there's something to be done to prevent this situation?

mr-manuel, 2024-02-15:
The battery aggregator should not switch to float until all batteries switched to float. Open an issue in the repository of the battery aggregator you are using.

Check also https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/52

SergioRius, 2024-02-15:
> The battery aggregator should not switch to float until all batteries switched to float. Open an issue in the repository of the battery aggregator you are using.
> 
> Check also [Dr-Gigavolt/dbus-aggregate-batteries#52](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/52)

I currently use https://github.com/pulquero/BatteryAggregator.
I've read all your conversation in the repo you attached and will see if it's better for me to switch to it. Taking into account how the system goes when a controlling BMS goes offline.

In the meanwhile, I've been happy with the Controlling BMS feature. Other that the balance time being so short for my likes.
[image]
* Two days charging from grid from the same starting %.

mr-manuel, 2024-02-15:
So all questions are answered?