Louisvdw/dbus-serialbattery #961: High Cerbo CPU usage

SergioRius, 2024-02-09:
### Describe the problem

I'm having a steady 50% CPU usage on the Cerbo. 
[image]

I've done the following steps to try to isolate the culprit. A top command doesn't really show the process that it's consuming and rather shows the GUI as the most consuming process.
Stopping all the external services that request from the Cerbo, like mqtt broker or node-red. Had no effect.
Stopping the BatteryAggregator by Pulquero only had a small impact (4-5%)

I did a uninstall of the driver, and that reduced the CPU usage to <15%. This is a parallel system, with grid meter, mppt and pv inverter.
[image]

Derived problems:
The Cerbo performance at some moments, due to VRM, GUI loads, etc... reduces enough to not being able to request the grid meter the readings in the minimum pace required by Victron. Then the meter shows offline for an instant and the systems goes pass-through.
[image]
Could or not could be caused or worsen by this.

Since I installed the driver, also experienced in the VRM the message: "The system is busy so realtime updates had been disabled".

I need more instructions for diagnosing or solving the problem.

### Driver version

v1.0.20240102beta

### Venus OS device type

Cerbo GX

### Venus OS version

3.14

### BMS type

JKBMS / Heltec BMS

### Cell count

18

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

MAX_BATTERY_CHARGE_CURRENT    = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

MIN_CELL_VOLTAGE   = 2.930
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.400

SOC_RESET_VOLTAGE  = 3.570

; No se aplica en modo Linear
; MAX_VOLTAGE_TIME_SEC = 3600
; SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

SOC_CALCULATION = True
; SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 180

; --------- Cell Voltage Current limitation (affecting CCL/DCL) ---------

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = True
; Discharge current control management referring to cell-voltage enable (True/False).
; !!! Dejar Activo solo porque la reducción está por debajo del voltaje admisible
DCCM_CV_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING   = 3.55, 3.50, 3.45, 3.40, 3.38
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.01, 0.08,  0.2,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.00, 3.02
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0, 0.05, 0.07,  0.1,    1

; --------- Cell Voltage limitation (affecting CVL) ---------

CVL_ICONTROLLER_MODE = True

; --------- Temperature limitation (affecting CCL/DCL) ---------

TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2,   1,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3,   1,  1,  0

; --------- SOC limitation (affecting CCL/DCL) ---------

CCCM_SOC_ENABLE = False
DCCM_SOC_ENABLE = False

BMS_TYPE = Jkbms

; CUSTOM_BATTERY_NAMES = /dev/ttyUSB1:JKBMS_1,/dev/ttyUSB2:JKBMS_2
```

### Relevant log output

```shell
Nothing wrong on logs. Will post if requested.
```

mr-manuel, 2024-02-12:
Take a look at https://github.com/Louisvdw/dbus-serialbattery/issues/808#issuecomment-1723817057

SergioRius, 2024-02-12:
> Take a look at [#808 (comment)](https://github.com/Louisvdw/dbus-serialbattery/issues/808#issuecomment-1723817057)

Thanks Manuel for your help. I've installed htop as per the link instructions and I'm attaching a screenshot below.
This is on a system that hasn't been rebooted for a while. I don't have any screen attached and usually I'm having the remote console open in a browser while I'm working. No node-red and this system was factory reset'd few weeks ago before adding a 2nd Multiplus to my system. Other than serialBattery and BatteryAggregator it doesn't have anything more.

[image]

What I don't understand is if I uninstalled the driver and the load reduced inmediately, why is not showing it's real CPU load on top?
Could be that the driver does something constantly to the GUI that makes it consume more CPU?
What do you think?

mr-manuel, 2024-02-12:
> I don't have any screen attached and usually I'm having the remote console open in a browser while I'm working.

As far as I know is the screen always displayed, even if you have no screen attached. If you connect to the old GUI, then you open a VNC connection in a browser.

>What I don't understand is if I uninstalled the driver and the load reduced inmediately,

Since the driver provides a lot of informations, a lot of informations are processed also in the GUI which probably causes the load.

> why is not showing it's real CPU load on top?

What do you mean with that? Probably CPU spikes are not always shown, therefore check the CPU time.

> Now I have rebooted and the dbus_mqtt establishes as the more time consuming, and cpu moves 60-75%

This was reworked by VIctron and in a future release Mosquitto is replaces by Flash MQ. Also the dbus_mqtt should be replaced, if I remember correctly. All data on the dbus is pushed to mqtt by dbus_mqtt, which causes the load, since the driver produces a lot of data, which is constantly changing.

Honusnap, 2024-02-14:
Here is mine ... 2 x jkbms_ble .. why ?

[image]

mr-manuel, 2024-02-15:
@Honusnap with F5 you can switch to tree mode. There you see the processes with their threads. The JKBMS BLE driver creates an additional thread to watchdog the main loop.

Honusnap, 2024-02-15:
> @Honusnap with F5 you can switch to tree mode. There you see the processes with their threads. The JKBMS BLE driver creates an additional thread to watchdog the main loop.

Hooo make sense for the watchdog, thanks for pointing that out.

After looking more closely to those loads, clearly there is something to do on dbus-serialbattery to be able to reduce this load, but the screen display also have a pretty big load and for example load on thje main page (graphic display with boxes) is cleary heavier then some menus => GuiMods have an impact on load that is not that small.

Having those dbus-fronius running while i got no fronius .. ... why victron ?

mr-manuel, 2024-02-15:
> clearly there is something to do on dbus-serialbattery to be able to reduce this load

This would take a huge amount of hours to debug and optimize this for only a little performance gain. The driver fetches, elaborates and writes a lot of data. This every second for most BMS, for some BMS every half second or every two seconds.
If you have a 16s battery then it processes every cell voltage, balancing status of every cell and much more.

> Having those dbus-fronius running while i got no fronius .. ... why victron ?

Because they support it natively and have to automatically recognize it. You could stop or remove the service.

To stop the service: `svc -t /service/dbus-fronius`

To get the list of the services: `ls -l /service/`

You could try to disable all you don't need and see what happens. I have no experience in that. Some others would appreciate for sure, if you create a list with not needed services that can be disabled.

kommando828, 2024-02-15:
Turning off the Gui on a Raspbian Pi has fixed high CPU issues for me in the past with other data hungry applications, Web access to Node-red and access to remote Terminal was unaffected so possible web access to Cerbo would still work, you would not have the screens showing on the LCD and be left with a terminal window so you could turn the GUI back on.

mr-manuel, 2024-02-15:
If you turn off the GUI you cannot connect over web to GUIv1. If you use GUIv2 then it works, since it get's it's data from MQTT. GUIv1 is a VNC connection to the screen more or less.

Honusnap, 2024-02-15:
> This would take a huge amount of hours to debug and optimize this for only a little performance gain. The driver fetches, elaborates and writes a lot of data. This every second for most BMS, for [...]

If data fetching is done every second then doing it every 5 sec would reduce the load by 4 (some things are not linearly dependent on fetching frequency)

> Because they support it natively and have to automatically recognize it. You could stop or remove the service.
Maybe an option in the interface would be smart from victron ?

I will disable things to see what happend yes.

Honusnap, 2024-02-16:
> This would take a huge amount of hours to debug and optimize this for only a little performance gain. The driver fetches, elaborates and writes a lot of data. This every second for most BMS, for [...]

In 2 days : 48h, i got 8hours for serialbattery, that's a constant 16% of load ... it's gigantic !

closed 2024-02-19 by mr-manuel

SergioRius, 2024-02-26:
CCL Doesn't need to be updated as often, for example. I'm near 100% charge and the pv inverter and mppt are going crazy with CCL jumping from 9A to 20A per bank. And it was worse some minutes ago.

mr-manuel, 2024-02-26:
https://github.com/Louisvdw/dbus-serialbattery/blob/ce2df1a3fed55ecabc8c5722d742f9c0f27aad6b/etc/dbus-serialbattery/config.default.ini#L100-L104