Louisvdw/dbus-serialbattery #963: Daly BMS 16s - Installed/available Capacity

legitas84, 2024-02-10:
### Describe the problem

Hi all,

iam curious about the shown "Installed/Available" Capacity.
BMS was configured correctly and already working fine. 

i made a change in config (once in config, also tried in config.default) on the parameter "BATTERY_CAPACITY = 280"

Questions: what is Parameter showing (installierte/verfügbare kapacität)? Should it show 280? or is it meant for something else?
if it should show 280AH, somehow its not working for me.

[image]

### Driver version

1.0.20230531

### Venus OS device type

Cerbo GX

### Venus OS version

3.14

### BMS type

Daly Smart BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BATTERY_CAPACITY = 280
```

### Relevant log output

```shell
@4000000065c70d950265d694 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065c70dbb0074052c INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065c70dc0115c9a5c INFO: Start service gps-dbus.ttyUSB0 once
@4000000065c70dc9297d066c INFO: Start service vedirect-interface.ttyUSB0 once
@4000000065c70dce3aa461ec INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065c70df43168857c INFO: Start service dbus-cgwacs.ttyUSB0 once
@4000000065c70dfa0596d6ec INFO: Start service gps-dbus.ttyUSB0 once
@4000000065c70e0328af562c INFO: Start service vedirect-interface.ttyUSB0 once
@4000000065c70e083b60206c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065c70e2e34aa9cd4 INFO: Start service dbus-cgwacs.ttyUSB0 once
[... 225 lines cut ...]
@4000000065c780d431724594 INFO: Create daemontools service mk2-dbus.ttyS4
@4000000065c780d60e216c8c INFO: Create daemontools service vedirect-interface.ttyS5
@4000000065c780d72d1ce1ac INFO: Create daemontools service vedirect-interface.ttyS6
@4000000065c780d9182f8164 INFO: Create daemontools service vedirect-interface.ttyS7
@4000000065c780db03abdb84 INFO: Start service mk2-dbus.ttyS4
@4000000065c780db17d320f4 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000065c780dc1ad5685c INFO: Start service vedirect-interface.ttyS5
@4000000065c780dd02bc5eec INFO: Create daemontools service dbus-cgwacs.ttyUSB1
@4000000065c780dd3511887c INFO: Start service vedirect-interface.ttyS6
@4000000065c780df25489fd4 INFO: Start service vedirect-interface.ttyS7
@4000000065c780e129d85184 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065c780e310675d1c INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000065c780ea385d216c INFO: Create daemontools service gps-dbus.ttyUSB1
@4000000065c780f112869f54 INFO: Start service gps-dbus.ttyUSB1 once
@4000000065c780fb2fedcc5c INFO: Create daemontools service vedirect-interface.ttyUSB1
@4000000065c7810136fe58a4 INFO: Start service vedirect-interface.ttyUSB1 once
@4000000065c7810511976524 INFO: Create daemontools service dbus-serialbattery.ttyUSB1
@4000000065c7810b17c4e854 INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000065c781151295975c INFO: Create daemontools service gps-dbus.ttyUSB0
@4000000065c7811b17dba4a4 INFO: Start service gps-dbus.ttyUSB0 once
@4000000065c781243490a464 INFO: Create daemontools service vedirect-interface.ttyUSB0
@4000000065c7812a36def194 INFO: Start service vedirect-interface.ttyUSB0 once
@4000000065c7812e1074f594 INFO: Create daemontools service dbus-cgwacs.ttyUSB0
@4000000065c7813412cbbea4 INFO: Start service dbus-cgwacs.ttyUSB0 once
```

mr-manuel, 2024-02-10:
It‘s only used, if your Daly BMS does not provide the value. In your case it‘s reading the value from the BMS. Have you setup the BMS correctly with the PC software?

legitas84, 2024-02-10:
I used the Mobile APP. So you mean I have to perform again with Desktop Software ?

mr-manuel, 2024-02-11:
Yes.

legitas84, 2024-02-11:
Ok, did it yesterday. Do I have to connect the UART /RS cable with BMS or Daly active balancer (to cerbo). Or it doesn’t matter?

mr-manuel, 2024-02-11:
Unfortunately I have no Daly BMS so I can't help you further. You could try to search all issues and discussions for Daly. There are lot's of informations which could help you.

legitas84, 2024-02-11:
Thx 4 Support

legitas84, 2024-02-11:
Fine

mr-manuel, 2024-02-11:
This means that you still need support or I can close the issue?

legitas84, 2024-02-11:
Can be closed

closed 2024-02-11 by mr-manuel