Louisvdw/dbus-serialbattery #964: It does not start to charge or discharge

kraber2, 2024-02-10:
### Describe the problem

I think I Installed it the correkt way. It shows the Battery BMS (JKBMS) with all information but it does not start to charge or discharge. The Battery SoC is up to 60%.

[image]

[image]

[image]

[image]

[image]

[image]

### Driver version

1.0.20231117dev

### Venus OS device type

Raspberry Pi

### Venus OS version

V.2.92

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting>
MAX_BATTERY_CHARGE_CURRENT = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E2:95:58
MIN_CELL_VOLTAGE   = 3.000
MAX_CELL_VOLTAGE   = 3.450
```

### Relevant log output

```shell
2024-02-10 13:29:38.916365500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-10 13:29:38.916936500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2024-02-10 13:29:39.200030500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 13:29:39.200324500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 13:29:43.709900500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 13:29:43.710618500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-02-10 13:29:43.710964500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce29558
2024-02-10 13:29:43.711243500 INFO:SerialBattery:========== Settings ==========
2024-02-10 13:29:43.711593500 INFO:SerialBattery:> Connection voltage: 51.64V | Current: 0.0A | SoC: 60%
2024-02-10 13:29:43.711877500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-02-10 13:29:43.712186500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-02-10 13:29:43.712484500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2024-02-10 13:29:43.712770500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2024-02-10 13:29:43.713034500 INFO:SerialBattery:> CVCM:     True
2024-02-10 13:29:43.713335500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX CELL VOLTAGE: 3.45V
2024-02-10 13:29:43.713613500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-02-10 13:29:43.713887500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-02-10 13:29:43.714169500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-02-10 13:29:43.714449500 INFO:SerialBattery:Serial Number/Unique Identifier: 2032809148
2024-02-10 13:29:43.746107500 INFO:SerialBattery:DeviceInstance = 2
2024-02-10 13:29:43.746459500 INFO:SerialBattery:com.victronenergy.battery.c8478ce29558
2024-02-10 13:29:43.746981500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 13:29:43.765930500 INFO:SerialBattery:publish config values = 1
2024-02-10 13:50:36.854305500 *** CCGX booted (0) ***
2024-02-10 13:51:35.872644500 
2024-02-10 13:51:35.872648500 INFO:Bluetooth details
2024-02-10 13:51:35.914732500 Attempting to disconnect from C8:47:8C:E2:95:58
2024-02-10 13:51:35.914736500 Successful disconnected
2024-02-10 13:51:40.970019500 Device C8:47:8C:E2:95:58 (public)
2024-02-10 13:51:40.970024500 	Alias: JK-B1A20S15P
2024-02-10 13:51:40.970026500 	Paired: no
2024-02-10 13:51:40.970029500 	Trusted: no
2024-02-10 13:51:40.970031500 	Blocked: no
2024-02-10 13:51:40.970033500 	Connected: no
2024-02-10 13:51:40.970035500 	LegacyPairing: no
2024-02-10 13:51:40.970038500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-10 13:51:40.970041500 	RSSI: -59
2024-02-10 13:51:40.971002500 
2024-02-10 13:51:41.652139500 INFO:SerialBattery:
2024-02-10 13:51:41.652394500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-10 13:51:41.653022500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2024-02-10 13:51:41.961775500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 13:51:41.962062500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 13:51:46.470525500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 13:51:46.471209500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-02-10 13:51:46.471547500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce29558
2024-02-10 13:51:46.471805500 INFO:SerialBattery:========== Settings ==========
2024-02-10 13:51:46.472126500 INFO:SerialBattery:> Connection voltage: 51.64V | Current: 0.0A | SoC: 60%
2024-02-10 13:51:46.472391500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-02-10 13:51:46.472737500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-02-10 13:51:46.473056500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2024-02-10 13:51:46.473333500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2024-02-10 13:51:46.473598500 INFO:SerialBattery:> CVCM:     True
2024-02-10 13:51:46.473877500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX CELL VOLTAGE: 3.45V
2024-02-10 13:51:46.474149500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-02-10 13:51:46.474421500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-02-10 13:51:46.474690500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-02-10 13:51:46.474964500 INFO:SerialBattery:Serial Number/Unique Identifier: 2032809148
2024-02-10 13:51:46.504606500 INFO:SerialBattery:DeviceInstance = 2
2024-02-10 13:51:46.505094500 INFO:SerialBattery:com.victronenergy.battery.c8478ce29558
2024-02-10 13:51:46.505667500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 13:51:46.524141500 INFO:SerialBattery:publish config values = 1
2024-02-10 14:02:25.981679500 
2024-02-10 14:02:25.981684500 INFO:Bluetooth details
2024-02-10 14:02:26.005639500 Attempting to disconnect from C8:47:8C:E2:95:58
2024-02-10 14:02:26.005644500 Successful disconnected
2024-02-10 14:02:31.060152500 Device C8:47:8C:E2:95:58 (public)
2024-02-10 14:02:31.060157500 	Alias: JK-B1A20S15P
2024-02-10 14:02:31.060160500 	Paired: no
2024-02-10 14:02:31.060162500 	Trusted: no
2024-02-10 14:02:31.060164500 	Blocked: no
2024-02-10 14:02:31.060166500 	Connected: no
2024-02-10 14:02:31.060169500 	LegacyPairing: no
2024-02-10 14:02:31.060171500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-10 14:02:31.060174500 	RSSI: -67
2024-02-10 14:02:31.061363500 
2024-02-10 14:02:31.757719500 INFO:SerialBattery:
2024-02-10 14:02:31.757991500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-10 14:02:31.758551500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2024-02-10 14:02:32.044049500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 14:02:32.044497500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E2:95:58
2024-02-10 14:02:36.559287500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 14:02:36.559293500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-02-10 14:02:36.559296500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce29558
2024-02-10 14:02:36.559300500 INFO:SerialBattery:========== Settings ==========
2024-02-10 14:02:36.559303500 INFO:SerialBattery:> Connection voltage: 51.64V | Current: 0.0A | SoC: 60%
2024-02-10 14:02:36.559306500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-02-10 14:02:36.559380500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-02-10 14:02:36.559384500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2024-02-10 14:02:36.559388500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2024-02-10 14:02:36.559392500 INFO:SerialBattery:> CVCM:     True
2024-02-10 14:02:36.559395500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX CELL VOLTAGE: 3.45V
2024-02-10 14:02:36.559422500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-02-10 14:02:36.559426500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-02-10 14:02:36.559429500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-02-10 14:02:36.559432500 INFO:SerialBattery:Serial Number/Unique Identifier: 2032809148
2024-02-10 14:02:36.596957500 INFO:SerialBattery:DeviceInstance = 2
2024-02-10 14:02:36.597313500 INFO:SerialBattery:com.victronenergy.battery.c8478ce29558
2024-02-10 14:02:36.598113500 INFO:SerialBattery:BAT: JKBMS 10.XG 16 cells (20220627)
2024-02-10 14:02:36.635244500 INFO:SerialBattery:publish config values = 1
```

DirkReudenbach, 2024-02-10:
can you post a picture of serial battery/ PARAMETER?

mr-manuel, 2024-02-10:
Additionally how did you connect your whole system? What have you selected under Settings -> ESS -> Mode?

mdkeil, 2024-02-10:
Whats your DVCC/Controlling-BMS?

mr-manuel, 2024-02-12:
Any updates? Why did you enable SCS?