Louisvdw/dbus-serialbattery #970: JBD_BLE 2 batteries with same shared VRM Instance

AlexdeKairos, 2024-02-12:
### Describe the bug

Im testing 1.1.20240211dev and found that my both batteries share the same VRM Instace so only one can be seen in VRM and both must share same name. If any of the battery names ar changed on cerbo, both get the same name.

### How to reproduce

Just get to BLE KBD batteries and check under "Device" the VRM version. Both batteries share the same instance number.

### Expected behavior

Each battery must have a unique VRM instance. ie 2 and 3 (that behaivour was correct up to version 

### Driver version

1.1.20240211dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.14

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

2

### Connection type

Bluetooth

### Config file

```ini
; --------- Charge Voltage limitation (affecting CVL) ---------
; Description: Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;              voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
;                    SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
;         Linear mode: After max voltage is reachend and cell voltage difference is smaller or equal to
;                      CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL it switches to float voltage after 300 (fixed)
;                      additional seconds.
;                      After cell voltage difference is greater or equal to CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT
;                      OR
;                      SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT
;                      it switches back to max voltage.
; Example: The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
;          float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
;          once below 80%
;          OR
;          The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
;          voltage of 53.6V after 300 additional seconds to don't stress the batteries. Allow max voltage of
;          55.2V again if max cell difference is above 0.080V or SoC below 80%.
; Charge voltage control management enable (True/False).
CVCM_ENABLE = False

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = False
; Discharge current control management referring to cell-voltage enable (True/False).
[... 163 lines cut ...]
2024-02-12 10:09:34.193409500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2024-02-12 10:09:34.194095500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 160.0A | MAX BATTERY DISCHARGE CURRENT: 160.0A (read from BMS)
2024-02-12 10:09:34.194696500 INFO:SerialBattery:> CVCM:     False
2024-02-12 10:09:34.195562500 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX CELL VOLTAGE: 3.6V
2024-02-12 10:09:34.196150500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2024-02-12 10:09:34.197144500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2024-02-12 10:09:34.197942500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-02-12 10:09:34.198871500 INFO:SerialBattery:Serial Number/Unique Identifier: 28_460.0Ah
2024-02-12 10:09:39.697993500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-02-12 10:09:39.959747500 INFO:SerialBattery:DeviceInstance = 1
2024-02-12 10:09:39.960568500 INFO:SerialBattery:Used device instances: ['1']
2024-02-12 10:09:39.961428500 INFO:SerialBattery:com.victronenergy.battery.blea5c2370595b6
2024-02-12 10:09:41.641660500 INFO:SerialBattery:publish config values = True
2024-02-12 10:09:43.883375500 ERROR:SerialBattery:Exception occurred: KeyError('/CurrentAvg') of type <class 'KeyError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #973
2024-02-12 10:09:44.478121500 INFO:SerialBattery:Saved AllowMaxVoltage. Before False, after False
2024-02-12 10:09:51.365489500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -60
2024-02-12 10:09:51.365498500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -60
2024-02-12 10:12:43.374851500 ERROR:SerialBattery:Exception occurred: KeyError('/CurrentAvg') of type <class 'KeyError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #973
2024-02-12 10:15:43.651986500 ERROR:SerialBattery:Exception occurred: KeyError('/CurrentAvg') of type <class 'KeyError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #973
2024-02-12 10:17:46.509548500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 40 expected 41
2024-02-12 10:18:43.817317500 ERROR:SerialBattery:Exception occurred: KeyError('/CurrentAvg') of type <class 'KeyError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #973
2024-02-12 10:21:31.763142500 ERROR:SerialBattery:>>> ERROR: BMS send insufficient data. Received 40 expected 41
2024-02-12 10:21:43.047781500 ERROR:SerialBattery:Exception occurred: KeyError('/CurrentAvg') of type <class 'KeyError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #973
2024-02-12 10:24:03.042772500 *** CCGX booted (0) ***
```

mr-manuel, 2024-02-12:
Please test the latest dev version of my repository.

AlexdeKairos, 2024-02-13:
I can confirm v1.1.20240212dev fixes de issue.
Thanks!

mr-manuel, 2024-02-13:
You're welcome.

closed 2024-02-13 by mr-manuel

AlexdeKairos, 2024-02-13:
Actually, your version is giving me a reallly hard time to connect to the JBD-BLE batteries.After the initial reset all was working fine, but I did an additional reset and is not able to mantain the connection with the batteries.

may connect briefly, but disconnects quickly.

Should I open a new bug report?

AlexdeKairos, 2024-02-13:
2024-02-13 15:04:26.877957500 [CHG] Device A5:C2:37:05:95:9E RSSI is nil
2024-02-13 15:04:26.877959500 [CHG] Device 66:62:EE:FF:AB:AE TxPower is nil
2024-02-13 15:04:26.877962500 [CHG] Device 66:62:EE:FF:AB:AE RSSI is nil
2024-02-13 15:04:26.878086500 [CHG] Device FC:97:F8:5D:A2:EC RSSI is nil
2024-02-13 15:04:26.878090500 [CHG] Device E9:CC:15:95:BA:68 RSSI is nil
2024-02-13 15:04:26.878093500 [CHG] Controller 08:BF:B8:56:B7:50 Discovering: no
2024-02-13 15:04:26.878096500 Successful disconnected
2024-02-13 15:04:32.033506500 Device A5:C2:37:05:95:B6 (public)
2024-02-13 15:04:32.033516500   Alias: Bateria Serv 2
2024-02-13 15:04:32.033519500   Paired: no
2024-02-13 15:04:32.033520500   Trusted: no
2024-02-13 15:04:32.033521500   Blocked: no
2024-02-13 15:04:32.033523500   Connected: no
2024-02-13 15:04:32.033524500   LegacyPairing: no
2024-02-13 15:04:32.033525500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-13 15:04:32.033528500   RSSI: -60
2024-02-13 15:04:32.034522500
2024-02-13 15:04:33.410003500 INFO:SerialBattery:
2024-02-13 15:04:33.411058500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-13 15:04:33.413115500 INFO:SerialBattery:Venus OS v3.14
2024-02-13 15:04:33.413765500 INFO:SerialBattery:dbus-serialbattery v1.1.20240212dev
2024-02-13 15:04:33.923307500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-13 15:04:33.924182500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-13 15:04:38.935206500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2024-02-13 15:04:38.938399500 ERROR:SerialBattery:No BMS found at A5:C2:37:05:95:B6
2024-02-13 15:04:38.939093500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-02-13 15:04:39.993888500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -64
2024-02-13 15:04:39.993899500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -60
2024-02-13 15:04:45.628097500 INFO:SerialBattery:BLE client disconnected
2024-02-13 15:04:45.966756500
2024-02-13 15:04:45.967154500 INFO:Bluetooth details
2024-02-13 15:04:46.191036500 Attempting to disconnect from A5:C2:37:05:95:B6
2024-02-13 15:04:46.191046500 [CHG] Device 5C:49:7D:77:F8:23 RSSI is nil
2024-02-13 15:04:46.191049500 [DEL] Device 5C:49:7D:77:F8:23 5C-49-7D-77-F8-23
2024-02-13 15:04:46.191052500 [CHG] Device A5:C2:37:05:95:9E RSSI is nil
2024-02-13 15:04:46.191055500 [CHG] Device 5A:72:A0:F5:72:92 RSSI is nil
2024-02-13 15:04:46.191057500 [CHG] Device 4A:D5:08:87:60:83 TxPower is nil
2024-02-13 15:04:46.191060500 [CHG] Device 4A:D5:08:87:60:83 RSSI is nil
2024-02-13 15:04:46.191178500 [CHG] Device 04:52:C7:B8:5F:6D TxPower is nil
2024-02-13 15:04:46.191182500 [CHG] Device 04:52:C7:B8:5F:6D RSSI is nil
2024-02-13 15:04:46.191185500 [CHG] Device FC:97:F8:5D:A2:EC RSSI is nil
2024-02-13 15:04:46.191188500 [CHG] Device 74:0A:67:AC:47:40 RSSI is nil
2024-02-13 15:04:46.191191500 [CHG] Device A5:C2:37:05:95:B6 RSSI is nil
2024-02-13 15:04:46.191193500 [CHG] Device 76:C9:2B:87:C5:26 TxPower is nil
2024-02-13 15:04:46.191220500 [CHG] Device 76:C9:2B:87:C5:26 RSSI is nil
2024-02-13 15:04:46.191223500 [CHG] Device E9:CC:15:95:BA:68 RSSI is nil
2024-02-13 15:04:46.191226500 [CHG] Controller 08:BF:B8:56:B7:50 Discovering: no
2024-02-13 15:04:46.191229500 Successful disconnected
2024-02-13 15:04:51.321739500 Device A5:C2:37:05:95:B6 (public)
2024-02-13 15:04:51.321748500   Alias: Bateria Serv 2
2024-02-13 15:04:51.321750500   Paired: no
2024-02-13 15:04:51.321751500   Trusted: no
2024-02-13 15:04:51.321753500   Blocked: no
2024-02-13 15:04:51.321754500   Connected: no
2024-02-13 15:04:51.321755500   LegacyPairing: no
2024-02-13 15:04:51.321757500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-13 15:04:51.321760500   RSSI: -64
2024-02-13 15:04:51.323920500
2024-02-13 15:04:52.865894500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -64
2024-02-13 15:04:52.874160500 INFO:SerialBattery:
2024-02-13 15:04:52.875254500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-13 15:04:52.879972500 INFO:SerialBattery:Venus OS v3.14
2024-02-13 15:04:52.881121500 INFO:SerialBattery:dbus-serialbattery v1.1.20240212dev
2024-02-13 15:04:53.476428500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-13 15:04:53.477336500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-13 15:04:58.491714500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2024-02-13 15:04:58.496369500 ERROR:SerialBattery:No BMS found at A5:C2:37:05:95:B6
2024-02-13 15:04:58.497444500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-02-13 15:05:00.183950500 ERROR:SerialBattery:BleakClient(): Exception occurred: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #100
2024-02-13 15:05:00.574527500
[... 15 lines cut ...]
2024-02-13 15:05:05.886259500   Connected: no
2024-02-13 15:05:05.886260500   LegacyPairing: no
2024-02-13 15:05:05.886262500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-13 15:05:05.886265500   RSSI: -60
2024-02-13 15:05:05.888171500
2024-02-13 15:05:07.219088500 INFO:SerialBattery:
2024-02-13 15:05:07.220433500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-13 15:05:07.224392500 INFO:SerialBattery:Venus OS v3.14
2024-02-13 15:05:07.225713500 INFO:SerialBattery:dbus-serialbattery v1.1.20240212dev
[... cut ...]
2024-02-13 15:05:19.685757500 Attempting to disconnect from A5:C2:37:05:95:B6
2024-02-13 15:05:19.685766500 [CHG] Device FC:97:F8:5D:A2:EC RSSI is nil
2024-02-13 15:05:19.685770500 [CHG] Device A5:C2:37:05:95:B6 RSSI is nil
2024-02-13 15:05:19.685773500 [CHG] Device E9:CC:15:95:BA:68 RSSI is nil
2024-02-13 15:05:19.685775500 [CHG] Device A5:C2:37:05:95:9E RSSI is nil
2024-02-13 15:05:19.685778500 [CHG] Controller 08:BF:B8:56:B7:50 Discovering: no
2024-02-13 15:05:19.685781500 Successful disconnected