Louisvdw/dbus-serialbattery #971: JBD_BLE battery temperature shown as MOSFET temperature in GUI

AlexdeKairos, 2024-02-12:
### Is your feature request related to a problem? Please describe.

I Can't use new temperature protections functionality due to wrong BMS temperature assignments. Seems that driver detects first temperature sensor as MOSFET temperature but actually many batteries in the market instead of MOSFET NTC, they have  just a termoswitch that triggers upon certain temperature. So in those BMSs, the temperature sensor is attached to the cells and not the MOS. My BMS is for instance, JBD DP04S007. Quite common model.

### Describe the solution you'd like

Option to select if NTC 1 Correspond to a Cell or MOSFET

mr-manuel, 2024-02-12:
Since there are a few BMS that have a different temperature sensor order maybe it would be an idea that there is a default sensor association which you can overwrite in the config file.

mr-manuel, 2024-02-23:
Can you please share a screenshot of the `dbus-spy` where we can see `/System` values?

AlexdeKairos, 2024-02-24:
com.victronenergy.system
Ac/ActiveIn/Frequency                                                                                                        50.1026
Ac/ActiveIn/L1/Current                                                                                                          1.96
Ac/ActiveIn/L1/Power                                                                                                             463
Ac/ActiveIn/L1/Voltage                                                                                                         245.3
Ac/ActiveIn/L2/Current                                                                                                             -
Ac/ActiveIn/L2/Power                                                                                                               -
Ac/ActiveIn/L2/Voltage                                                                                                             -
Ac/ActiveIn/L3/Current                                                                                                             -
Ac/ActiveIn/L3/Power                                                                                                               -
Ac/ActiveIn/L3/Voltage                                                                                                             -
Ac/ActiveIn/NumberOfPhases                                                                                                         1
Ac/ActiveIn/Source                                                                                                                 3
Ac/Consumption/Frequency                                                                                                     49.8214
Ac/Consumption/L1/Current                                                                                                        0.4
Ac/Consumption/L1/Power                                                                                                          112
Ac/Consumption/L1/Voltage                                                                                                      245.3
Ac/Consumption/L2/Current                                                                                                          -
Ac/Consumption/L2/Power                                                                                                            -
Ac/Consumption/L2/Voltage                                                                                                          -
Ac/Consumption/L3/Current                                                                                                          -
Ac/Consumption/L3/Power                                                                                                            -
Ac/Consumption/L3/Voltage                                                                                                          -
Ac/Consumption/NumberOfPhases                                                                                                      1
Ac/ConsumptionOnInput/Frequency                                                                                              50.1026
Ac/ConsumptionOnInput/L1/Current                                                                                                   0
Ac/ConsumptionOnInput/L1/Power                                                                                                     0
Ac/ConsumptionOnInput/L1/Voltage                                                                                               245.3
Ac/ConsumptionOnInput/L2/Current                                                                                                   -
Ac/ConsumptionOnInput/L2/Power                                                                                                     -
Ac/ConsumptionOnInput/L2/Voltage                                                                                                   -
Ac/ConsumptionOnInput/L3/Current                                                                                                   -
Ac/ConsumptionOnInput/L3/Power                                                                                                     -
Ac/ConsumptionOnInput/L3/Voltage                                                                                                   -
Ac/ConsumptionOnInput/NumberOfPhases                                                                                               1
Ac/ConsumptionOnOutput/Frequency                                                                                             49.8214
Ac/ConsumptionOnOutput/L1/Current                                                                                                0.4
Ac/ConsumptionOnOutput/L1/Power                                                                                                  112
Ac/ConsumptionOnOutput/L1/Voltage                                                                                              245.3
Ac/ConsumptionOnOutput/L2/Current                                                                                                  -
Ac/ConsumptionOnOutput/L2/Power                                                                                                    -
Ac/ConsumptionOnOutput/L2/Voltage                                                                                                  -
Ac/ConsumptionOnOutput/L3/Current

AlexdeKairos, 2024-02-24:
Here is my \System 

com.victronenergy.system
Ac/ActiveIn/Frequency                                                                                                        50.1026
Ac/ActiveIn/L1/Current                                                                                                          9.55Ac/ActiveIn/L1/Power                                                                                                            1846
Ac/ActiveIn/L1/Voltage                                                                                                        240.25
Ac/ActiveIn/L2/Current                                                                                                             -Ac/ActiveIn/L2/Power                                                                                                               -
Ac/ActiveIn/L2/Voltage                                                                                                             -
Ac/ActiveIn/L3/Current                                                                                                             -Ac/ActiveIn/L3/Power                                                                                                               -
Ac/ActiveIn/L3/Voltage                                                                                                             -
Ac/ActiveIn/NumberOfPhases                                                                                                         1Ac/ActiveIn/Source                                                                                                                 3
Ac/Consumption/Frequency                                                                                                     49.9488
Ac/Consumption/L1/Current                                                                                                       8.91Ac/Consumption/L1/Power                                                                                                         1720
Ac/Consumption/L1/Voltage                                                                                                     240.25
Ac/Consumption/L2/Current                                                                                                          -Ac/Consumption/L2/Power                                                                                                            -
Ac/Consumption/L2/Voltage                                                                                                          -
Ac/Consumption/L3/Current                                                                                                          -Ac/Consumption/L3/Power                                                                                                            -
Ac/Consumption/L3/Voltage                                                                                                          -
Ac/Consumption/NumberOfPhases                                                                                                      1Ac/ConsumptionOnInput/Frequency                                                                                              50.1026
Ac/ConsumptionOnInput/L1/Current                                                                                                   0
Ac/ConsumptionOnInput/L1/Power                                                                                                     0Ac/ConsumptionOnInput/L1/Voltage                                                                                              240.25
Ac/ConsumptionOnInput/L2/Current                                                                                                   -
Ac/ConsumptionOnInput/L2/Power                                                                                                     -Ac/ConsumptionOnInput/L2/Voltage                                                                                                   -
Ac/ConsumptionOnInput/L3/Current                                                                                                   -
Ac/ConsumptionOnInput/L3/Power                                                                                                     -Ac/ConsumptionOnInput/L3/Voltage                                                                                                   -
Ac/ConsumptionOnInput/NumberOfPhases                                                                                               1
Ac/ConsumptionOnOutput/Frequency                                                                                             49.9488Ac/ConsumptionOnOutput/L1/Current                                                                                               8.91
Ac/ConsumptionOnOutput/L1/Power                                                                                                 1720
Ac/ConsumptionOnOutput/L1/Voltage                                                                                             240.25Ac/ConsumptionOnOutput/L2/Current                                                                                                  -
Ac/ConsumptionOnOutput/L2/Power                                                                                                    -
Ac/ConsumptionOnOutput/L2/Voltage                                                                                                  -
[... cut ...]
Timers/TimeOnGenerator                                                                                                             0
Timers/TimeOnGrid                                                                                                               8006
Timers/TimeOnInverter                                                                                                           2724
VebusInstance                                                                                                                    276

AlexdeKairos, 2024-02-24:
@mr-manuel I'll stay in the boat for few hours up to 17h PM CET. If you need to test anithing. As youll see I updated the driver to: 1.2.20240220dev

AlexdeKairos, 2024-02-26:
@mr-manuel jus to let you know that my system config is published as requested...

mr-manuel, 2024-02-26:
Unfortunately we misunderstood each other. I need the `/System` values of the `com.victronenergy.battery.*` devices like this:

[image]

AlexdeKairos, 2024-02-27:
OK... sorry... @mr-manuel here they are

com.victronenergy.battery.ble_a5c2370595b6
Alarms/BmsCable                                                                                                                    0
Alarms/CellImbalance                                                                                                               -
Alarms/HighChargeCurrent                                                                                                           0
Alarms/HighChargeTemperature                                                                                                       0
Alarms/HighDischargeCurrent                                                                                                        0
Alarms/HighInternalTemperature                                                                                                     -
Alarms/HighTemperature                                                                                                             0
Alarms/HighVoltage                                                                                                                 0
Alarms/InternalFailure                                                                                                             -
Alarms/LowCellVoltage                                                                                                              -
Alarms/LowChargeTemperature                                                                                                        0
Alarms/LowSoc                                                                                                                      0
Alarms/LowTemperature                                                                                                              0
Alarms/LowVoltage                                                                                                                  0
Balances/Cell1                                                                                                                     0
Balances/Cell2                                                                                                                     0
Balances/Cell3                                                                                                                     0
Balances/Cell4                                                                                                                     0
Balancing                                                                                                                          0
Capacity                                                                                                                      197.75
Connected                                                                                                                          1
ConsumedAmphours                                                                                                              262.25
CurrentAvg                                                                                                                       6.5
CustomName                                                                                                            Bateria Serv 2
Dc/0/Current                                                                                                                    39.5
Dc/0/MidVoltage                                                                                                                    -
Dc/0/MidVoltageDeviation                                                                                                           -
Dc/0/Power                                                                                                                    526.14
Dc/0/Temperature                                                                                                                   -
Dc/0/Voltage                                                                                                                   13.32
DeviceInstance                                                                                                                     3
DeviceName                                                                                                                         -
FirmwareVersion                                                                                                      1.2.20240224dev
HardwareVersion                                                                                                                  2.8
Info/BatteryLowVoltage                                                                                                            12
Info/ChargeLimitation                                                                                     Max Battery Charge Current
Info/ChargeMode                                                                                                   Bulk (Linear Mode)
Info/ChargeModeDebug
Info/Config/AUTO_RESET_SOC                                                                                                         1
Info/Config/BATTERY_CAPACITY                                                                                                      50
Info/Config/BATTERY_CELL_DATA_FORMAT                                                                                               1
[... cut ...]
Voltages/Cell3                                                                                                                 3.343
Voltages/Cell4                                                                                                                 3.341
Voltages/Diff                                                                                                                  0.003
Voltages/Sum                                                                                                                   13.37

mr-manuel, 2024-02-27:
Could you test `v1.2.20240227dev`? Option 6 with the latest installer script:

```bash
wget -O /tmp/install.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh

bash /tmp/install.sh
```

If there is only one temperature sensor it is used as battery sensor.

AlexdeKairos, 2024-02-27:
Nice idea. Single temp sensor always on battery cells.
Unfortunately, there is an error and no BLE connection. here logs
2024-02-27 11:18:10.529207500 Device A5:C2:37:05:95:B6 (public)
2024-02-27 11:18:10.529218500   Alias: Bateria Serv 2
2024-02-27 11:18:10.529220500   Paired: no
2024-02-27 11:18:10.529221500   Trusted: no
2024-02-27 11:18:10.529222500   Blocked: no
2024-02-27 11:18:10.529224500   Connected: no
2024-02-27 11:18:10.529225500   LegacyPairing: no
2024-02-27 11:18:10.529227500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-02-27 11:18:10.529230500   RSSI: -58
2024-02-27 11:18:10.530128500
2024-02-27 11:18:11.922318500 INFO:SerialBattery:
2024-02-27 11:18:11.923035500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-27 11:18:11.925089500 INFO:SerialBattery:Venus OS v3.14
2024-02-27 11:18:11.925743500 INFO:SerialBattery:dbus-serialbattery v1.2.20240227dev

==> /data/log/dbus-blebattery.0/current <==
2024-02-27 11:18:17.923959500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:05:95:9E
2024-02-27 11:18:17.924928500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:05:95:9E
2024-02-27 11:18:22.937010500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2024-02-27 11:18:22.940435500 ERROR:SerialBattery:No BMS found at A5:C2:37:05:95:9E
2024-02-27 11:18:22.941297500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-02-27 11:18:25.010415500 INFO:SerialBattery:BLE client disconnected
2024-02-27 11:18:25.015829500 ERROR:SerialBattery:BleakClient(): Exception occurred: BleakError('failed to discover services, device disconnected') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #101
2024-02-27 11:18:25.344481500
2024-02-27 11:18:25.344904500 INFO:Bluetooth details

AlexdeKairos, 2024-02-27:
On the first restart another error came. 

2024-02-27 11:22:57.224888500 INFO:SerialBattery:
2024-02-27 11:22:57.225617500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-27 11:22:57.227680500 INFO:SerialBattery:Venus OS v3.14
2024-02-27 11:22:57.228388500 INFO:SerialBattery:dbus-serialbattery v1.2.20240227dev
2024-02-27 11:23:02.668509500 [CHG] Device A5:C2:37:05:95:B6 RSSI: -56
2024-02-27 11:23:13.802748500 INFO:SerialBattery:Init of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-27 11:23:13.803624500 INFO:SerialBattery:Test of LltJbd_Ble at A5:C2:37:05:95:B6
2024-02-27 11:23:19.310874500 ERROR:SerialBattery:Exception occurred: TypeError("unsupported operand type(s) for -: 'NoneType' and 'float'") of type <class 'TypeError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd.py line #275
2024-02-27 11:23:19.311937500 ERROR:SerialBattery:No BMS found at A5:C2:37:05:95:B6
2024-02-27 11:23:19.312633500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2024-02-27 11:23:19.312968500 #0
2024-02-27 11:23:19.312972500 #1
2024-02-27 11:23:19.312973500 #2
2024-02-27 11:23:19.312974500 #3

mr-manuel, 2024-02-27:
The problem was fixed with `v1.2.20240227`.

closed 2024-02-27 by mr-manuel