Louisvdw/dbus-serialbattery #981: JBD Serial Connection - Wrong SoC after restart and duplicate VRM ID

dersch81, 2024-02-23:
### Describe the bug

All Versions (instead of v1.0.20230531) create duplicate VRM Id's with JBD BMS with serial connection. Also after restarting dbus-serialbattery the SOC is wrong, for example it had counted 77% and after restart it shows 97%.  Due to = SoC: None%

The duplicate VRM Id is due to the new Serial Number field. I have 3 BMS. One has a unique serial and the two other get the same. I don't know from where it get the serial but i cant control this. It is not match the serial field i can set in the bms setting. 

### How to reproduce

Install any version later than v1.0.20230531 with USB connected JBD BMS

### Expected behavior

Take the serial out of the BMS serial setting in the right way and read the SoC value of the BMS

### Driver version

v1.1.20240121

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.21

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Battery count

3

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 85.0
MAX_BATTERY_DISCHARGE_CURRENT = 95.0
CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False
CCCM_SOC_ENABLE = False

MAX_CELL_VOLTAGE = 3.463

CUSTOM_BATTERY_NAMES = /dev/ttyACM0:Das Pack Drei,/dev/ttyACM1:Das Pack Zwei,/dev/ttyACM2:Das Pack Eins
```

### Relevant log output

```shell
@4000000065d91b59340bb524 INFO:SerialBattery:Testing Jkbms
@4000000065d91b5a08bfa59c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065d91b5a0949b174 INFO:SerialBattery:Testing Lifepower
[... 219 lines cut ...]
@4000000065d9218b2a859114 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065d9218b2a8b0784 INFO:SerialBattery:Testing HLPdataBMS4S
@4000000065d921900e0d38ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065d921900e19851c INFO:SerialBattery:Testing Jkbms
@4000000065d921901f52361c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065d921901fa38e54 INFO:SerialBattery:Testing Lifepower
@4000000065d9219030122124 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065d921903085322c INFO:SerialBattery:Testing LltJbd
@4000000065d921910a750a44 INFO:SerialBattery:Connection established to LltJbd
@4000000065d921910a8440cc INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM1
@4000000065d921910a84583c INFO:SerialBattery:========== Settings ==========
@4000000065d921910a872314 INFO:SerialBattery:> Connection voltage: 53.67V | Current: 24.46A | SoC: 77.29%
@4000000065d921910a95bd5c INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@4000000065d921910a96b75c INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000065d921910afca51c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 85.0A | MAX BATTERY DISCHARGE CURRENT: 95.0A
@4000000065d921910afcc074 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 300.0A | MAX BATTERY DISCHARGE CURRENT: 300.0A (read from BMS)
@4000000065d921910b01a65c INFO:SerialBattery:> CVCM:     True
@4000000065d921910b1107dc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.463V
@4000000065d921910b111f4c INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
@4000000065d921910b14b544 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
@4000000065d921910b1b0e44 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
@4000000065d9219113581c3c INFO:SerialBattery:DeviceInstance = 4
@4000000065d92191136ba054 INFO:SerialBattery:com.victronenergy.battery.ttyACM1
@4000000065d921911f416954 INFO:SerialBattery:publish config values = 1
```

mr-manuel, 2024-02-23:
Please install todays version, there the issue is fixed.

mr-manuel, 2024-02-24:
Since the JBD does not seem to provide a serial number that we can read, we have to create a unique identifier. In this case it contains also the capacity. Change the capacity of one battery by increasing or decreasing 1 Ah. The SoC is only None when starting. I'm working on this issue, but it is not that urgent.

dersch81, 2024-02-24:
It would be good to have that Unique ID configurable in the config.ini

Also the Soc is always none. I had it running for a couple of hrs and it did not change.

mr-manuel, 2024-02-24:
> It would be good to have that Unique ID configurable in the config.ini

That would make no sense. The unique identifier is needed to identify a BMS indipendent of where and how it is attached. Therefore the value has to be fetched from the BMS.

> Also the Soc is always none. I had it running for a couple of hrs and it did not change.

Please share a screenshot and dbus-spy of the battery where we can see the SoC value.

mr-manuel, 2024-02-25:
The problem with the None SoC at driver startup is solved in the next release.

mr-manuel, 2024-02-27:
The problem was fixed with `v1.2.20240227dev`.

closed 2024-02-27 by mr-manuel

gustogyver, 2024-02-28:
Where can I find the version  v1.2.20240227dev? Is there a download link somewhere or only over the install script?

mr-manuel, 2024-02-28:
https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases