Louisvdw/dbus-serialbattery #985: Serialnumber at device is double

mactoolz, 2024-02-25:
### Describe the problem

Hello,

i use the seplos BMS at my Victron system. At each seplos BMS i have configured the parameter to see which is the right storage at the driver, i configured the capacity parameter with different value.

But the driver shows me two serialnumber twice the same seralnumber.
I checked all my storages and there are different capacities values.

how can i correct the results at the driver.

Thank you

### Driver version

1.1.20240121

### Venus OS device type

Cerbo GX

### Venus OS version

3.21

### BMS type

Seplos

### Cell count

16

### Battery count

5

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = DEBUG

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 80.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.4
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;                - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
[... 388 lines cut ...]
@4000000065da2a532b20468c INFO: Create daemontools service gps-dbus.ttyUSB1
@4000000065da2a5439d4b5dc INFO: Create daemontools service gps-dbus.ttyUSB2
@4000000065da2a580a9071e4 INFO: Create daemontools service gps-dbus.ttyUSB3
@4000000065da2a5825d6264c INFO: Start service gps-dbus.ttyUSB0 once
@4000000065da2a59189c684c INFO: Create daemontools service gps-dbus.ttyUSB4
@4000000065da2a592dd9d8ac INFO: Start service gps-dbus.ttyUSB1 once
@4000000065da2a5b018adbfc INFO: Start service gps-dbus.ttyUSB2 once
@4000000065da2a5e0cfd91b4 INFO: Start service gps-dbus.ttyUSB3 once
@4000000065da2a5f1e944904 INFO: Start service gps-dbus.ttyUSB4 once
@4000000065da2a6231a7c4e4 INFO: Create daemontools service vedirect-interface.ttyUSB0
@4000000065da2a640001462c INFO: Create daemontools service vedirect-interface.ttyUSB1
@4000000065da2a650d8f0be4 INFO: Create daemontools service vedirect-interface.ttyUSB2
@4000000065da2a681dca2ad4 INFO: Create daemontools service vedirect-interface.ttyUSB3
@4000000065da2a68343c3eec INFO: Start service vedirect-interface.ttyUSB0 once
@4000000065da2a6928ccd8dc INFO: Create daemontools service vedirect-interface.ttyUSB4
@4000000065da2a6a027a1244 INFO: Start service vedirect-interface.ttyUSB1 once
@4000000065da2a6b1173b46c INFO: Start service vedirect-interface.ttyUSB2 once
@4000000065da2a6c381d0b2c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065da2a6e23cd5a14 INFO: Start service vedirect-interface.ttyUSB3 once
@4000000065da2a6e3143fa04 INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000065da2a6f2d87e434 INFO: Start service vedirect-interface.ttyUSB4 once
@4000000065da2a701ed3c304 INFO: Start service dbus-serialbattery.ttyUSB2 once
@4000000065da2a7400125d2c INFO: Start service dbus-serialbattery.ttyUSB3 once
@4000000065da2a751f30851c INFO: Start service dbus-serialbattery.ttyUSB4 once
```

mr-manuel, 2024-02-25:
Please read all carefully, which logs we need and how you get them: https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#driver-log-files

There are a few thing missing, we need.

mactoolz, 2024-02-25:
which logs will you see, i have 5 battery packs ?

mactoolz, 2024-02-25:
again my problem, all packs are working fine, but the problem is that the serial number for identifying a pack is created based on the capacity, e.g. for PAck1 = 281Ah.

Under the device in the Victron you can see the
Serial number = SeplosBMS16cells_281_0Ah 
This serial number does not change even if I change the capacity in the Seplos BMS, even to one that no longer exists.

mactoolz, 2024-02-25:
root@einstein:~# ls -l /dev/ttyUSB*
crw-rw----    1 root     dialout   188,   0 Feb 25 09:11 /dev/ttyUSB0
crw-rw----    1 root     dialout   188,   1 Feb 25 09:11 /dev/ttyUSB1
crw-rw----    1 root     dialout   188,   2 Feb 25 09:11 /dev/ttyUSB2
crw-rw----    1 root     dialout   188,   3 Feb 25 09:11 /dev/ttyUSB3
crw-rw----    1 root     dialout   188,   4 Feb 25 09:11 /dev/ttyUSB4

root@einstein:~# head /data/var/lib/serial-starter/*
==> /data/var/lib/serial-starter/ttyS4 <==
mkx

==> /data/var/lib/serial-starter/ttyS5 <==
vedirect

==> /data/var/lib/serial-starter/ttyS6 <==
vedirect

==> /data/var/lib/serial-starter/ttyS7 <==
vedirect

==> /data/var/lib/serial-starter/ttyUSB0 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB1 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB2 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB3 <==
sbattery

==> /data/var/lib/serial-starter/ttyUSB4 <==
sbattery

mr-manuel, 2024-03-01:
@mactoolz any update?

closed 2024-03-11 by mr-manuel