Louisvdw/dbus-serialbattery #992: Missing IO controls for Charge/discharge/Balance on JBD BLE BMS

AlexdeKairos, 2024-02-28:
### Describe the bug

Yesterday after update to 1.2.20240227dev the IO controls have disappeared.  Are no longer available for JBD BLE BMS
[image]

### How to reproduce

1 Go the dbus-seral battery IO section.
2 controls are no longer avail. Just the info of how they are configured but not the switches.

### Expected behavior

Below the info Allow to balance, there used to be 3 switches to control all options.

### Driver version

1.2.20240227dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.14

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
On the way
```

### Relevant log output

```shell
Shortly
```

mr-manuel, 2024-02-28:
Can you re-install this version with the installer script?

AlexdeKairos, 2024-02-28:
Thanks! Fixed!

closed 2024-02-28 by AlexdeKairos