Louisvdw/dbus-serialbattery #995: JKBMS with RS485 connection: Battery SoC does not reset to 100%

tfboy, 2024-03-01:
### Describe the problem

Battery SoC is slowly reducing, despite being fully charged. I know there's a "100% reset SoC" function available but I believe only with Bluetooth connection and not RS485 which is what I'm using.
I did read a thread which suggested some tweaks and the newer driver would fix the reset value which I believe JK say is not possible over serial, only BLE.
Right now, I believe the battery is fully charged showing over 3.5V per cell when solar is providing charge. No current is going in, battery is in absorption, but battery SoC never goes above 83% when I believe it should be 95%+.

### Driver version

v1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.700
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.510
FLOAT_CELL_VOLTAGE = 3.400

; --------- SOC reset voltage ---------
; Description: May be needed to reset the SoC to 100% once in a while for some BMS, because of SoC drift.
;              Specify the cell voltage where the SoC should be reset to 100% by the BMS.
;                - JKBMS: SoC is reset to 100% if one cell reaches OVP (over voltage protection) voltage
;              As you have to adopt this value to your system, I reccomend to start with
;              OVP voltage - 0.030 (see Example).
;                - Try to increase (add) by 0.005 in steps, if the system does not switch to float mode, even if
;                  the target voltage SOC_RESET_VOLTAGE * CELL_COUNT is reached.
;                - Try to decrease (lower) by 0.005 in steps, if the system hits the OVP too fast, before all
;                  cells could be balanced and the system goes into protection mode multiple times.
; Example: If OVP is 3.650, then start with 3.620 and increase/decrease by 0.005
; Note: The value has to be higher as the MAX_CELL_VOLTAGE
SOC_RESET_VOLTAGE = 3.525
; Specify after how many days the soc reset voltage should be reached again
; The timer is reset when the soc reset voltage is reached
; Leave empty if you don't want to use this
; Example: Value is set to 15
; day 1: soc reset reached once
; day 16: soc reset reached twice
[... 266 lines cut ...]
2024-03-01 17:32:36.834293500 INFO:SerialBattery:Testing HLPdataBMS4S
2024-03-01 17:32:37.655299500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-03-01 17:32:37.656285500 INFO:SerialBattery:Testing Jkbms
2024-03-01 17:32:37.802112500 INFO:SerialBattery:Connection established to Jkbms
2024-03-01 17:32:37.802852500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2024-03-01 17:32:37.803446500 INFO:SerialBattery:========== Settings ==========
2024-03-01 17:32:37.804110500 INFO:SerialBattery:> Connection voltage: 53.6V | Current: -0.0A | SoC: 82%
2024-03-01 17:32:37.804737500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2024-03-01 17:32:37.805384500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-03-01 17:32:37.806022500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.7V | MAX CELL VOLTAGE: 3.51V
2024-03-01 17:32:37.806692500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 120.0A | MAX BATTERY DISCHARGE CURRENT: 140.0A
2024-03-01 17:32:37.807447500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 120.0A (read from BMS)
2024-03-01 17:32:37.808130500 INFO:SerialBattery:> CVCM:     True
2024-03-01 17:32:37.808818500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-03-01 17:32:37.809548500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-03-01 17:32:37.810187500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-03-01 17:32:37.811203500 INFO:SerialBattery:Serial Number/Unique Identifier: XavBMS-1
2024-03-01 17:32:38.499508500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-03-01 17:32:38.522090500 INFO:SerialBattery:Remove /Settings/Devices/serialbattery_ttyUSB0 from dbus. Old entry. Delete result: False
2024-03-01 17:32:38.899433500 INFO:SerialBattery:DeviceInstance = 1
2024-03-01 17:32:38.901790500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-03-01 17:32:38.903703500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-03-01 17:32:38.905192500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-03-01 17:32:38.961731500 INFO:SerialBattery:publish config values = False
```

tfboy, 2024-03-01:
Sorry, forgot to say I'm fairly certain my cells are reaching 3.525V which is the SoC value, but it doesn't seem to do anything.
Maybe I should lower this? Is the value her required for just one cell or do all cells need to exceed the SoC reset value for it to actually reset?

mr-manuel, 2024-03-01:
There are many issues with the same behaviour. JKBMS setz  SOC to 95% when reaching OVPR and 100 % when one cell reaches OVP.

kommando828, 2024-03-02:
The JK firmware is suspect, here is a fudge fix from Youtube.

https://www.youtube.com/watch?v=maTqnSs_XAI

Title is Reset the old JK-BMS to 100% SOC when fully charged with no OVP. Here is how.

But it also covers the new version.

tfboy, 2024-03-03:
Thank you both. I've not yet tweaked the settings within the JK bluetooth app, but it seems to have reset to 95% today. I'll keep an eye on it and apply the workaround Andy described if I have the problem again. :)

closed 2024-03-04 by mr-manuel