Louisvdw/dbus-serialbattery #1001: CCL drops before reaching Absorption

NorthTown2022, 2024-03-08:
### Describe the problem

CCL drops before reaching Absorption:
[image]
[image]

### Driver version

v1.2.20240227beta

### Venus OS device type

Cerbo GX

### Venus OS version

v3.13

### BMS type

JKBMS / Heltec BMS

### Cell count

16s

### Battery count

1

### Connection type

Serial USB adapter to RS485

### Config file

```ini
GNU nano 4.9.3                                                                                      /data/etc/dbus-serialbattery/config.ini
[DEFAULT]
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

BMS_TYPE = Jkbms
LOGGING = DEBUG

MAX_BATTERY_CHARGE_CURRENT = 140
MAX_BATTERY_DISCHARGE_CURRENT = 100

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max ba                                                                             >
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V =                                                                             >
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.350
; SOC to reset CVL to allow MAX VOLTAGE
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 80

; --------- SOC reset voltage ---------
SOC_RESET_VOLTAGE = 3.620

; --------- SOC calculation ---------
SOC_CALCULATION = TRUE
SOC_RESET_CURRENT = 7
SOC_RESET_TIME = 60
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, 300
SOC_CALC_CURRENT_MEASURED_BY_USER = -300, 300

; Temperature sensor 1 name
TEMP_1_NAME = Temp Cells 4|5

; Temperature sensor 2 name
TEMP_2_NAME = Temp Cells 12|13

LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True
CCCM_CV_ENABLE = True
DCCM_CV_ENABLE = True
CCCM_T_ENABLE = True
DCCM_T_ENABLE = True

; --------- Cell Voltage limitation (affecting CVL) ---------
CVL_ICONTROLLER_MODE = True
CVL_ICONTROLLER_FACTOR = 0.2

;CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_TIME_RESTART = 0.030
VOLTAGE_DROP = 0.12
```

### Relevant log output

```shell
Dbus-serialbattery is installed and runnning.
```

mr-manuel, 2024-03-11:
You have to check what the `Charge Current Limitation` says in that moment the CCL reaches 0.

NorthTown2022, 2024-03-12:
I think the problem is somehow BMS related. Last night I reset the BMS by changing chemistry and reverting back to LiFeP04 and re-entering my settings: 
[image]

CCL went to zero at the same time max cell voltage hit 3.45 (same as SOC 100% in BMS) 
[image]
[image]

I have set SOC-100% to 3.460V in BMS and will see if the highest cell kicks out at 3.450V again tomorrow or perhaps it will hit absorption and float.

According to the config CCL should not hit zero until 3.55V?
[image]

NorthTown2022, 2024-03-12:
Just notice I have both ESS errors and cannot toggle inverter on in VRM dashboard. Need to go into remote console to turn it on.
[image]

NorthTown2022, 2024-03-13:
> You have to check what the `Charge Current Limitation` says in that moment the CCL reaches 0.

SoC has been the only limiting factor throughout charging: Here is a screenshot from the moment CCL reached zero.
[image]

mr-manuel, 2024-03-13:
The wording is wrong. "only" is not correct, if there is at least one limiting factor then the CCL is limited.

https://github.com/Louisvdw/dbus-serialbattery/blob/ac892a108ee8623effeac23ff64c899d0de32c18/etc/dbus-serialbattery/config.default.ini#L275-L281

You could try to change it to:

```ini
 SOC_WHILE_CHARGING                 =  100,   95,   90,   85 
 MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.02, 0.15, 0.50, 1.00 
```

NorthTown2022, 2024-03-13:
> You could try to change it to:
> 
> ```ini
>  SOC_WHILE_CHARGING                 =  100,   95,   90,   85 
>  MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.02, 0.15, 0.50, 1.00 
> ```

Making that change seemed to fix some issues. I will test this more in the coming days :)

[image]

With this new setting, I suppose the BMS will permanently allow a minimum of 2.80A and have a negligible effect on the duration of float transition?

[image]

mr-manuel, 2024-03-14:
Thats not entirely correct. It allows a minimum of `0.02`(which are 2%) of `MAX_BATTERY_CHARGE_CURRENT` at 100% SoC for the `CCCM_SOC_ENABLE` setting, which in your case are `2.80A`. BUT if you also have `CCCM_CV_ENABLE` or `CCCM_T_ENABLE` enabled then this parameters could also limit the `CCL`.

I would recommend to disable `CCCM_SOC_ENABLE` and `DCCM_SOC_ENABLE` since it's not that accurate as `CCCM_CV_ENABLE` and `DCCM_CV_ENABLE` is. The old limits where that high, that it had no real impact for most of the systems.

I will change the default behaviour from

```ini
; --------- SOC limitation (affecting CCL/DCL) ---------
; Description:
;     Maximal charge / discharge current will be increased / decreased depending on State of Charge,
;     see CC_SOC_LIMIT1 etc.
; Example:
;     The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = True
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
SOC_WHILE_CHARGING                 =  100,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.00, 0.15, 0.50, 1.00

SOC_WHILE_DISCHARGING              =    0,    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.00, 0.10, 0.20, 0.50, 1.00
```

to

```ini
; --------- SOC limitation (affecting CCL/DCL) ---------
; Description:
;     Maximal charge / discharge current will be increased / decreased depending on State of Charge,
;     see CC_SOC_LIMIT1 etc.
; Example:
;     The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = False
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = False

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
SOC_WHILE_CHARGING                 =   98,   95,   90,   85
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.10, 0.20, 0.50, 1.00

SOC_WHILE_DISCHARGING              =    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.10, 0.20, 0.50, 1.00
```

closed 2024-03-14 by mr-manuel