Louisvdw/dbus-serialbattery #1002: Multiple daly BMS -

11011010, 2024-03-11:
### Describe the bug

In any version after v1.0.20230531, my second daly battery no longer shows up in the "device list" but under "remote console" I see both.
[image]
[image]

I am using "battery aggregator" but I installed that later, it should have no influence and removing it shows the same result.

I have no idea if something has changed or if it is a bug in general.

### How to reproduce

I tried all versions of the drivers available, same result.

### Expected behavior

Both battaries show up in the "device list"

### Driver version

v1.1.2024121

### Venus OS device type

Cerbo GX

### Venus OS version

v3.22

### BMS type

Daly Smart BMS

### Cell count

16

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
std
```

### Relevant log output

```shell
Last login: Sun Mar 10 16:18:15 2024
root@einstein:~# tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
;
^C
root@einstein:~# tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
^C
root@einstein:~# tail -F /data/log/serial-starter/current | tai64nlocal
2024-03-10 16:18:30.342265500 INFO: Create daemontools service gps-dbus.ttyUSB0
2024-03-10 16:18:32.448067500 INFO: Create daemontools service gps-dbus.ttyUSB1
2024-03-10 16:18:36.648013500 INFO: Start service gps-dbus.ttyUSB0 once
2024-03-10 16:18:38.672907500 INFO: Start service gps-dbus.ttyUSB1 once
2024-03-10 16:18:47.386447500 INFO: Create daemontools service vedirect-interface.ttyUSB0
2024-03-10 16:18:49.184806500 INFO: Create daemontools service vedirect-interface.ttyUSB1
2024-03-10 16:18:53.592346500 INFO: Start service vedirect-interface.ttyUSB0 once
2024-03-10 16:18:55.421489500 INFO: Start service vedirect-interface.ttyUSB1 once
2024-03-10 16:18:59.440829500 INFO: Start service dbus-serialbattery.ttyUSB0 once
2024-03-10 16:19:01.417700500 INFO: Start service dbus-serialbattery.ttyUSB1 once
^C
```

mr-manuel, 2024-03-11:
> I have no idea if something has changed or if it is a bug in general.

Did you read the [CHANGELOG](https://github.com/Louisvdw/dbus-serialbattery/blob/master/CHANGELOG.md)?

Please read again, [which log files](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#serial-bms-connection) are needed. 

[image]

What is the `VRM Instance` and `Serial number` of both batteries? You can find it under `SerialBattery -> Device`.

11011010, 2024-03-12:
> > I have no idea if something has changed or if it is a bug in general.
> 
> Did you read the [CHANGELOG](https://github.com/Louisvdw/dbus-serialbattery/blob/master/CHANGELOG.md)?
> 
> Please read again, [which log files](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#serial-bms-connection) are needed.
> 
> ![grafik](https://private-user-images.githubusercontent.com/12592765/311785025-1782facf-2fd4-4c21-ad70-4ab619b219ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyMjk5NjIsIm5iZiI6MTcxMDIyOTY2MiwicGF0aCI6Ii8xMjU5Mjc2NS8zMTE3ODUwMjUtMTc4MmZhY2YtMmZkNC00YzIxLWFkNzAtNGFiNjE5YjIxOWFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDA3NDc0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0YWU1YWExMDcxNGE0Y2RhNDljMTMwNTQ1ODBjZWY4OWRjNDIzZWI5ZTYwZmVhZWI1MmY2NGFlYmM5NjJkMTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YqdOQ_sHPwNczMYdgCYShUAM1UPw4QZfalykPnjmp6E)
> 
> What is the `VRM Instance` and `Serial number` of both batteries? You can find it under `SerialBattery -> Device`.

Sorry, I should really read the changelog. Found the problem