Louisvdw/dbus-serialbattery #1005: Sync SOC of BMS with SOC of Smart- or Lynx-Shunt

RichieS67, 2024-03-11:
### Is your feature request related to a problem? Please describe.

The problem is that the SOC of the Daly BMS and the SOC of the Victron Lynx Shunt drift apart, especially if the SOC is not charged to 100% for a long time.

### Describe the solution you'd like

Maybe it is possible to sync SOC of BMS with more accurate SOC of Shunt.
Selection should be possible: FALSE/TRUE
And when it is TRUE the update interval should be selectable e.g. 10 min

### Describe alternatives you've considered

Maybe via NodeRed

BipedalPrimate, 2024-03-12:
I **strongly second** this request with an alternative implementation, but first a little digression.

I am nowhere near a competent Python coder but, looking at the daly.py code, it maybe that the Daly BMS SOC is reset to 100% when dbus-serialbattery changes to Float.  This suggests that dbus-serialbattery writes that SOC back to the Daly BMS.

However, that solution does not appear to be available for JKBMS, which I have.

So, @RichieS67, is there a long time between your Daly system achieving Float?

The solution, as currently implemented, to deal with incorrect BMS current readings is to specify in config.ini some adjusted current readings by:
```
; Example to set small currents to zero
;     SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300, -0.5, 0.5, 300
;     SOC_CALC_CURRENT_MEASURED_BY_USER = -300,    0,   0, 300
```
In systems without an external current monitor, this may work.  However, in systems with a SmartShunt or LynxShunt, it seems to be workaround for low quality BMS current sensors (which in the case of my JKBMS cannot be current calibrated to 0A).

After capturing and comparing the BMS & SmartShunt current readings over time, my settings for these parameters are:
```
SOC_CALC_CURRENT_REPORTED_BY_BMS  = -300.0, -4.9, -4.7, -4.5, -4.3, -4.1, -3.9, -3.7, -3.5, -3.3, -3.1, -2.9, -2.6, -2.4, -2.2, -1.8, -1.6, 0.0, 2.4, 2.6, 2.9, 3.1, 3.3, 3.5, 3.7, 3.9, 4.1, 4.3, 4.5, 4.7, 300.0
SOC_CALC_CURRENT_MEASURED_BY_USER = -300.0, -4.6, -4.3, -3.9, -3.7, -3.6, -3.4, -3.2, -3.0, -2.9, -2.9, -1.8, -1.1, -0.6, -0.4, -0.3, -0.2, 0.0, 0.4, 0.9, 1.4, 1.7, 2.3, 2.7, 2.9, 3.3, 3.6, 4.1, 4.3, 4.6, 300.0
```
This is probably a little excessive specification, but that is what the data suggests and my system spends a lot of time with very little current flow. 

Be that as it may, an alternative implementation of the enhancement.

VenusOS already has an option to specify which Battery Monitor to use.  I think that setting is available on dbus at:
```
com.victronenergy.system
Settings/SystemSetup/BatteryService                           com.victronenergy.battery/289
ActiveBatteryService                                          com.victronenergy.battery/289
ActiveBmsService                                              com.victronenergy.battery.ttyUSB2
Dc/Battery/BatteryService                                     com.victronenergy.battery.ttyUSB3
```

The path to any Battery Monitor SOC seems to be conditioned on those settings:
```
com.victronenergy.battery.ttyUSB2
Soc                                                                           38

```

```
com.victronenergy.battery.ttyUSB3
Soc                                                                            55.5
```
Yes, there is currently a 17.5% SoC difference and getting bigger day by day.  I trust the 55.5% from the SmartShunt.

For the JKBMS, the code in jkbms.py at 140-145 could be modified to refer to dbus to find the BatteryService and obtain SOC from that source.

```
        # the JKBMS resets to
        # 95% SoC, if all cell voltages are above or equal to OVPR (Over Voltage Protection Recovery)
        # 100% Soc, if all cell voltages are above or equal to OVP (Over Voltage Protection)
        offset = cellbyte_count + 18
        self.soc = unpack_from(">B", self.get_data(status_data, b"\x85", offset, 1))[0]

```
[https://github.com/Louisvdw/dbus-serialbattery/blob/ac892a108ee8623effeac23ff64c899d0de32c18/etc/dbus-serialbattery/bms/jkbms.py#L140-L145](url)
Note: The comment in that code about SoC reset to 95% if all cell voltages are above or equal to OVPR (Over Voltage Protection Recovery) does not work on my JKBMS.  Only the 100% SoC on OVP works (must be done manually via Android app) which disconnects the battery and generates various VRM alarms.

As the disclaimer in the first sentence mentions, I am no Python coder and don't know if this is realistic or achievable.

This approach would allow selection of Battery Monitor from VenusOS console:
[image]
[image]

RichieS67, 2024-03-12:
Yes, I selected the Lynx Shunt as the battery monitor, but I always get the low battery alarm messages from the Daly BMS. I don't know if this will shut down my system at some point even though there is still enough power in my batteries. My house's power consumption is also very low (200-300W). Therefore, the error of the Daly BMS is extremely large. Yesterday I had 55% SOC from the shunt and only 28% from the Daly BMS. With the new version of the dbus-serialbattery you can now manually set the SOC of the Daly BMS to any value from 0 to 100%. Therefore I think it must be possible to do this automatically.