Louisvdw/dbus-serialbattery #1006: Battery connection breaks at ESS min soc

Micha854, 2024-03-12:
### Describe the bug

we have 2 batteries. a pylontech 9.6kw connected via can and a diy 9.6kw with daly bms connected via rs485 to usb on the rasperry pi

Venus is the current version 3.22

our ess is set to a minimum soc of 35%. Every time the DIY memory reaches around 35% (+, - 2%) it is no longer displayed in the VRM portal or in the remote console

As soon as something has been recharged in the morning, the memory is visible again. Can the serial battery driver be responsible for this?

In the BMS we have set the min cell voltage to 2.8V and the min total voltage to 42V

### How to reproduce

no idea

### Expected behavior

I want the connection to the battery to remain when the min SOC is reached

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi 3

### Venus OS version

3.22

### BMS type

Daly Smart BMS

### Cell count

15

### Battery count

2

### Connection type

Serial USB adapter to RS485

### Config file

```ini
default config
```

### Relevant log output

```shell
dont have
```

mr-manuel, 2024-03-12:
You have to increase the Daly BMS sleep time with the PC software.

Micha854, 2024-03-14:
Super thank you! I would never have thought of the idea

I was able to set sleep time to the value 65535 using the PC software.

now the problem is solved. Thanks

closed 2024-03-14 by mr-manuel