Louisvdw/dbus-serialbattery #1012: Bluetooth connection to Liontron LiFePo4 with possible Jbk Bms

Thomasinator, 2024-03-19:
### Describe the bug

No connection to the bms

### How to reproduce

If the system is running the error is persistent there

### Expected behavior

Bluetooth connection should work

### Driver version

v1.2.20240318dev

### Venus OS device type

Raspberry Pi 4

### Venus OS version

v3.22

### BMS type

JKBMS / Heltec BMS

### Cell count

8

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = INFO

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0

; --------- Bluetooth BMS ---------
; Description:
;     Specify the Bluetooth BMS and it's MAC address that you want to install. Leave emty to disable
; Available Bluetooth BMS:
;     Jkbms_Ble, LltJbd_Ble
; Example for one BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
; Example for multiple BMS:
;     BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS = Jkbms_Ble A4:C1:38:AD:41:7C

; --------- SOC calculation ---------
SOC_CALCULATION   = False

; --------- Time-To-Go ---------
; Description:
;     Calculates the time to go shown in the GUI
;     Recalculation is done based on TIME_TO_SOC_RECALCULATE_EVERY
TIME_TO_GO_ENABLE = True

; Publish the config settings to the dbus path "/Info/Config/"
PUBLISH_CONFIG_VALUES = True

; -- Daly settings
; Battery capacity (amps), if the BMS does not support reading it
BATTERY_CAPACITY = 110
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1
```

### Relevant log output

```shell
2024-03-19 20:37:41.720557500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-03-19 20:37:43.061406500 ERROR:SerialBattery:No BMS found at A4:C1:38:AD:41:7C
2024-03-19 20:37:43.061703500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-03-19 20:37:43.807202500 Starting bluetooth: bluetoothd.
2024-03-19 20:37:43.826031500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-03-19 20:37:46.013329500
2024-03-19 20:37:46.013433500 INFO:Bluetooth details
2024-03-19 20:37:46.039025500 Attempting to disconnect from A4:C1:38:AD:41:7C
2024-03-19 20:37:46.039031500 Successful disconnected
2024-03-19 20:37:51.082950500 Device A4:C1:38:AD:41:7C (public)
2024-03-19 20:37:51.082955500   Alias: 032224100147
2024-03-19 20:37:51.082958500   Paired: no
2024-03-19 20:37:51.082960500   Trusted: no
2024-03-19 20:37:51.082962500   Blocked: no
2024-03-19 20:37:51.082964500   Connected: no
2024-03-19 20:37:51.082967500   LegacyPairing: no
2024-03-19 20:37:51.082969500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-03-19 20:37:51.082972500   RSSI: -31
2024-03-19 20:37:51.083948500
2024-03-19 20:37:51.780503500 INFO:SerialBattery:
2024-03-19 20:37:51.780937500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-19 20:37:51.781946500 INFO:SerialBattery:Venus OS v3.22
2024-03-19 20:37:51.782340500 INFO:SerialBattery:dbus-serialbattery v1.2.20240318dev
2024-03-19 20:38:08.042697500 INFO:SerialBattery:Init of Jkbms_Ble at A4:C1:38:AD:41:7C
2024-03-19 20:38:08.043170500 INFO:SerialBattery:Test of Jkbms_Ble at A4:C1:38:AD:41:7C
2024-03-19 20:38:09.229022500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-03-19 20:38:09.229579500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-03-19 20:38:09.232853500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-03-19 20:38:09.306923500 [CHG] Device A4:C1:38:AD:41:7C RSSI: -31
2024-03-19 20:38:09.306928500 [CHG] Device A4:C1:38:AD:41:7C RSSI: -31
2024-03-19 20:38:09.306931500 [CHG] Device A4:C1:38:AD:41:7C RSSI is nil
2024-03-19 20:38:11.326270500 Starting bluetooth: bluetoothd.
2024-03-19 20:38:11.344080500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-03-19 20:38:14.517215500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-03-19 20:38:14.517759500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-03-19 20:38:14.520739500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-03-19 20:38:16.611020500 Starting bluetooth: bluetoothd.
2024-03-19 20:38:16.629505500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-03-19 20:38:17.558470500 ERROR:SerialBattery:No BMS found at A4:C1:38:AD:41:7C
2024-03-19 20:38:17.558475500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-03-19 20:38:18.814621500
2024-03-19 20:38:18.814712500 INFO:Bluetooth details
2024-03-19 20:38:18.839814500 Attempting to disconnect from A4:C1:38:AD:41:7C
2024-03-19 20:38:18.839819500 Successful disconnected
2024-03-19 20:38:23.883694500 Device A4:C1:38:AD:41:7C (public)
2024-03-19 20:38:23.883698500   Alias: 032224100147
2024-03-19 20:38:23.883701500   Paired: no
2024-03-19 20:38:23.883703500   Trusted: no
2024-03-19 20:38:23.883706500   Blocked: no
2024-03-19 20:38:23.883708500   Connected: no
2024-03-19 20:38:23.883710500   LegacyPairing: no
2024-03-19 20:38:23.883713500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-03-19 20:38:23.883716500   RSSI: -37
2024-03-19 20:38:23.885538500
2024-03-19 20:38:24.588405500 INFO:SerialBattery:
2024-03-19 20:38:24.588906500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-19 20:38:24.590910500 INFO:SerialBattery:Venus OS v3.22
2024-03-19 20:38:24.590915500 INFO:SerialBattery:dbus-serialbattery v1.2.20240318dev
2024-03-19 20:38:40.858853500 INFO:SerialBattery:Init of Jkbms_Ble at A4:C1:38:AD:41:7C
2024-03-19 20:38:40.859379500 INFO:SerialBattery:Test of Jkbms_Ble at A4:C1:38:AD:41:7C
2024-03-19 20:38:42.227780500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-03-19 20:38:42.228442500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-03-19 20:38:42.232912500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-03-19 20:38:42.324001500 [CHG] Device A4:C1:38:AD:41:7C RSSI: -37
2024-03-19 20:38:42.324006500 [CHG] Device A4:C1:38:AD:41:7C RSSI: -31
2024-03-19 20:38:42.324010500 [CHG] Device A4:C1:38:AD:41:7C RSSI is nil
2024-03-19 20:38:44.320240500 Starting bluetooth: bluetoothd.
2024-03-19 20:38:44.338395500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-03-19 20:38:47.367318500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakDBusError('org.bluez.Error.Failed', 'Software caused connection abort') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #474
2024-03-19 20:38:47.367676500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-03-19 20:38:47.371054500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-03-19 20:38:49.462054500 Starting bluetooth: bluetoothd.
2024-03-19 20:38:49.480177500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-03-19 20:38:50.371813500 ERROR:SerialBattery:No BMS found at A4:C1:38:AD:41:7C
2024-03-19 20:38:50.372129500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-03-19 20:38:51.669664500
2024-03-19 20:38:51.669754500 INFO:Bluetooth details
2024-03-19 20:38:51.695143500 Attempting to disconnect from A4:C1:38:AD:41:7C
2024-03-19 20:38:51.695149500 Successful disconnected
```

mr-manuel, 2024-03-20:
Did you reboot Venus OS?

Which JKBMS do you have? Harware and Software version?

Thomasinator, 2024-03-21:
I made a mistake by my research what BMS type is in a liontron, not JK. There should be a JBD inside.
So i think it is not possible to connect to this type of BMS with your library?
Or can i provide something for you to implement it?

mr-manuel, 2024-03-21:
Just check the docs. JKBMS and JBDBMS are supported. Update your config file accordingly and it should work.

mr-manuel, 2024-03-31:
Any update on this?

closed 2024-04-02 by mr-manuel