Louisvdw/dbus-serialbattery #1013: Jdb Bluetooth connection to battery not stable

dsrawding, 2024-03-20:
### Describe the bug

The Pi has no problem connecting to the battery and data flows for a few seconds then disconnects.  I have added a second Bluetooth adapter hoping it would help but it does not.  I have tried all available versions.  With debugging on I notice this error:  ERROR: BMS send insufficient data. Received 40 expected 41.  That seems to be when it dies.  The XiaoXiang app runs fine and has no issue staying connected to the battery.
[image]
[image]
[image]

### How to reproduce

Install and run bus-serialbattery.

### Expected behavior

Data would be received from battery more than once

### Driver version

1.2.20240227beta

### Venus OS device type

Raspberry Pi 3

### Venus OS version

v3.30

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

1

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Set logging level ---------
; ERROR: Only errors are logged
; WARNING: Errors and warnings are logged
; INFO: Errors, warnings and info messages are logged
; DEBUG: Errors, warnings, info and debug messages are logged
LOGGING = WARNING
BLOCK_ON_DISCONNECT = False

; --------- Bluetooth BMS ---------
; Description:
;     Specify the Bluetooth BMS and it's MAC address that you want to install. Leave emty to disable
; Available Bluetooth BMS:
;     Jkbms_Ble, LltJbd_Ble
BLUETOOTH_BMS = LltJbd_Ble A5:C2:37:08:0A:89
```

### Relevant log output

```shell
2024-03-20 15:01:53.285119500 *** CCGX booted (0) ***
tail: /data/log/dbus-blebattery.0/current has been replaced; following end of new file
2024-03-20 15:03:07.576519500 
[... 130 lines cut ...]

### Any other information that may be helpful

bluetoothctl
[Lisuateli12V100aH]# info
Device A5:C2:37:08:0A:89 (public)
	Name: Lisuateli12V100aH
	Alias: Lisuateli12V100aH
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010203-0405-0607-0809-0a0b0c0d1912)
	Modalias: usb:v248Ap8266d0001
	ManufacturerData Key: 0x0a89
	ManufacturerData Value:
  08 37 c2 a5                                      .7..            
	RSSI: -48
	AdvertisingFlags:
  06                                               .

dsrawding, 2024-03-20:
I have been trying different versions and trying to get this to work for a few days with no luck.

mr-manuel, 2024-03-20:
I have no idea on how to fix that. Some help of the community would be appreciated. The Bluetooth connection is also not stable for the JKBMS, maybe we have some design mistake or anything else.

dsrawding, 2024-03-22:
Can someone point me to what code parses the data from the BMS?  I'm happy to try to figure out what piece of data is missing or modify my local code to only look for 40 rather than 41.

Marvo2011, 2024-03-26:
> I have added a second Bluetooth adapter hoping it would help but it does not.

Have you tried latest master with `BLUETOOTH_USE_USB = True` in config.ini?
Have your Bluetooth adapter BLE support ? Bluetooth version >4.0

Marvo2011, 2024-03-26:
@dsrawding Can you post your output from `hciconfig -a`

dsrawding, 2024-03-26:
I don't believe the issue was with the Bluetooth.  I tested by disabling the internal interface and it didn't make a difference.  I have given up on using Bluetooth and cut the battery case open and connected using the uart port on the BMS and it's working with dubs-serialbattery.  Thanks.

closed 2024-03-26 by dsrawding