Louisvdw/dbus-serialbattery #1018: SOC reset voltage not reachable and SOC not resetting.

Rutgerdek, 2024-03-22:
### Describe the problem

Probably a user error, but here's what I encounter.

I'm unable to get the system to charge to SOC_RESET_VOLTAGE as this voltage needs to be higher than MAX_CELL_VOLTAGE, at which the Victron system stops charging. I've set both values at 0.001v apart and the battery sees that voltage regularly for a few seconds but no reset.

I also tried raising the voltages, thus charging the batteries, then lovering them and rebooting the Cerbo. This causes the cells to be higher then SOC_RESET_VOLTAGE for a few minutes but still no reset.

### Driver version

1.120240121

### Venus OS device type

Cerbo GX

### Venus OS version

3.13 (LargeOS)

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

8

### Battery count

2

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.295
; Max voltage (can seen as absorption voltage)
FLOAT_CELL_VOLTAGE = 3.294
; Float voltage (can be seen as resting voltage), lager dan max
SOC_RESET_VOLTAGE = 3.296
; moet hoger zijn dan max_cell_voltage
SOC_RESET_AFTER_DAYS = 1

LINEAR_LIMITATION_ENABLE = False
CVCM_ENABLE = False
CCCM_CV_ENABLE = False
DCCM_CV_ENABLE = False
CCCM_T_ENABLE = False
DCCM_T_ENABLE = False
TIME_TO_GO_ENABLE = False
BMS_TYPE = LltJbd
EXCLUDED_DEVICES = /dev/ttyusb0
CUSTOM_BATTERY_NAMES = /dev/ttyusb1
[... 316 lines cut ...]
@4000000065fd457f3006b35c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065fd458012c3db1c ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
@4000000065fd459113fc052c INFO:SerialBattery:
@4000000065fd4591140e50c4 INFO:SerialBattery:Starting dbus-serialbattery
@4000000065fd45911424e21c INFO:SerialBattery:dbus-serialbattery v1.1.20240121
@4000000065fd45a1152d93d4 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000065fd45a115388c6c INFO:SerialBattery:Testing LltJbd
@4000000065fd45a12663916c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000065fd45a2095d54cc INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@4000000065fd45a2095d7bdc INFO:SerialBattery:Testing LltJbd
@4000000065fd45a21b49409c ERROR:SerialBattery:>>> ERROR: No reply - returning
```

### Any other information that may be helpful

This might be completely irrelevant but when I'm opening config.default.ini externally, via file explorer in "MeshCentral", the file starts at:

d on dbus [Valid values 0,1,2,3]
; 0 Do not publish all the cells (only the min/max cell data as used by the default GX)
; 1 Format: /Voltages/Cell (also available for display on Remote Console)
; 2 Format: /Cell/#/Volts
; 3 Both formats 1 and 2
BATTERY_CELL_DATA_FORMAT = 1

All text above this is just missing. If I open the file (via SSH) locally on the Cerbo, the whole file is there so probably just an issue with MeshCentral.

Rutgerdek, 2024-03-22:
Hi Manuel,

Thanks for your answer. What do you mean by check the parameters page in the GUI?
In the Cerbo I've tried disabling DVCC and changing/turning off BMS control, ESS is at "Keep batteries charged".

I'm now at:
MAX_CELL_VOLTAGE   = 3.295
FLOAT_CELL_VOLTAGE = 3.285
SOC_RESET_VOLTAGE = 3.305

Changing that seems to have changed things a bit. The MPPT's are now charging to 26.64v/3.33v. System voltage is at 26.50v now, with all cells above 3.305v. Still, no reset though.

The driver does indeed start and sometimes the connection with the battery drops (it was a bug then and I assumed it still is). It always comes back within a minute though. I've had that issue with multiple interfaces and with Venus as well as Cerbo.

Logging changed to debug and attached current
[current.txt](https://github.com/Louisvdw/dbus-serialbattery/files/14723454/current.txt)
[current2.txt](https://github.com/Louisvdw/dbus-serialbattery/files/14723456/current2.txt)
 logfiles.

mr-manuel, 2024-03-22:
> What do you mean by check the parameters page in the GUI?

Open the Remote Console/GUI and navigate to your Battery -> Parameters. Then post a screenshot of that page.

> Still, no reset though.

I think you missunderstood the function. The driver itself is not resetting the SOC, but the BMS. Different BMS resets the SOC at different cell voltages with different settings. You have to check first at which voltage your BMS resets the SOC to 100% and then set this voltage - some little diff like `0.005` V.

E.g. if your BMS resets the SOC at `3.600` V then try to set `SOC_RESET_VOLTAGE` to `3.595` V. You have to try and see which voltage fits best for you. There are many other issues where this is described. https://github.com/Louisvdw/dbus-serialbattery/issues?q=SOC_RESET_VOLTAGE

Rutgerdek, 2024-03-22:
Hi Manuel, attached the screenshots of both batteries.
I'm expecting the driver to overrule the BMS setting that is indeed resetting SOC at 28.6v/8. I understand my current settings are crazy low but it's just for testing.

Sorry but I cannot find any topics in your link that explain how the system should behave when SOC_RESET_VOLTAGE is reached. In the config, I read: "

_"Specify the cell voltage where the SoC should be reset to 100% by the BMS."_

That is exactly what I want, to overrule the BMS setting and tell it to reset SOC at a lower voltage than the setting it has in it's own configuration. If that's not how I should read the sentence above, please let me know.
[image]
[image]

mr-manuel, 2024-03-22:
> "Specify the cell voltage where the SoC should be reset to 100% by the BMS."

The driver and the BMS need to work together. The driver cannot override BMS settings.

mr-manuel, 2024-03-22:
Now I found the threat I meaned: https://github.com/Louisvdw/dbus-serialbattery/issues/703

Rutgerdek, 2024-03-26:
That's unfortunate, I was indeed looking for a way to "unconditionally" override the BMS settings. As my target cell voltage and the BMS settings are 0.1125v apart, I now understand this is not going to work anyway.

Thanks anyway for your support and great battery driver! Good to see active development and support.

mr-manuel, 2024-03-26:
Then you could use the soc calculation from the driver.

Rutgerdek, 2024-03-26:
Thanks again, I will give that a try.

closed 2024-03-26 by Rutgerdek