Louisvdw/dbus-serialbattery #1019: Unstable Cerbo after  Update from 3.0 to  3.2/3.3/3.4

gsxarne, 2024-03-22:
### Describe the problem

Hey guys, I have a vicctron ees running for more than year now , it was running fine at  Venus Os 3.0 and for some strange reason i got the idea to update all the Software packages...it was a bad Idea

After going to  3.2 (with the Serialbattery Version from  a year ago) Problems started. The Connection to the Cerbo got very unstable. This results in a lost connection on the remote console, ssh connection also dont work. MQTT connection is also lost. Sometimes the Cerbo reboots itself, sometimes it works again without rebooting.

Some days later 3.3 was released , but problems havent changed. Same for Beta3.4

I also deleted and reinstalled the SerialBattery Driver  Version to 0.1.2 but still the cerbo crashes a after some minutes of operation.

This morning i have vacation and used may time to analyze the problem a bit deeper.
I am now pretty sure that this problem is caused by the serialbattery driver (sorry to say).
When i disconnect all devices (multiplus, 2 Meters, the Serial  Connection (via  USB) to a JKBMS) the cerbo runs stable.

If i connect all Meters and the Multiplus, the Cerbo runs stable (testes for about 2h).
If i connect the Battery the Cerbo crashed twice within the last 30minutes.

Since the System worked fine for over a year, i would say that  the hardware is fine and it cant be related  to any EMC / Shield issues.

Unfortunatley the tai64nlocal command seems not to provide any helpfull information. But have a look at your own (attached below)

### Driver version

1.0.20230531

### Venus OS device type

Cerbo GX

### Venus OS version

3.3

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Battery count

1

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 35
 MAX_BATTERY_DISCHARGE_CURRENT = 70

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 3.000
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; Charge current control management referring to cell-voltage enable (True/False).
CCCM_CV_ENABLE = True
; Discharge current control management referring to cell-voltage enable (True/False).
DCCM_CV_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING   = 3.45, 3.420, 3.4, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.2,  0.5,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 3, 3.1, 3.15,
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    

; --------- Temperature limitation (affecting CCL/DCL) ---------
; Description: Maximal charge / discharge current will be in-/decreased depending on temperature
; Example: The temperature limit will be monitored to control the currents. If there are two temperature senors,
;          then the worst case will be calculated and the more secure lower current will be set.
; Charge current control management referring to temperature enable (True/False).
CCCM_T_ENABLE = True
; Charge current control management referring to temperature enable (True/False).
DCCM_T_ENABLE = True

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  15, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 1,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 15, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3, 0.4,  1,  1,  0

; --------- SOC limitation (affecting CCL/DCL) ---------
; Description: Maximal charge / discharge current will be increased / decreased depending on State of Charge,
;              see CC_SOC_LIMIT1 etc.
; Example: The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = True
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = True

; Charge current soc limits
CC_SOC_LIMIT1 = 98
CC_SOC_LIMIT2 = 95
CC_SOC_LIMIT3 = 91

; Charge current limits
CC_CURRENT_LIMIT1_FRACTION = 1
CC_CURRENT_LIMIT2_FRACTION = 1
CC_CURRENT_LIMIT3_FRACTION = 1

; Discharge current soc limits
DC_SOC_LIMIT1 = 10
DC_SOC_LIMIT2 = 20
DC_SOC_LIMIT3 = 30

; Discharge current limits
DC_CURRENT_LIMIT1_FRACTION = 0.1
DC_CURRENT_LIMIT2_FRACTION = 0.3
DC_CURRENT_LIMIT3_FRACTION = 0.5

; --------- Additional settings ---------
; Specify only one BMS type to load else leave empty to try to load all available
; -- Available BMS:
; Daly, Ecs, HeltecModbus, HLPdataBMS4S, Jkbms, Lifepower, LltJbd, Renogy, Seplos
; -- Available BMS, but disabled by default:
; https://louisvdw.github.io/dbus-serialbattery/general/install#how-to-enable-a-disabled-bms
; Ant, MNB, Sinowealth
BMS_TYPE =  Jkbms
```

### Relevant log output

```shell
@4000000065fd61870392d544 INFO:SerialBattery:Starting dbus-serialbattery
@4000000065fd61870518a63c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000065fd61870579f07c INFO:SerialBattery:Testing Jkbms
@4000000065fd61870ed270f4 INFO:SerialBattery:Connection established to Jkbms
@4000000065fd61870ee7ecf4 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
@4000000065fd61870f161944 INFO:SerialBattery:========== Settings ==========
@4000000065fd61870f547234 INFO:SerialBattery:> Connection voltage: 53.91V | Current: 17.0A | SoC: 88%
@4000000065fd61870f61b4bc INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
@4000000065fd61870f6e0ce4 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000065fd61870f7b91d4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
@4000000065fd61870f8847bc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 80.0A (read from BMS)
@4000000065fd61870ffab0cc INFO:SerialBattery:> CVCM:     True
@4000000065fd6187103350e4 INFO:SerialBattery:> MIN CELL VOLTAGE: 3.0V | MAX CELL VOLTAGE: 3.45V
@4000000065fd6187108e4ddc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000065fd618710a010bc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000065fd618710ad7284 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@4000000065fd618710b941f4 INFO:SerialBattery:Serial Number/Unique Identifier: JK-B2A24S15P
@4000000065fd61871c1f842c INFO:SerialBattery:DeviceInstance = 1
@4000000065fd61871c2cd26c INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
@4000000065fd61872980b3d4 INFO:SerialBattery:publish config values = 1
```

mr-manuel, 2024-03-22:
There are no Venus OS 3.2, 3.3 and 3.4 versions. Probably you mean 3.20, 3.30 and 3.40.

Also there is a BIG message that no one wants to read, when opening an issue.

[image]

What is your system load? You can check it with `uptime` and/or `top` command.

gsxarne, 2024-03-22:
Hi Manuel,
yes i was talking about 3.20,3.30  and so on.
sorry for ignoring the important notice.  Actually i concentrated on the "how to troubleshoot" Guide. Maybe you should add the info about installing the nightly there, too.

I installed 1.2.20240318dev, and will monitor the  behaviour this afternoon.

Systemload seems to be normal (with 1.2.20240318dev), serialbattery running at 1-2 percent:

CPU:  26% usr   6% sys   0% nic  65% idle   0% io   1% irq   0% sirq
Load average: 3.70 2.10 1.12 1/303 4629

I'll give you a feedback later wheter the cerbo crashes  again or not.

mr-manuel, 2024-03-22:
The load already seems pretty high. Is this only because you recently restarted? You can also check with htop: https://github.com/mr-manuel/venus-os_helpful-scripts/tree/master/htop/armv7

Albarge, 2024-03-22:
I also had similar issues yesterday after deciding to update GuiMods to v10.16 and SetupHelper from v7.1 to v7.8 and VenusOS from v3.22 to v3.30. I had already been running dbus-serialbattery v1.2.20240227beta for a few days with no issues, but decided to edit the config file to include the: CVL_ICONTROLLER_MODE   = True 

After doing a serialbattery reinstall I lost all communication with my 3 JK BMS's and had the dreaded ERROR #67 and multiple shutdowns. There were some GuiMods issues from v7.1 -v7.8, but I think my issue also may have been related to serialbattery.

I'll try the Nightly too, as I've had to run the system dumb overnight until I could figure out what to check.
I did wonder if having my old custom battery names in the config could have confused the latest beta that doesn't use them?

CerboGX GuiMods
JK BMS x 3 RS485-USB
Battery Aggregator

gsxarne, 2024-03-22:
> The load already seems pretty high. Is this only because you recently restarted? You can also check with htop: https://github.com/mr-manuel/venus-os_helpful-scripts/tree/master/htop/armv7

Her is a screenshot from htop. Now CPU load is constantly above 30% with serialbattery below 1%

[image]

gsxarne, 2024-03-22:
The Issues don't  change with 1.2.20240318dev

Albarge, 2024-03-22:
Yep, v1.0.20240102beta seems stable for me running VenusOS v3.30, but both v1.2.20240227beta and 1.2.20240318dev didn't show connected BMS's.

mr-manuel, 2024-03-22:
Are there no errors in the log? You should see some errors, if the driver crashes. If there are no errors, check if the system restarted.

Albarge, 2024-03-22:
This is my log now, but only seems to be today since reverting back to the older driver, I don't really know what I'm looking for though, but I can see some errors.

/data/etc/dbus-serialbattery$ tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
2024-03-22 14:57:43.555701500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:44.557163500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:45.558619500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:46.559372500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:47.560727500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:48.561390500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:49.562042500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:57:50.568285500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:58:03.582721500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:58:04.584167500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:58:05.586042500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:58:06.587760500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 14:58:07.588409500 ERROR:SerialBattery:[Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
2024-03-22 15:08:43.758641500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:205/35329]
2024-03-22 15:36:03.422655500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:205/35329]
2024-03-22 15:48:18.650431500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:254/22019]
2024-03-22 16:19:50.723147500 INFO:SerialBattery:
2024-03-22 16:19:50.723841500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-22 16:19:50.725777500 INFO:SerialBattery:dbus-serialbattery v1.0.20240102beta
2024-03-22 16:20:06.741291500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-03-22 16:20:06.742170500 INFO:SerialBattery:Testing Jkbms
2024-03-22 16:20:06.808506500 INFO:SerialBattery:Connection established to Jkbms
2024-03-22 16:20:06.809638500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2024-03-22 16:20:06.810533500 INFO:SerialBattery:========== Settings ==========
2024-03-22 16:20:06.811689500 INFO:SerialBattery:> Connection voltage: 27.99V | Current: 2.75A | SoC: None%
2024-03-22 16:20:06.812376500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 0
2024-03-22 16:20:06.813067500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2024-03-22 16:20:06.813775500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 170.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A
2024-03-22 16:20:06.814657500 INFO:SerialBattery:> CVCM:     True
2024-03-22 16:20:06.815468500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2024-03-22 16:20:06.816184500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  False
2024-03-22 16:20:06.816834500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-03-22 16:20:06.817449500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-03-22 16:20:06.818031500 INFO:SerialBattery:Serial Number/Unique Identifier: JK_Bms_2
2024-03-22 16:20:17.427009500 INFO:SerialBattery:Found existing battery with DeviceInstance = 3
2024-03-22 16:20:18.283808500 INFO:SerialBattery:DeviceInstance = 3
2024-03-22 16:20:18.285055500 INFO:SerialBattery:Used device instances: ['3', '2', '1']
2024-03-22 16:20:18.286656500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-03-22 16:20:18.363911500 INFO:SerialBattery:publish config values = True
2024-03-22 16:20:27.560649500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-03-22 16:20:29.559977500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-03-22 16:25:12.725620500 INFO:SerialBattery:
2024-03-22 16:25:12.736100500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-22 16:25:12.738541500 INFO:SerialBattery:dbus-serialbattery v1.0.20240102beta
2024-03-22 16:25:28.758525500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2024-03-22 16:25:28.758532500 INFO:SerialBattery:Testing Jkbms
2024-03-22 16:25:29.084668500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-03-22 16:25:29.587674500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2024-03-22 16:25:29.587683500 INFO:SerialBattery:Testing Jkbms
2024-03-22 16:25:29.868283500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2024-03-22 16:25:30.382378500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2024-03-22 16:25:30.382387500 INFO:SerialBattery:Testing Jkbms
2024-03-22 16:25:30.451056500 INFO:SerialBattery:Connection established to Jkbms
[... cut ...]
2024-03-22 16:38:50.998216500 INFO:SerialBattery:DeviceInstance = 3
2024-03-22 16:38:50.999305500 INFO:SerialBattery:Used device instances: ['3', '2', '1']
2024-03-22 16:38:51.000511500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2024-03-22 16:38:51.198672500 INFO:SerialBattery:publish config values = True
2024-03-22 16:39:15.336468500 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:14/47560]
2024-03-22 16:39:17.074684500 INFO:SerialBattery:Saved MaxVoltageStartTime. Before 1711125555, after 1711125555

gsxarne, 2024-03-23:
So on my side  the log file didn't show any errors:

> @4000000065ff30ad072af7f4 *** starting serial-starter ***
@4000000065ff30ad2e20b164 serstart starting
@4000000065ff30ad2f63570c INFO: loading config file /etc/venus/serial-starter.conf
@4000000065ff30ae27eeb994 INFO: loading config file /data/conf/serial-starter.d/dbus-serialbattery.conf
@4000000065ff30b03b10c7ec INFO: Create daemontools service dbus-cgwacs.ttyACM0
@4000000065ff30b31028aa54 INFO: Create daemontools service mk2-dbus.ttyS4
@4000000065ff30b42b920f74 INFO: Create daemontools service vedirect-interface.ttyS5
@4000000065ff30b60e5ab884 INFO: Create daemontools service vedirect-interface.ttyS6
@4000000065ff30b712400134 INFO: Start service dbus-cgwacs.ttyACM0 once
@4000000065ff30b7262991c4 INFO: Create daemontools service vedirect-interface.ttyS7
@4000000065ff30b91691c754 INFO: Create daemontools service dbus-serialbattery.ttyUSB0
@4000000065ff30b91ef27a4c INFO: Start service mk2-dbus.ttyS4
@4000000065ff30bb0082841c INFO: Start service vedirect-interface.ttyS5
@4000000065ff30bc03ba9cdc INFO: Create daemontools service dbus-cgwacs.ttyUSB1
@4000000065ff30bc1a637c4c INFO: Start service vedirect-interface.ttyS6
@4000000065ff30bd2e492cac INFO: Start service vedirect-interface.ttyS7
@4000000065ff30bf249787e4 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000065ff30c218b2326c INFO: Start service dbus-cgwacs.ttyUSB1 once
> 

When it  crashes and restart  this log file overwrites itself.

I wanted to  install an old Version of serialbattery, v.0.14.3 for example.
When i use the install script, chose 7 - specific version, paste the link from github (https://github.com/Louisvdw/dbus-serialbattery/releases/download/v0.14.3/venus-data.tar.gz) , the installprocess stops with this error after downloading the tar-file:

tar: conf/serial-starter.d: Cannot open: File exists
-sh: tar:: command not found

Do you  know what went wrong here?

I also went back to Venus OS V3.01, but my issues with your nightly is still there.

I could imagine that my GridMeter (EM540) could  cause the  issues. it is  reporting values with a quite high frequency. I will unplug it  and give you a feedback.

Best Regards

Albarge, 2024-03-23:
So what version are you on that is still causing lost connections?
Did you try the version that worked for me: v1.0.20240102beta?

gsxarne, 2024-03-24:
So yeah, Serialbattery and the EM540 are fighting against each other causing the Cerbo to crash:

The whole night the system was running stable in Phasthrough Mode, so the cerbo was only logging the data from Serialbattery (and HomeAssistant logged the Cell-Voltages via MQTT).

This morning I plugged in the EM540. It worked for  ~1.5h and crashed again. This time also the Connection to the BMS was lost according to the Cerbo and i had to unplug the USB Connection (for what ever reason?)

To Sum up what i found out so far:

After Going from Venus OS 3.00 to 3.20 and later 3.30 and back to V3.00

- Cerbo is running stable with EM540 without Serialbattery running
- Cerbo is running stable with Serial Battery running but without Em540 connected
- Crashes occur with nightlys, beta and final releases of Serialbattery
- Crashes occur with  Venus Os V3.00  but also V3.30 and Beta 3.40
- when cerbo crashes the log files get overwritten (or i make  something wrong?)
- Installing v0.14  with serialbattery installer gives error (file already exists, see above)

mr-manuel, 2024-03-24:
Please specify what for crashes. Does the driver or Venus OS reboot?

> when cerbo crashes the log files get overwritten (or i make something wrong?)

The logs are rotated. Use
~~~
tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/* | tai64nlocal
~~~
to see all.

> Installing v0.14 with serialbattery installer gives error (file already exists, see above)

See https://louisvdw.github.io/dbus-serialbattery/general/install#downgrade-from--v100-to--v0143

> Could it be, that the Cerbo cannot deliver enough power for 3 USB Connections?

Yes. There are lots of issues because also the USB hub gave problems.

gsxarne, 2024-03-24:
> Please specify what for crashes. Does the driver or Venus OS reboot?
The Cerbo Crashes 

> > when cerbo crashes the log files get overwritten (or i make something wrong?)
> 
> The logs are rotated. Use
> 
> ```
> tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/* | tai64nlocal
> ```
> 
> to see all.

I'll use it  and report later
 
> See https://louisvdw.github.io/dbus-serialbattery/general/install#downgrade-from--v100-to--v0143
>
>
> Yes. There are lots of issues because also the USB hub gave problems.

Now i  let it  run with one the EM540 and Serialbattery, but without the USBHub and without the ET340  (PV Meter).
If the USB Hub is the root cause, why was it  never a problem before all the Update from Venus Os 3.00  to 3.20 and updating serial battery from 0,14 to the nightly / beta?

gsxarne, 2024-03-24:
Ok, also wihtout USB Hub & the ET340, these "Crashes" occur... So the issues is probably not related to to any USBHub related Issue.

I used
`tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/* | tai64nlocal`

but the files in /data/log/dbus-serialbattery.ttyUSB0 are anyhow created on restart.
I added the content of the putty console to the attached txt file.
[log.txt](https://github.com/Louisvdw/dbus-serialbattery/files/14735533/log.txt)

gsxarne, 2024-03-24:
Soooo
i have news...

I took my  Spare-Cerbo, which i had on stock in case I brick my current one (which has a high probability due to my limited Linux-Knowlegde) :-D

I used the V0.14 of Serialbattery and Venus OS 3.30, AND .....got the same issues.
BUT, after i disconnected the Comunication of my EM540 and could actually use the RemoteConsolei got an Info that my Multiplus should get a Software Update. Hell Yeah, after the experience of the past days, another Software Update made me anxious (insert Ralph Wigum's I am in Danger meme). But obviously i am not  capable of  learning and so I hit the button and  updated the MK3 Version (now 1170216), and after sweating of  fear  the system is at least up and running again.

CPU load is now lower,rather 20-30% than 30-40% like before. But will it  help? I doubt it and will report tomorrow.

gsxarne, 2024-03-25:
I still had 5 "loss  of communication" since my  last post.

darkdaydreamer, 2024-03-26:
i have also the same issue.... i have 2 JKBMS and only one is working since the update.... i tried
2) latest release "v1.1.20240121" (mr-manuel's repo, stable, most up to date)
4) beta build "v1.2.20240227beta" (mr-manuel's repo, no errors after 72 h runtime, long time testing needed)
6) nightly build "v1.2.20240318dev" (mr-manuel's repo, newest features and fixes, bugs possible)
and also the link from above
https://github.com/Louisvdw/dbus-serialbattery/releases/download/v1.0.20240102beta/venus-data.tar.gz

in case i added the working JKBMS the trail look like 

2024-03-26 19:01:09.356506500
2024-03-26 19:01:11.990921500 INFO:SerialBattery:
2024-03-26 19:01:11.992300500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-26 19:01:11.999613500 INFO:SerialBattery:dbus-serialbattery v1.0.20240102beta
2024-03-26 19:01:12.581343500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E8:E0:62
2024-03-26 19:01:12.582415500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E8:E0:62
2024-03-26 19:01:17.102519500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20240118)
2024-03-26 19:01:17.113570500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-03-26 19:01:17.114743500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce8e062
2024-03-26 19:01:17.119311500 INFO:SerialBattery:========== Settings ==========
2024-03-26 19:01:17.120561500 INFO:SerialBattery:> Connection voltage: 53.23V | Current: 0.0A | SoC: None%
2024-03-26 19:01:17.126381500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-03-26 19:01:17.131688500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-03-26 19:01:17.135877500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 400.0A
2024-03-26 19:01:17.138752500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A (read from BMS)
2024-03-26 19:01:17.143210500 INFO:SerialBattery:> CVCM:     True
2024-03-26 19:01:17.144590500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-03-26 19:01:17.145886500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-03-26 19:01:17.147204500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-03-26 19:01:17.148550500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-03-26 19:01:17.149816500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092447590
2024-03-26 19:01:18.117762500 [CHG] Device C8:47:8C:E8:E0:62 RSSI: -64
2024-03-26 19:01:21.635118500 INFO:SerialBattery:Remove /Settings/Devices/serialbattery_2092447188 from dbus. Old entry. Delete result: False
2024-03-26 19:01:21.636082500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-03-26 19:01:21.898517500 INFO:SerialBattery:DeviceInstance = 1
2024-03-26 19:01:21.899383500 INFO:SerialBattery:Used device instances: ['1']
2024-03-26 19:01:21.900261500 INFO:SerialBattery:com.victronenergy.battery.c8478ce8e062
2024-03-26 19:01:21.902316500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20240118)
2024-03-26 19:01:21.938610500 INFO:SerialBattery:publish config values = True

in case i try to add the not working one following was written to the log
2024-03-26 17:07:14.029740500 INFO:SerialBattery:
2024-03-26 17:07:14.030907500 INFO:SerialBattery:Starting dbus-serialbattery
2024-03-26 17:07:14.032654500 INFO:SerialBattery:dbus-serialbattery v1.1.20240121
2024-03-26 17:07:14.596695500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E8:E0:94
2024-03-26 17:07:14.597800500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E8:E0:94
2024-03-26 17:07:20.118210500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20240119)
2024-03-26 17:07:20.121195500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-03-26 17:07:20.122543500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce8e094
2024-03-26 17:07:20.123373500 INFO:SerialBattery:========== Settings ==========
2024-03-26 17:07:20.124917500 INFO:SerialBattery:> Connection voltage: 52.7V | Current: -30.0A | SoC: None%
2024-03-26 17:07:20.125133500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-03-26 17:07:20.128180500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-03-26 17:07:20.129898500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 400.0A
2024-03-26 17:07:20.135057500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A (read from BMS)
2024-03-26 17:07:20.135798500 INFO:SerialBattery:> CVCM:     True
2024-03-26 17:07:20.136511500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2024-03-26 17:07:20.137277500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-03-26 17:07:20.137996500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-03-26 17:07:20.138643500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2024-03-26 17:07:20.139407500 INFO:SerialBattery:Serial Number/Unique Identifier: 2092447188
2024-03-26 17:07:25.185917500 Traceback (most recent call last):
2024-03-26 17:07:25.185926500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 239, in <module>
2024-03-26 17:07:25.185930500     main()
2024-03-26 17:07:25.185932500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 222, in main
2024-03-26 17:07:25.185935500     if not helper.setup_vedbus():
[... cut ...]
2024-03-26 17:07:28.415627500 [CHG] Device C8:47:8C:E8:E0:94 RSSI: -64

when i take a look with dbus-spy the working one will stay there.... the not working one just apear short and disapear directly again.....

possible some interesting info.... today I updated from 3.13 to 3.30..... and yeah... as I know it in front I told to my friend I am not sure if I should do.... oh, i switched to large OS with this step.... but not sure if that has also an effect (needing large OS was the reason for my fail =D )

THX in advance!

mr-manuel, 2024-03-28:
In this issue there is now to much confusion and too many different errors. Please open a new issue with fresh logs of the latest version (updated today) and write also which was the latest version where all worked properly.

gsxarne, 2024-03-31:
Good News Everybody , at least i hope so.

Yesterday i recognized that Home Assitant had an issue with one Entity which was sourced from a 
Victron Parameter via MQTT. Since i  haverepaired  it  (roughly 18hs ago) i had no loss  of  communication or Cerbo reboots.

Please don't ask me how a MQTT client (Home Assistant) can mess up the Broker so much, that it crashe, And please dont ask me why it started after the Updates...but apperently there is a correlation between not having this Entity-Issue and a stable Cerbo. ¯\_(ツ)_/¯

So please  excuse me for falsely accusing SerialBattery. I  will pay tribute later

mr-manuel, 2024-03-31:
Thanks for the feedback. If everything is OK please close the issue.

closed 2024-04-01 by gsxarne