Louisvdw/dbus-serialbattery #1024: CCL and DCL must go to 0a in the event of Fet soft Charge and/or Discharge goes to off

AlexdeKairos, 2024-03-27:
### Is your feature request related to a problem? Please describe.

Hi, Im thinkig about scenarios with several batteries. I the event that one of the batteries goes to protection due to for instance, an overload cell, and closes the charge, it CCL will remain untouched and the remaining batteries may become overcharged with more than its actual CCL. Same happends if during a high discharge one cell goes to protection and ce BMS closes the discharge, its DCL should go to 0a to avoid the remaining batteries in paralell go tu overdischarge.
Battery agregators sum the CCL to obtain whatever the chargers can put in the battery bank. Lets say we have 2 batteries with 100a CCL each. Chargers and MPPT will limit the max charge to 200a. If one of the bateries goes to protection and closes its Charge fets, the other battery will recive, if available, 200a for charge because the driver will still report 100a for each battery. Damaging its cells of the life one.

### Describe the solution you'd like

In the event of a battery closes the charge, its asociated CCL should go to 0a (reason maybe something like: Charge Closed by BMS . And as soon it's opened allowing to charge, it should go to whatever limit is calculated by the driver.

Similar with the discharge. If the battery closes the discharge, its DCL should go to 0a. (reason like: Discharge closed by BMS)
Understand that IO manu is not avail to control in all BMS trought the driver but can be read in many BMSs brands.
Mine is JBD_BLE for instance.

mr-manuel, 2024-03-28:
Do the `Allow to charge` and `Allow to discharge` switch to `No` in this cases, right?

AlexdeKairos, 2024-03-28:
Yes, if properly read, in my JBD_BLE works properly, it goes to "No" when battery BMS closes charge or discharge.

mr-manuel, 2024-03-28:
Download the latest nightly and test please.

mr-manuel, 2024-03-31:
Enhancement included from driver version `v1.2.20240401`.

closed 2024-03-31 by mr-manuel

AlexdeKairos, 2024-04-01:
Hi Manuel, sorry but i've been busy all week without access to my boat. I have tried 1.2.20240401dev but in my JBD BLE bms the function is not working. One of my batteries is in Allow to Charge NO and Allow to discharge NO, and still the CCL is 115 and DCL is 150a. So mod is not working atm with JBD BLE.

mr-manuel, 2024-04-01:
I found a typo. Please reinstall this version.

mr-manuel, 2024-04-01:
You can no longer control the switches through the dbus-serialbattery driver?

AlexdeKairos, 2024-04-01:
I can actually switch off discharge but can't go ON again from GUI. Switches actually go active or desactive but there is no change on Yes or No. I was able to force discharging OFF but the it does not respond anymore and YES is not avail. Balancing works both directions. For some reason charging off was not responding but after a several minutes delay, it went to No. no possible to recover to Yes regardless of the switch position. 
Unfortunatelly today I must leave the boat but expect to come back tomorrow.

mr-manuel, 2024-04-01:
Please open a new issue for that.