Louisvdw/dbus-serialbattery #1030: Force charging OFF and F. Discharging OFF works slow but no longer can be disabled

AlexdeKairos, 2024-04-01:
### Describe the bug

Testing new functionality on CCL and DCL I have discovered that Forcing OFF works but when disabled, Allow to Charge and Allow to discharge remain in No all the time regardless of the switch position.  Switch behaviour is correct and works, but Allow charge and discharge, not.

### How to reproduce

Just go to IO menu, activate Force charging off , wait for Allow to charge to go No. Then deactivate Force Charging off and whait for Allow to charge to go Yes. It remains in No.

Same happens with Force discharging. 

### Expected behavior

Upon disabling,  the BMS should open again the fees and show "
[image]
[image]
Yes"

### Driver version

1.2.20240401dev

### Venus OS device type

Cerbo GX

### Venus OS version

3.30

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Battery count

2

### Connection type

Bluetooth

### Config file

```ini
Irrelevant
```

### Relevant log output

```shell
NA
```

AlexdeKairos, 2024-04-01:
I dont really understand your question. Can be more specific?

mr-manuel, 2024-04-02:
1. What do you see in the Bluetooth App? Is there any indication that charging/discharging is disabled? If the switch in the driver is off, then the BMS decides, if charging/discharging is allowed. If the switch is on, then it forces to be off.
2. Open the GUI/Remote Console -> Serialbattery -> Parameters. Post a screenshot of this page.

AlexdeKairos, 2024-04-02:
OK. Yes I understand. In the app the indication is in agreement with the GUI IO Yes or No indications. Changing from the Overkill App works and is properly reflected in the GUI by the driver. (I had to install a serial connection... bluetooth was used. anyhow, the Gui can disable the charging or discharging, but does not allow to tound then back on. It was previously possible before.
You already have the screen shoot in the report. 
IO have been testing even auto off, changind overvioltage settings in the BMS and they are properly reflected in the GUI (without switch movement as expected)

mr-manuel, 2024-04-02:
What was the latest driver version where it worked?

AlexdeKairos, 2024-04-02:
Just before the CCL and DCL TO 0a.

mmy6x7, 2024-04-02:
In the same boat as @AlexdeKairos, been messing with /data/etc/dbus-serialbattery/bms/lltjbd.py.  I can get the "force disable charge" toggle to work again when I added sections and followed the log for debugging.  Sections updated:
```  
def force_charging_off_callback(self, path, value):
        if value is None:                
            return False                 
                                                     
        if value == 0:                                      
            self.trigger_force_disable_charge = False       
            self.control_allow_charge = True                
            return True                                     
                                                            
        if value == 1:                                                                 
            self.trigger_force_disable_charge = True                                   
            return True                                                                
                                                                                                           
        return False                                               
```
Then just below it for the write:
```
   def write_charge_discharge_mos(self):                                                                  
        if (                                                                                               
            self.trigger_force_disable_charge is None                                                      
            and self.trigger_force_disable_discharge is None                                               
        ):                                                  
            return False                                    
                                                                                 
        charge_disabled = 0 if self.charge_fet else 1                            
        if self.trigger_force_disable_charge is not None and self.control_allow_charge:                          
            logger.info( f"allow val : {self.control_allow_charge}" )                                            
            charge_disabled = 1 if self.trigger_force_disable_charge else 0                                      
            logger.info( f"charge_disabled val : {charge_disabled}" )                                            
            logger.info( f"fet val : {self.charge_fet}" )                                                        
            logger.info(                                                                                         
                f"write force disable charging: {'true' if self.trigger_force_disable_charge else 'false'}"      
            )                                                                                              
        self.trigger_force_disable_charge = None  
```
Reran the the reinstall (bash /data/etc/dbus-serialbattery/reinstall-local.sh) and followed along in the log:
```
2024-04-02 22:25:12.070637500 INFO:SerialBattery:Starting dbus-serialbattery
2024-04-02 22:25:12.072654500 INFO:SerialBattery:Venus OS v3.13
2024-04-02 22:25:12.073280500 INFO:SerialBattery:dbus-serialbattery v1.2.20240401
2024-04-02 22:25:28.614864500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:31:56:9C
2024-04-02 22:25:28.615688500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:31:56:9C
2024-04-02 22:25:35.509041500 INFO:SerialBattery:Connection established to LltJbd_Ble
2024-04-02 22:25:38.350187500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-04-02 22:25:38.660180500 INFO:SerialBattery:DeviceInstance = 1
2024-04-02 22:25:38.662525500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-04-02 22:25:38.663676500 INFO:SerialBattery:Used DeviceInstances = ['1']
2024-04-02 22:25:38.664860500 INFO:SerialBattery:com.victronenergy.battery.ble_a4c13731569c
2024-04-02 22:25:39.770852500 INFO:SerialBattery:publish config values = True
2024-04-02 22:25:39.805030500 INFO:SerialBattery:Battery LltJbd_Ble connected to dbus from ble_a4c13731569c
2024-04-02 22:25:39.805689500 INFO:SerialBattery:========== Settings ==========
2024-04-02 22:25:39.806434500 INFO:SerialBattery:> Connection voltage: 26.43V | Current: 2.28A | SoC: None%
2024-04-02 22:25:39.807044500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 8
2024-04-02 22:25:39.807620500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2024-04-02 22:25:39.808230500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.6V
2024-04-02 22:25:39.808840500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 110.0A
2024-04-02 22:25:39.809443500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 110.0A | MAX BATTERY DISCHARGE CURRENT: 110.0A (read from BMS)
2024-04-02 22:25:39.810135500 INFO:SerialBattery:> CVCM:     True
2024-04-02 22:25:39.811142500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2024-04-02 22:25:39.811721500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2024-04-02 22:25:39.812378500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-04-02 22:25:39.813045500 INFO:SerialBattery:Serial Number/Unique Identifier: a4c13731569c
2024-04-02 22:26:29.864990500 INFO:SerialBattery:allow val : True
2024-04-02 22:26:29.865714500 INFO:SerialBattery:charge_disabled val : 1
2024-04-02 22:26:29.866326500 INFO:SerialBattery:fet val : True
[... cut ...]
2024-04-02 22:29:58.111868500 INFO:SerialBattery:fet val : True
2024-04-02 22:29:58.112450500 INFO:SerialBattery:write force disable charging: true
2024-04-02 22:30:25.142934500 INFO:SerialBattery:allow val : True
2024-04-02 22:30:25.142942500 INFO:SerialBattery:charge_disabled val : 0
2024-04-02 22:30:25.143584500 INFO:SerialBattery:fet val : False
2024-04-02 22:30:25.147013500 INFO:SerialBattery:write force disable charging: false
```
Seems to work almost immediately now when toggling that value in the cerbo.

AlexdeKairos, 2024-04-04:
Hi @mmy6x7. I'm not very good with programing skills. This mods are for writing both in the lltjbd.py?
Would you mind to attach the file already modified to test it?

mr-manuel, 2024-04-04:
Could you all try to install `v1.2.20240404dev`?

mr-manuel, 2024-04-04:
Can you write it on Discord, so I can test on your system?

mr-manuel, 2024-04-04:
Main problem was solved, but in some weird circumstances charging/discharing needs to be enabled first that discharging/charing is working then. Maybe @idstein has an idea why?

AlexdeKairos, 2024-04-04:
Hi I found some that may help in fixinfg this. On my Battery One upn selection of Force charging OFF allow to charge changes from YES to NO in and enless loop and parametes Charge limitation goes from BMS CCL 0a to Max Batery charge current in an endless loop.

May be @mr-manuel something related with CCL to 0 when Allow to charge goes to No?

closed 2024-04-08 by Louisvdw